zigpy/zigpy-zigate #1: Tag release versions and make PyPi release(s) of zigpy-zigate

Hedda, 2019-01-22:
Please remember to tag an initial release version of zigpy-zigate and make PyPi release of zigpy-zigate.

@doudz I guess that you are probably not yet to make the first release of this zigpy-zigate library but I just wanted to add a friendly reminder here under issues for tracking purposes only, so there is no rush.

An initial PyPi release version and then continues versions/releases will make it easier for Linux OS distributions like Hass.io and Raspbian to update and make use of each new versions of this library.

Version tagging and PyPi release(s) is something that is already available for the zigpy library as well as similar radio libraries that is depends on such as; bellows, zigpy-deconz, and zigpy-xbee

https://github.com/zigpy/zigpy/releases
https://github.com/zigpy/bellows/releases
https://github.com/zigpy/zigpy-xbee/releases
https://github.com/zigpy/zigpy-deconz/releases

https://pypi.org/project/zigpy
https://pypi.org/project/bellows
https://pypi.org/project/zigpy-xbee
https://pypi.org/project/zigpy-deconz

Oh! And one more thing; FIRST! ;)  ...though I should add don't actually own ZiGate hardware myself

Hedda, 2019-01-22:
Maybe you could talk @rcloran @AndreasBomholtz @damarco & @Adminiuga about having your zigpy-zigate repository move the GitHub repo to the zigpy organization @ https://github.com/zigpy/ ?

Perhaps they would prefer reserving names with "zigpy" prefix on PyPi together so that more than one person has access to that project for maintenance? Hope that you could all come to an agreement :)

Hedda, 2019-03-20:
@doudz maybe talk with the existing Zigpy devs now about you joining their org and moving the repo?

Again, congratulations on what looks to be great progress with your zigpy-zigate project for Zigpy!

doudz, 2019-03-20:
it's to early, the current zigpy-zigate is based on zigate lib , it's overkill since zigate lib as the same purpose as zigpy.
now I'm gonna remove this dependency before saying zigpy-zigate is ready

Hedda, 2019-08-20:
@doudz Maybe good idea to now tag 0.1.0 version in this repo?

https://github.com/doudz/zigpy-zigate/releases

That is to match the commit revision used by the PyPI release:

https://pypi.org/project/zigpy-zigate/

Best practice would be to always try to tag release on GitHub first

doudz, 2019-08-20:
Yes you're right

closed 2019-09-11 by doudz