zigpy/zigpy-zigate #4: Init avec ZHA

Adorem, 2019-08-25:
Hello @doudz,

Aujourd'hui j'ai voulu essayer ma Zigate (flashée avec le firmware 3.1a) dans Home Assistant (0.98.0b1) avec ZHA et donc zigpy-zigate et je dois dire que ce n'est pas très concluant.

Pour mes tests j'ai utilisé une lampe Philips hue et une commande Philips.

Le premier problème rencontré c'est lorsque que j'essaye d'appairé les devices depuis HA lorsque je fais Configuration > ZHA > ADD Devices la zigate ne passe pas toujours en mode appairage (LED bleu qui flash). Souvent ça fonctionne la première fois et pas les suivantes. Quand je suis dans ce cas là je dois redémarrer mon système complet (pas seulement HA mais aussi hassio).

Ma lampe n'est pas détectée correctement (unk_manufacturer unk_model) et ma commande ne fonctionne pas non plus.

Je vais essayer d'aller plus loin dans le debug mais si tu des pistes à suivre ou des choses à faire je prends.

[image]

doudz, 2019-08-25:
Merci pour ce retour.
J'ai déjà eu ce même phénomène unk_manufacturer etc avec un capteur xiaomi mais je ne sais pas trop d'où ça vient.
il faut que je fasse plus de test

Adorem, 2019-08-26:
Merci pour cette réponse.
Et pour le zha.permit qui ne fonctionne pas à chaque fois une idée ?

doudz, 2019-08-26:
Là je ne vois pas. attention la différence par rapport à mon composant est le délai de 60sec contre 30sec avant

Rommie19, 2019-08-27:
i think i have the same problem

the zigate will only go in to join mode after a reboot (blue LED blinks, all the other times the LED stays on)
When it is in join mode i can only add 1 device at the time

founded:
[image]

output:
`Device 0x35d1 (90:fd:9f:ff:fe:19:24:a9) joined the network
[0x35d1] Requesting 'Node Descriptor'
[0x35d1] Node Descriptor: <Optional byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4476 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0x35d1] Discovering endpoints
[0x35d1] Discovered endpoints: [1]
[0x35d1:1] Discovering endpoint information
[0x35d1:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=49246 device_type=544 device_version=2 input_clusters=[0, 3, 4, 5, 6, 8, 768, 2821, 4096] output_clusters=[5, 25, 32, 4096]>
[0x35d1:zdo] ZDO request 0x0013: [0x35d1, 90:fd:9f:ff:fe:19:24:a9, 142]
Error calling listener.attribute_updated: Error binding parameter 0 - probably unsupported type.
Error calling listener.attribute_updated: Error binding parameter 0 - probably unsupported type.
[0x35d1:1] Manufacturer: IKEA of Sweden
[0x35d1:1] Model: TRADFRI bulb E14 WS opal 400lm
Error calling listener.raw_device_initialized: Error binding parameter 0 - probably unsupported type.
Checking quirks for IKEA of Sweden TRADFRI bulb E14 WS opal 400lm (90:fd:9f:ff:fe:19:24:a9)
Considering <class 'zigpy.quirks.ikea.TradfriPlug'>
Fail because endpoint list mismatch: {1, 2, 242} {1}
Considering <class 'zigpy.quirks.smartthings.SmartthingsMultiPurposeSensor'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XbeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because profile_id mismatch on at least one endpoint
device - 0x35d1:90:fd:9f:ff:fe:19:24:a9 entering async_device_initialized - is_new_join: True
device - 0x35d1:90:fd:9f:ff:fe:19:24:a9 has joined the ZHA zigbee network
[0x35d1](TRADFRI bulb E14 WS opal 400lm): started configuration
[0x35d1:ZDO](TRADFRI bulb E14 WS opal 400lm): channel: 'async_configure' stage succeeded
[0x35d1:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0x35d1:1:0x0006]: reporting 'on_off' attr on 'on_off' cluster: 0/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0x35d1:1:0x0006]: finished channel configuration
[0x35d1:1:0x0006]: channel: 'async_configure' stage succeeded
[0x35d1:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0x35d1:1:0x0000]: finished channel configuration
Error calling listener.attribute_updated: Error binding parameter 0 - probably unsupported type.
[0x35d1:1:0x0000]: initializing channel: from_cache: False
[0x35d1:1:0x0000]: channel: 'async_configure' stage succeeded
[0x35d1:1:0x0008]: Failed to bind 'level' cluster: 
Error calling listener.attribute_updated: Error binding parameter 0 - probably unsupported type.
[0x35d1:1:0x0300]: initializing channel: from_cache: False
[0x35d1:1:0x0300]: bound 'light_color' cluster: Status.SUCCESS
[0x35d1:1:0x0300]: reporting 'current_x' attr on 'light_color' cluster: 30/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0x35d1:1:0x0300]: reporting 'current_y' attr on 'light_color' cluster: 30/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0x35d1:1:0x0300]: reporting 'color_temperature' attr on 'light_color' cluster: 30/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0x35d1:1:0x0300]: finished channel configuration
[0x35d1:1:0x0300]: channel: 'async_configure' stage succeeded
[0x35d1:1:0x0005]: bound 'scenes' cluster: Status.SUCCESS
[0x35d1:1:0x0005]: finished channel configuration
[0x35d1:1:0x0005]: channel: 'async_configure' stage succeeded
[... 14 lines cut ...]
light.ikea_of_sweden_tradfri_bulb_e14_ws_opal_400lm_fe1924a9_1: polling current state
[0x35d1:1:0x0008]: initializing channel: from_cache: False
[0x35d1:1:0x0008]: channel: 'async_initialize' stage succeeded
[0x35d1:1:0x0006]: initializing channel: from_cache: False
[0x35d1:1:0x0006]: channel: 'async_initialize' stage succeeded
[0x35d1:1:0x1000]: initializing channel: from_cache: False
[0x35d1:1:0x1000]: channel: 'async_initialize' stage succeeded
[0x35d1:1:0x0005]: initializing channel: from_cache: False
[0x35d1:1:0x0005]: channel: 'async_initialize' stage succeeded
[0x35d1:1:0x0000]: initializing channel: from_cache: False
[0x35d1:1:0x0000]: channel: 'async_initialize' stage succeeded
Error calling listener.attribute_updated: Error binding parameter 0 - probably unsupported type.
[0x35d1:1:0x0300]: channel: 'async_initialize' stage succeeded
[0x35d1](TRADFRI bulb E14 WS opal 400lm): power source: Mains
[0x35d1](TRADFRI bulb E14 WS opal 400lm): completed initialization`

then it is added to my devices but i cannot control it

[image]

after a reboot it is not reachable anymore

[image]

please let me know if you need more info

Adorem, 2019-08-27:
IDK if it can help you but here is the log when I try to add the Philips bulb:
`
2019-08-27 19:27:22 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 9, b'']
2019-08-27 19:27:22 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800900169e000000158d0001ed44a8e52b9b7ae232a1149e420f00
2019-08-27 19:27:22 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8009 b'000000158d0001ed44a8e52b9b7ae232a1149e420f' LQI:0
2019-08-27 19:27:22 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 00:15:8d:00:01:ed:44:a8, 58667, 11203515730116255298, 15]
2019-08-27 19:27:22 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x21 b'00008000'
2019-08-27 19:27:22 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x00!\x00\x04\xa5\x00\x00\x80\x00'
2019-08-27 19:27:22 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x10!\x02\x10\x02\x14\xa5\x02\x10\x02\x10\x80\x02\x10'
2019-08-27 19:27:22 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000005a40000002100
2019-08-27 19:27:22 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00000021' LQI:0
2019-08-27 19:27:22 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 33, b'']
2019-08-27 19:28:54 DEBUG (MainThread) [zigpy_zigate.zigbee.application] Broadcast not implemented.
2019-08-27 19:28:54 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x49 b'fffc3c00'
2019-08-27 19:28:54 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x00I\x00\x04r\xff\xfc<\x00'
2019-08-27 19:28:54 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x10I\x02\x10\x02\x14r\xff\xfc<\x02\x10'
2019-08-27 19:28:54 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800000057e00b2004900
2019-08-27 19:28:54 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00b20049' LQI:0
2019-08-27 19:28:54 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 178, 73, b'']
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 004d000cfbcc5d00178801030c0e3a8e00
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.api] data received 0x4d b'cc5d00178801030c0e3a8e' LQI:0
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [52317, 00:17:88:01:03:0c:0e:3a, 142]
2019-08-27 19:29:01 INFO (MainThread) [zigpy.application] Device 0xcc5d (00:17:88:01:03:0c:0e:3a) joined the network
2019-08-27 19:29:01 INFO (MainThread) [zigpy.device] [0xcc5d] Requesting 'Node Descriptor'
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0xcc5d, 0, 0x0002, 0, 0, 2, b'\x02]\xcc', True, 10)
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'02cc5d000000020000000003025dcc'
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x0f;\x02\xcc]\x00\x00\x00\x02\x00\x00\x00\x00\x03\x02]\xcc'
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x02\x1f;\x02\x12\xcc]\x02\x10\x02\x10\x02\x10\x02\x12\x02\x10\x02\x10\x02\x10\x02\x10\x02\x13\x02\x12]\xcc'
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001a3b0000000013000002cc5d02fffd005dcc3a0e0c03018817008e99
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'0000000013000002cc5d02fffd005dcc3a0e0c03018817008e' LQI:153
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 19, 0, 0, <Address address_mode=ADDRESS_MODE.NWK address=52317>, <Address address_mode=ADDRESS_MODE.NWK address=65533>, b'\x00]\xcc:\x0e\x0c\x03\x01\x88\x17\x00\x8e']
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy.zdo] [0xcc5d:zdo] ZDO request 0x0013: [0xcc5d, 00:17:88:01:03:0c:0e:3a, 142]
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 870100037a0000ff
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8701 b'0000' LQI:255
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0]
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000005d30063053000
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00630530' LQI:0
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 99, 1328, b'']
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001f180000008002000002cc5d02000002005dcc01408e0b10472d0000002d000096
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'0000008002000002cc5d02000002005dcc01408e0b10472d0000002d0000' LQI:150
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 32770, 0, 0, <Address address_mode=ADDRESS_MODE.NWK address=52317>, <Address address_mode=ADDRESS_MODE.NWK address=0>, b'\x02\x00]\xcc\x01@\x8e\x0b\x10G-\x00\x00\x00-\x00\x00']
2019-08-27 19:29:01 INFO (MainThread) [zigpy.device] [0xcc5d] Node Descriptor: <Optional byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4107 maximum_buffer_size=71 maximum_incoming_transfer_size=45 server_mask=0 maximum_outgoing_transfer_size=45 descriptor_capability_field=0>
2019-08-27 19:29:01 INFO (MainThread) [zigpy.device] [0xcc5d] Discovering endpoints
2019-08-27 19:29:01 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0xcc5d, 0, 0x0005, 0, 0, 3, b'\x03]\xcc', True, 10)
[... cut ...]
2019-08-27 19:29:14 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xcc5d:11:0x0000]: initializing channel: from_cache: False
2019-08-27 19:29:14 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xcc5d:11:0x0000]: channel: 'async_initialize' stage succeeded
2019-08-27 19:29:14 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xcc5d](unk_model): power source: Mains
2019-08-27 19:29:14 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xcc5d](unk_model): completed initialization
`

doudz, 2019-08-30:
Some things are broken... The fact is I've used HA 0.97 and a pre-release 3.a1 Zigate firmware and I don't notice some changes

Adorem, 2019-08-30:
Ah bad luck :'(
You think the changes causing this are from zigate 3.1a or from HA 0.98?

doudz, 2019-08-31:
Got the same hang problem trying to pair a xiaomi door sensor

```
2019-08-31 08:05:06 WARNING (MainThread) [zigpy.endpoint] [0x61b0:1] Failed ZDO request during endpoint initialization: Message send failure 166
2019-08-31 08:05:06 INFO (MainThread) [zigpy.application] Removing device 0x61b0 (00:15:8d:00:01:de:74:3c)
2019-08-31 08:05:06 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x61b0, 0, 0x0034, 0, 0, 21, b'\x15<t\xde\x01\x00\x8d\x15\x00\x02', True, 10)
2019-08-31 08:05:06 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'0261b000000034000000000a153c74de01008d150002'
2019-08-31 08:05:06 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x16\xd6\x02a\xb0\x00\x00\x004\x00\x00\x00\x00\n\x15<t\xde\x01\x00\x8d\x15\x00\x02'
2019-08-31 08:05:06 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x16\xd6\x02\x12a\xb0\x02\x10\x02\x10\x02\x104\x02\x10\x02\x10\x02\x10\x02\x10\x02\x1a\x15<t\xde\x02\x11\x02\x10\x8d\x15\x02\x10\x02\x12'
2019-08-31 08:05:06 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8000000516a600053000
2019-08-31 08:05:06 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'a6000530' LQI:0
2019-08-31 08:05:06 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [166, 0, 1328, b'']
2019-08-31 08:05:06 DEBUG (MainThread) [zigpy.application] Sending 'zdo_leave_req' failed: Message send failure 166
```
Looks like zdo_leave_req fails and cause hang, don't if it's zigate hang or zha

biau, 2019-08-31:
Don't know if this could help but I've tried to pair a xiaomi cube and it was not working at all:
https://termbin.com/21vob
I've revert to zigate component but at first nothing works with a lot of error like 

> 2019-08-31 11:38:09 ERROR (ZiGate-Decode data) [zigate] Command 0x0092 failed Failed (ZigBee event codes) 14 : b''

But after a zigate.reset the problem was gone. 😅

Adorem, 2019-09-02:
And then after that you try to pair your xiaomi cube and it worked?

biau, 2019-09-02:
No I did not retry I do not have a zigate for testing and I don't want to mess with my hass instance.

doudz, 2019-09-11:
Many things have been fixed, please wait until new HA release with zigpy-zigate 0.3.0

Adorem, 2019-09-14:
I copied all the components/zha from the dev branch to custom_components. Then yes now manifest.json refer to 0.3.1. But I'm not certain that 0.3.1 is used. There is way to check? Maybe with some logs?
I ask this because I still have the same problems (unk_manufacturer unk_model).

doudz, 2019-09-14:
Could you provide debug log at pairing time ?
unk_manufacturer means unknow manufacturer, what kind of device are you trying to use ?

Adorem, 2019-10-06:
Bon je viens de faire le test avec HA 0.100.0b1 qui inclus zigpy-zigate 0.4.0. J'essaye à chaque fois de repartir de 0 avec reset complet de la zigate et de la configuration d'HA.
Peut-être que le problème vient plus de zha que de zigpy-zigate ?
J'essaye pour le moment juste d'ajouter une ampoule philips hue.
Voici le log:
2019-10-06 11:12:00 DEBUG (MainThread) [zigpy_zigate.zigbee.application] Broadcast not implemented.
2019-10-06 11:12:00 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x49 b'fffc3c00'
2019-10-06 11:12:00 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x00I\x00\x04r\xff\xfc<\x00'
2019-10-06 11:12:00 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x10I\x02\x10\x02\x14r\xff\xfc<\x02\x10'
2019-10-06 11:12:00 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000005ed0021004900
2019-10-06 11:12:00 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00210049' LQI:0
2019-10-06 11:12:00 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 33, 73, b'']
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 004d000c671a1700178801030c0e3a8e00
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.api] data received 0x4d b'1a1700178801030c0e3a8e' LQI:0
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0x1a17, 00:17:88:01:03:0c:0e:3a, 142]
2019-10-06 11:12:07 INFO (MainThread) [zigpy.application] Device 0x1a17 (00:17:88:01:03:0c:0e:3a) joined the network
2019-10-06 11:12:07 INFO (MainThread) [zigpy.device] [0x1a17] Requesting 'Node Descriptor'
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy.device] [0x1a17] Extending timeout for 0x02 request
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x1a17, 0, 0x0002, 0, 0, 2, b'\x02\x17\x1a', True, False)
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'021a1700000002000000000302171a'
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x0f;\x02\x1a\x17\x00\x00\x00\x02\x00\x00\x00\x00\x03\x02\x17\x1a'
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x02\x1f;\x02\x12\x1a\x17\x02\x10\x02\x10\x02\x10\x02\x12\x02\x10\x02\x10\x02\x10\x02\x10\x02\x13\x02\x12\x17\x1a'
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001adc00000000130000021a1702fffd00171a3a0e0c03018817008e7e
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00000000130000021a1702fffd00171a3a0e0c03018817008e' LQI:126
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 19, 0, 0, <Address address_mode=ADDRESS_MODE.NWK address=0x1a17>, <Address address_mode=ADDRESS_MODE.NWK address=0xfffd>, b'\x00\x17\x1a:\x0e\x0c\x03\x01\x88\x17\x00\x8e']
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy.zdo] [0x1a17:zdo] ZDO request 0x0013: [0x1a17, 00:17:88:01:03:0c:0e:3a, 142]
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 870100037a0000ff
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8701 b'0000' LQI:255
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0]
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800000053e008e053000
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'008e0530' LQI:0
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 142, 1328, b'']
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001ff000000080020000021a170200000200171a01408e0b10472d0000002d00007e
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00000080020000021a170200000200171a01408e0b10472d0000002d0000' LQI:126
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 32770, 0, 0, <Address address_mode=ADDRESS_MODE.NWK address=0x1a17>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x02\x00\x17\x1a\x01@\x8e\x0b\x10G-\x00\x00\x00-\x00\x00']
2019-10-06 11:12:07 INFO (MainThread) [zigpy.device] [0x1a17] Node Descriptor: <Optional byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4107 maximum_buffer_size=71 maximum_incoming_transfer_size=45 server_mask=0 maximum_outgoing_transfer_size=45 descriptor_capability_field=0>
2019-10-06 11:12:07 INFO (MainThread) [zigpy.device] [0x1a17] Discovering endpoints
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x1a17, 0, 0x0005, 0, 0, 4, b'\x04\x17\x1a', True, False)
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'021a1700000005000000000304171a'
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x0f:\x02\x1a\x17\x00\x00\x00\x05\x00\x00\x00\x00\x03\x04\x17\x1a'
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x02\x1f:\x02\x12\x1a\x17\x02\x10\x02\x10\x02\x10\x02\x15\x02\x10\x02\x10\x02\x10\x02\x10\x02\x13\x02\x14\x17\x1a'
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800000053f008f053000
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'008f0530' LQI:0
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 143, 1328, b'']
[... cut ...]
2019-10-06 11:14:11 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8024 b'01000000158d0001ed44a80f' LQI:0
2019-10-06 11:14:11 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [1, 0x0000, 00:15:8d:00:01:ed:44:a8, 15]
2019-10-06 11:14:11 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80060002850100
2019-10-06 11:14:11 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8006 b'01' LQI:0
2019-10-06 11:14:12 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8006

doudz, 2019-10-07:
Le problème commence ici
```
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x1a17, 260, 0, 11, 11, 8, b'\x00\x08\x00\x04\x00\x05\x00', True, False)
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'021a170b0b0000010400000700080004000500'
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x13"\x02\x1a\x17\x0b\x0b\x00\x00\x01\x04\x00\x00\x07\x00\x08\x00\x04\x00\x05\x00'
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x13"\x02\x12\x1a\x17\x02\x1b\x02\x1b\x02\x10\x02\x10\x02\x11\x02\x14\x02\x10\x02\x10\x02\x17\x02\x10\x02\x18\x02\x10\x02\x14\x02\x10\x02\x15\x02\x10'
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8000000513a300053000
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'a3000530' LQI:0
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [163, 0, 1328, b'']
2019-10-06 11:12:07 DEBUG (MainThread) [zigpy.device] [0x1a17] Delivery error for seq # 0x0a, on endpoint id 11 cluster 0x0000: Message send failure %s
```
On reçoit le statut 0xa3 soit ZPS_APL_APS_E_ILLEGAL_REQUEST
A partir de là ça on a soit le statut 0xa3 soit 0xa6  ZPS_APL_APS_E_INVALID_PARAMETER 

Je ne sais pas vraiment ce que tente de faire zha à ce moment là.
Il faudrait je pense contacter les personnes en charge de zha, il y aura peut-être une explication.
Peut-être que cet équipement n'est pas bien géré dans zha

@Adminiuga Could you help us with the error above ?
The error happens right after `2019-10-06 11:12:07 INFO (MainThread) [zigpy.endpoint] [0x1a17:11] Discovered endpoint information: <SimpleDescriptor endpoint=11 profile=49246 device_type=256 device_version=2 input_clusters=[0, 3, 4, 5, 6, 8, 4096] output_clusters=[25]>`

Adminiuga, 2019-10-07:
Hrm, looks like it is trying to read manufacturer and model attributes of
the basic cluster, but I cannt tell why it doesn't like it. The ZCL payload
looks fine to me. It's a zcl global command I'd 0x00 - read attributes, and
two attributes 0x0004 and 0x0005

Adminiuga, 2019-10-08:
@doudz out of the box, what endpoints are configured on zigate and does zigate allows sending requests using an arbitrary source endpoint?
I'm thinking Zigate could be similar `zigpy-deconz` in the sense that it allows originating requests only from the configured endpoints on coordinator. In case of `zigpy-deconz` it would be endpoint id 1. If you send from any other endpoint, Conbee would produce a similar error.

doudz, 2019-10-08:
Hum yes ZiGate only have endpoint 1 as far as I know so it's probably the problem
on `zigpy-deconz` , is it fixed here ? https://github.com/zigpy/zigpy-deconz/blob/master/zigpy_deconz/zigbee/application.py#L107

Adminiuga, 2019-10-08:
Yes. Essentially it is using either `0` or `1` as source endpoint in `zigpy-deconz`. Maybe a `1 if dst_ep else 0` would be more readable as `dst_ep == 0 ` for ZDO requests.

doudz, 2019-10-08:
ok, thanks, I'll try that

doudz, 2019-10-08:
@Adorem J'ai apporté la correction proposée, pourrais-tu faire l'essai ?
Il faudrait modifier manuellement le fichier application.py pour ajouter la ligne suivante
https://github.com/doudz/zigpy-zigate/blob/master/zigpy_zigate/zigbee/application.py#L126

Adorem, 2019-10-08:
Yeah thanks all for your time. Seems we get good progress:

[image]

I'll continue my tests but this patch sound good :)

doudz, 2019-10-09:
Good ! I'll keep this issue open for few days

Adorem, 2019-10-09:
Yes please keep open. Yeasterday I try to pair the next device (Philips hue remote) and it seems not to works correctly too. I'll provide you logs asap.

Adorem, 2019-10-11:
I try now to pair the philips remote and I get this:

[image]

The device is detected but do not works for the button an the battery level for example.

```
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Connection made
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy.appdb] Loading application state from /config/zigbee.db
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy.ota] Initialize OTA providers
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x2 b'01'
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x00\x02\x00\x01\x02\x01'
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x10\x02\x12\x02\x10\x02\x11\x02\x12\x02\x11'
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy.ota.provider] OTA image directory '/config/zigpy_ota/' does not exist
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000005870000000200
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00000002' LQI:0
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 2, b'']
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x10 b''
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x00\x10\x00\x00\x10'
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x10\x10\x02\x10\x02\x10\x10'
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000005950000001000
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00000010' LQI:0
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 16, b'']
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 801000058f0003031a00
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8010 b'0003031a' LQI:0
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [3, 794]
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x21 b'00008000'
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x00!\x00\x04\xa5\x00\x00\x80\x00'
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x10!\x02\x10\x02\x14\xa5\x02\x10\x02\x10\x80\x02\x10'
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000005a40000002100
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00000021' LQI:0
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 33, b'']
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x24 b''
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x00$\x00\x00$'
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x10$\x02\x10\x02\x10$'
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8024000d3a04000000158d0001ed44a80f00
2019-10-11 22:34:30 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8024 b'04000000158d0001ed44a80f' LQI:0
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [4, 0x0000, 00:15:8d:00:01:ed:44:a8, 15]
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000005a10000002400
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00000024' LQI:0
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 36, b'']
2019-10-11 22:34:31 INFO (MainThread) [zigpy_zigate.zigbee.application] Network started 0x0000 00:15:8d:00:01:ed:44:a8
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x9 b''
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x00\t\x00\x00\t'
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x10\x02\x19\x02\x10\x02\x10\x02\x19'
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800000058c0000000900
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00000009' LQI:0
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 9, b'']
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80090016f7000000158d0001ed44a8be0067c2d189cd2db1ed0f00
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8009 b'000000158d0001ed44a8be0067c2d189cd2db1ed0f' LQI:0
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0x0000, 00:15:8d:00:01:ed:44:a8, 48640, 7476768721171493357, 15]
2019-10-11 22:34:31 INFO (MainThread) [homeassistant.setup] Setting up zha
2019-10-11 22:34:31 INFO (MainThread) [homeassistant.setup] Setup of domain zha took 0.0 seconds.
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.uart] Connection made
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy.appdb] Loading application state from /config/zigbee.db
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x2 b'01'
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x00\x02\x00\x01\x02\x01'
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x10\x02\x12\x02\x10\x02\x11\x02\x12\x02\x11'
2019-10-11 22:34:31 DEBUG (MainThread) [zigpy.ota] Initialize OTA providers
[... cut ...]
2019-10-11 22:36:28 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x4b62](RWL021): completed initialization
2019-10-11 22:36:54 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.philips_rwl021_01881700_2_1, old_state=<state sensor.philips_rwl021_01881700_2_1=unavailable; unit_of_measurement=%, friendly_name=Philips RWL021, device_class=battery @ 2019-10-11T22:36:25.461181+02:00>, new_state=<state sensor.philips_rwl021_01881700_2_1=unknown; unit_of_measurement=%, friendly_name=Philips RWL021, device_class=battery @ 2019-10-11T22:36:54.013233+02:00>>
```

Adminiuga, 2019-10-11:
Philips is picky about binding clusters and configuring attribute reporting. Do you get anything when you press buttons?

Adorem, 2019-10-11:
Nope nothing in the logs when I push the buttons.

Adminiuga, 2019-10-12:
are you using dev zigpy branch by any chance?

Adorem, 2019-10-12:
Yes I can try. Want me to test something?
I'm using hass.io. Is it enough to replace the files in the directory /usr/local/lib/python3.7/site-packages/zigpy of my homeassistant container? Need to reboot after changing that? No risk to be overwritten after reboot for example?

Adminiuga, 2019-10-13:
Nah, don't use zigpy dev yet. I think somewhere the "byte order" is messed up.

@doudz 
Device IEEE address is `00:17:88:01:02:d5:b9:8c` and Zigate's IEEE is `00:15:8d:00:01:ed:44:a8`
But in ZDO Bind request, the byte order is not "little endian" 
```
2019-10-11 22:35:57 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x4b62, 0, 0x0021, 0, 0, 30, b'\x1e\x00\x17\x88\x01\x02\xd5\xb9\x8c\x01\x06\x00\x03\x00\x15\x8d\x00\x01\xedD\xa8\x01', True, False)
```

the payload should be `b'\x1e\x8c\xb9\xd5\x02\x01\x88\x17\x00\x01\x06\x00\x03\xa8D\xed\x01\x00\x8d\x15\x00\x01'` I think, which make me wonder how it was working before. Unless devices were actually ignoring the payload and binding just to request originator.

Can you get a  device joining log for a working device (device which successfuly sends attribute updates) as long as the device is not a xiaomi. Xiaomi don't really care about binding and attribe report configuration.

doudz, 2019-10-13:
Could it be my custom EUI64 class ? it reverse byte order in `__repr__`, but since it's only in `__repr__` I supposed it won't affect zigpy but I could be wrong

Adminiuga, 2019-10-14:
ah, yep. It seems to me zigate is using "big-endian" and reversing the `__repr__` make it appears it was processed correctly, but the byte order is going to be messed up between the radio and `zigpy`. Now, I actually was planning to refactor zigpy EUI64 in https://github.com/zigpy/zigpy/pull/240 mandating that between the radio and zigpy, the LSB of IEEE being in the lowest list index and MSB of IEEE being in the highest list index.  Then you could do `zigpy.types.EUI64(zigpy_zigate_EUI64_instance)` and zigpy or radio will serialize data correctly. This would also allow you to get rid off https://github.com/doudz/zigpy-zigate/blob/81d44c4bee89722786982183d054cf123a868e55/zigpy_zigate/types.py#L8-L21
Give me a day or two to publish new zigpy version cause I really think that's the way to go.

doudz, 2019-10-14:
Yes Zigate is using big-endian.
I'll wait for your change. 
Could it be a problem for NWK too ?

Adminiuga, 2019-10-16:
@doudz 
Hrm, so NWK should be OK because of https://github.com/zigpy/zigpy/blob/166b473240b8ed20fa3c10070aa8baa0347e7a3d/zigpy/device.py#L38

Can you review and test #8 ? I think it should address this issue

doudz, 2019-10-20:
@Adminiuga v0.5.0 is released with your PR
Should be ok now

Adorem, 2019-10-20:
Thanks for this new version and the progress. It's still not working. If needed I will provide you full logs but maybe for the moment this one is enough:
```
2019-10-20 20:36:12 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 856, in _handle_cluster_request
    await self._handle_query_next_image(*args, tsn=tsn)
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 925, in _handle_query_next_image
    foundation.Status.NO_IMAGE_AVAILABLE, tsn=tsn
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 159, in request
    with self._pending.new(sequence) as req:
  File "/usr/local/lib/python3.7/site-packages/zigpy/util.py", line 236, in new
    return Request(self, sequence)
  File "/usr/local/lib/python3.7/site-packages/zigpy/util.py", line 205, in __init__
    assert sequence not in pending
AssertionError
```

Adminiuga, 2019-10-20:
In zha config panel pick the remote, press reconfigure button and provide full logs.

The error above is for OTA requests and it could happen if the reply to the 1st request is not acknowledged

Adminiuga, 2019-10-20:
Wait till https://github.com/home-assistant/home-assistant/pull/27972 gets merged into dev, then either install dev version of ha or copy zha folder from the dev tree.
After that 
In zha config panel pick the remote, press reconfigure button and provide full logs.

The error above is for OTA requests and it could happen if the reply to the 1st request is not acknowledged

Adminiuga, 2019-10-21:
@Adorem the https://github.com/home-assistant/home-assistant/pull/27972 got merged. 
Can you try:
1. stop HA container (otherwise it tends to install old deps),
2. copy `homeassistant/components/zha` folder from dev tree into your container where the HA sources are located
3. enable debug loggin
4. start HA and try to reconfigure the HUE remote. 

I'm interested to see ZCL payload of the binding requests.

Adorem, 2019-10-21:
Thanks. Here it is:
```
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy.zcl] [0x2ab1:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=19 command_id=Command.Read_Attributes_rsp>
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800200289900010400000101022ab102000018130104000042075068696c697073050000420652574c3032318a
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00010400000101022ab102000018130104000042075068696c697073050000420652574c303231' LQI:138
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 0, 1, 1, <Address address_mode=ADDRESS_MODE.NWK address=0x2ab1>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18\x13\x01\x04\x00\x00B\x07Philips\x05\x00\x00B\x06RWL021']
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy.zcl] [0x2ab1:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=19 command_id=Command.Read_Attributes_rsp>
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy.device] [0x2ab1] Invalid state on future for 0x13 seq -- probably duplicate response
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800200289900010400000101022ab102000018130104000042075068696c697073050000420652574c3032318a
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00010400000101022ab102000018130104000042075068696c697073050000420652574c303231' LQI:138
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 0, 1, 1, <Address address_mode=ADDRESS_MODE.NWK address=0x2ab1>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18\x13\x01\x04\x00\x00B\x07Philips\x05\x00\x00B\x06RWL021']
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy.zcl] [0x2ab1:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=19 command_id=Command.Read_Attributes_rsp>
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy.device] [0x2ab1] Invalid state on future for 0x13 seq -- probably duplicate response
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy.endpoint] [0x2ab1:1] Manufacturer: Philips
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy.endpoint] [0x2ab1:1] Model: RWL021
2019-10-21 20:32:54 INFO (MainThread) [zigpy.endpoint] [0x2ab1:2] Discovering endpoint information
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy.device] [0x2ab1] Extending timeout for 0x15 request
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x2ab1, 0, 0x0004, 0, 0, 21, b'\x15\xb1*\x02', True, False)
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'022ab100000004000000000415b12a02'
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x100\x02*\xb1\x00\x00\x00\x04\x00\x00\x00\x00\x04\x15\xb1*\x02'
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x100\x02\x12*\xb1\x02\x10\x02\x10\x02\x10\x02\x14\x02\x10\x02\x10\x02\x10\x02\x10\x02\x14\x15\xb1*\x02\x12'
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000005f40044053000
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00440530' LQI:0
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 68, 1328, b'']
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800200274000000080040000022ab10200001500b12a140204010c0000050000010003000f0000fc01190087
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00000080040000022ab10200001500b12a140204010c0000050000010003000f0000fc011900' LQI:135
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 32772, 0, 0, <Address address_mode=ADDRESS_MODE.NWK address=0x2ab1>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x15\x00\xb1*\x14\x02\x04\x01\x0c\x00\x00\x05\x00\x00\x01\x00\x03\x00\x0f\x00\x00\xfc\x01\x19\x00']
2019-10-21 20:32:54 INFO (MainThread) [zigpy.endpoint] [0x2ab1:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=260 device_type=12 device_version=0 input_clusters=[0, 1, 3, 15, 64512] output_clusters=[25]>
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for Philips RWL021 (00:17:88:01:03:e7:2a:68)
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zigpy.quirks.smartthings.SmartthingsMultiPurposeSensor'>
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2}
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {11, 13} {1, 2}
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
2019-10-21 20:32:54 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2}
[... cut ...]
2019-10-21 20:33:29 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
2019-10-21 20:33:29 DEBUG (MainThread) [zigpy.device] [0x2ab1] Delivery error for seq # 0x2b, on endpoint id 1 cluster 0x0000: ZiGate doesn't answer to command
2019-10-21 20:33:29 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
2019-10-21 20:33:29 DEBUG (MainThread) [zigpy.device] [0x2ab1] Delivery error for seq # 0x2d, on endpoint id 2 cluster 0x0000: ZiGate doesn't answer to command
```

Adorem, 2019-10-22:
@Adminiuga are these logs correct/what you expect. Did you want me to redo something?

Adminiuga, 2019-10-22:
But if I interpret the log correctly, the bind command for cluster 6 was successful:
```
2019-10-21 20:32:57 DEBUG (MainThread) [zigpy.device] [0x2ab1] Extending timeout for 0x25 request
2019-10-21 20:32:57 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x2ab1, 0, 0x0021, 0, 0, 37, b'%h*\xe7\x03\x01\x88\x17\x00\x01\x06\x00\x03\xa8D\xed\x01\x00\x8d\x15\x00\x01', True, False)
2019-10-21 20:32:57 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'022ab100000021000000001625682ae7030188170001060003a844ed01008d150001'
2019-10-21 20:32:57 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00"9\x02*\xb1\x00\x00\x00!\x00\x00\x00\x00\x16%h*\xe7\x03\x01\x88\x17\x00\x01\x06\x00\x03\xa8D\xed\x01\x00\x8d\x15\x00\x01'
2019-10-21 20:32:57 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10"9\x02\x12*\xb1\x02\x10\x02\x10\x02\x10!\x02\x10\x02\x10\x02\x10\x02\x10\x16%h*\xe7\x02\x13\x02\x11\x88\x17\x02\x10\x02\x11\x02\x16\x02\x10\x02\x13\xa8D\xed\x02\x11\x02\x10\x8d\x15\x02\x10\x02\x11'

...

2019-10-21 20:32:58 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800200100700000080210000022ab102000025008a
2019-10-21 20:32:58 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00000080210000022ab10200002500' LQI:138
2019-10-21 20:32:58 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 32801, 0, 0, <Address address_mode=ADDRESS_MODE.NWK address=0x2ab1>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'%\x00']
```

So if the last line is the response to the bind req, then the status of that request was SUCCESS.

Adorem, 2019-10-22:
Here is the full log when I add the remote:
```
2019-10-23 00:41:49 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 870100037a0000ff
2019-10-23 00:41:49 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8701 b'0000' LQI:255
2019-10-23 00:41:49 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0]
2019-10-23 00:41:49 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000005bb000b053000
2019-10-23 00:41:49 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'000b0530' LQI:0
2019-10-23 00:41:49 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 11, 1328, b'']
2019-10-23 00:41:56 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 004d000ce57f630017880103e72a688000
2019-10-23 00:41:56 DEBUG (MainThread) [zigpy_zigate.api] data received 0x4d b'7f630017880103e72a6880' LQI:0
2019-10-23 00:41:56 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0x7f63, 00:17:88:01:03:e7:2a:68, 128]
2019-10-23 00:41:56 INFO (MainThread) [zigpy.application] Device 0x7f63 (00:17:88:01:03:e7:2a:68) joined the network
2019-10-23 00:41:56 INFO (MainThread) [zigpy.device] [0x7f63] Requesting 'Node Descriptor'
2019-10-23 00:41:56 DEBUG (MainThread) [zigpy.device] [0x7f63] Extending timeout for 0x16 request
2019-10-23 00:41:56 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x7f63, 0, 0x0002, 0, 0, 22, b'\x16c\x7f', True, False)
2019-10-23 00:41:56 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'027f6300000002000000000316637f'
2019-10-23 00:41:56 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x0f/\x02\x7fc\x00\x00\x00\x02\x00\x00\x00\x00\x03\x16c\x7f'
2019-10-23 00:41:56 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x02\x1f/\x02\x12\x7fc\x02\x10\x02\x10\x02\x10\x02\x12\x02\x10\x02\x10\x02\x10\x02\x10\x02\x13\x16c\x7f'
2019-10-23 00:41:56 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001aa000000000130000027f6302fffd07637f682ae703018817008096
2019-10-23 00:41:56 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00000000130000027f6302fffd07637f682ae7030188170080' LQI:150
2019-10-23 00:41:56 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 19, 0, 0, <Address address_mode=ADDRESS_MODE.NWK address=0x7f63>, <Address address_mode=ADDRESS_MODE.NWK address=0xfffd>, b'\x07c\x7fh*\xe7\x03\x01\x88\x17\x00\x80']
2019-10-23 00:41:56 DEBUG (MainThread) [zigpy.zdo] [0x7f63:zdo] ZDO request 0x0013: [0x7f63, 00:17:88:01:03:e7:2a:68, 128]
2019-10-23 00:41:56 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000005bc000c053000
2019-10-23 00:41:56 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'000c0530' LQI:0
2019-10-23 00:41:56 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 12, 1328, b'']
2019-10-23 00:41:57 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 004d000ce57f630017880103e72a688000
2019-10-23 00:41:57 DEBUG (MainThread) [zigpy_zigate.api] data received 0x4d b'7f630017880103e72a6880' LQI:0
2019-10-23 00:41:57 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0x7f63, 00:17:88:01:03:e7:2a:68, 128]
2019-10-23 00:41:57 INFO (MainThread) [zigpy.application] Device 0x7f63 (00:17:88:01:03:e7:2a:68) joined the network
2019-10-23 00:41:57 DEBUG (MainThread) [zigpy.application] Skip initialization for existing device 00:17:88:01:03:e7:2a:68
2019-10-23 00:41:57 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001aab00000000130000027f6302fffd09637f682ae703018817008093
2019-10-23 00:41:57 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00000000130000027f6302fffd09637f682ae7030188170080' LQI:147
2019-10-23 00:41:57 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 19, 0, 0, <Address address_mode=ADDRESS_MODE.NWK address=0x7f63>, <Address address_mode=ADDRESS_MODE.NWK address=0xfffd>, b'\tc\x7fh*\xe7\x03\x01\x88\x17\x00\x80']
2019-10-23 00:41:57 DEBUG (MainThread) [zigpy.zdo] [0x7f63:zdo] ZDO request 0x0013: [0x7f63, 00:17:88:01:03:e7:2a:68, 128]
2019-10-23 00:42:07 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.signal_wifi_led_sous_sol, old_state=<state sensor.signal_wifi_led_sous_sol=-95; unit_of_measurement=dB, friendly_name=Signal WiFi LED Sous Sol, icon=mdi:wifi @ 2019-10-23T00:41:07.476415+02:00>, new_state=<state sensor.signal_wifi_led_sous_sol=-94; unit_of_measurement=dB, friendly_name=Signal WiFi LED Sous Sol, icon=mdi:wifi @ 2019-10-23T00:42:07.468411+02:00>>
2019-10-23 00:42:18 DEBUG (MainThread) [zigpy.device] [0xfffe] Extending timeout for 0x18 request
2019-10-23 00:42:18 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0xfffe, 0, 0x0002, 0, 0, 24, b'\x18\xfe\xff', True, False)
2019-10-23 00:42:18 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'02fffe00000002000000000318feff'
2019-10-23 00:42:18 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x0f!\x02\xff\xfe\x00\x00\x00\x02\x00\x00\x00\x00\x03\x18\xfe\xff'
2019-10-23 00:42:18 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x02\x1f!\x02\x12\xff\xfe\x02\x10\x02\x10\x02\x10\x02\x12\x02\x10\x02\x10\x02\x10\x02\x10\x02\x13\x18\xfe\xff'
2019-10-23 00:42:18 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000005be000e053000
[... cut ...]
  File "/config/custom_components/zha/api.py", line 548, in async_binding_operation
    await asyncio.gather(*bind_tasks)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 184, in request
    self.nwk, dst_ep, cluster
zigpy.exceptions.DeliveryError: [0x7f63:0:0x0021]: Message send failure
2019-10-23 00:46:46 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
2019-10-23 00:46:46 DEBUG (MainThread) [zigpy.device] [0x7f63] Delivery error for seq # 0x4a, on endpoint id 0 cluster 0x0021: ZiGate doesn't answer to command

```

Adminiuga, 2019-10-23:
per logs, cluster `0x0006` was successfully bound this time
```
2019-10-23 00:42:59 DEBUG (MainThread) [custom_components.zha.core.channels] [0x7f63:1:0x0006]: initializing channel: from_cache: False
2019-10-23 00:42:59 DEBUG (MainThread) [custom_components.zha.core.channels] [0x7f63:1:0x0006]: channel: 'async_initialize' stage succeeded
```

1. Do you get anything on the uart when you press buttons on the remote?
2. When trying to bind the remote, since it is battery operated it might be sleeping. So immediatly after you press "bind" in zha config panel, press buttons on the remote to keep it awake and see if the binding to the light succeeds.  

cause when you try to bind remote to a light, apparently remote does not acknowledge reception of the request
```
2019-10-23 00:46:46 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
2019-10-23 00:46:46 DEBUG (MainThread) [zigpy.device] [0x7f63] Delivery error for seq # 0x4a, on endpoint id 0 cluster 0x0021: ZiGate doesn't answer to command
```

Adorem, 2019-10-23:
1. No I get nothing in the logs when I press the remote buttons.
2. I retry this this morning and no change (same logs as seen in my last message).
Is it normal the error we have just before the `No response to command 0x0530`
```
2019-10-23 07:25:30 DEBUG (MainThread) [custom_components.zha.api] processing binding operation for: source_ieee: [00:17:88:01:03:e7:2a:68] target_ieee: [00:17:88:01:03:0c:0e:3a] cluster: 6
2019-10-23 07:25:30 DEBUG (MainThread) [custom_components.zha.api] processing binding operation for: source_ieee: [00:17:88:01:03:e7:2a:68] target_ieee: [00:17:88:01:03:0c:0e:3a] cluster: 8
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy.device] [0x7f63] Extending timeout for 0xfe request
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x7f63, 0, 33, 0, 0, 254, b'\xfeh*\xe7\x03\x01\x88\x17\x00\x01\x06\x00\x03:\x0e\x0c\x03\x01\x88\x17\x00\x0b', True, False)
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'027f63000000210000000016fe682ae70301881700010600033a0e0c03018817000b'
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00"R\x02\x7fc\x00\x00\x00!\x00\x00\x00\x00\x16\xfeh*\xe7\x03\x01\x88\x17\x00\x01\x06\x00\x03:\x0e\x0c\x03\x01\x88\x17\x00\x0b'
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10"R\x02\x12\x7fc\x02\x10\x02\x10\x02\x10!\x02\x10\x02\x10\x02\x10\x02\x10\x16\xfeh*\xe7\x02\x13\x02\x11\x88\x17\x02\x10\x02\x11\x02\x16\x02\x10\x02\x13:\x02\x1e\x02\x1c\x02\x13\x02\x11\x88\x17\x02\x10\x02\x1b'
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy.device] [0x7f63] Extending timeout for 0xff request
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x7f63, 0, 33, 0, 0, 255, b'\xffh*\xe7\x03\x01\x88\x17\x00\x01\x08\x00\x03:\x0e\x0c\x03\x01\x88\x17\x00\x0b', True, False)
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'027f63000000210000000016ff682ae70301881700010800033a0e0c03018817000b'
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00"]\x02\x7fc\x00\x00\x00!\x00\x00\x00\x00\x16\xffh*\xe7\x03\x01\x88\x17\x00\x01\x08\x00\x03:\x0e\x0c\x03\x01\x88\x17\x00\x0b'
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10"]\x02\x12\x7fc\x02\x10\x02\x10\x02\x10!\x02\x10\x02\x10\x02\x10\x02\x10\x16\xffh*\xe7\x02\x13\x02\x11\x88\x17\x02\x10\x02\x11\x02\x18\x02\x10\x02\x13:\x02\x1e\x02\x1c\x02\x13\x02\x11\x88\x17\x02\x10\x02\x1b'
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8000000516a600053000
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'a6000530' LQI:0
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [166, 0, 1328, b'']
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8000000516a600053000
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'a6000530' LQI:0
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [166, 0, 1328, b'']
2019-10-23 07:25:30 DEBUG (MainThread) [zigpy.device] [0x7f63] Delivery error for seq # 0xff, on endpoint id 0 cluster 0x0021: Message send failure 166
2019-10-23 07:25:30 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.139900713236112] Error handling message: Unknown error
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 19, in _handle_async_response
    await func(hass, connection, msg)
  File "/config/custom_components/zha/api.py", line 485, in websocket_bind_devices
    await async_binding_operation(zha_gateway, source_ieee, target_ieee, BIND_REQUEST)
  File "/config/custom_components/zha/api.py", line 548, in async_binding_operation
    await asyncio.gather(*bind_tasks)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 184, in request
    self.nwk, dst_ep, cluster
zigpy.exceptions.DeliveryError: [0x7f63:0:0x0021]: Message send failure
2019-10-23 07:25:33 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
2019-10-23 07:25:33 DEBUG (MainThread) [zigpy.device] [0x7f63] Delivery error for seq # 0xfe, on endpoint id 0 cluster 0x0021: ZiGate doesn't answer to command
```

Adminiuga, 2019-10-24:
Hrm, at this point I"m not sure what to try next. Both requests  req id 0xfe to bind cluster 6 (on off cluster) and req id 0xff to bind cluster 8 (level cluster) failed for some reason. 

I assume you don't have anything to run a zigbee sniffer to capture the traffic between zigate and hue? An ezsp based or a Conbee or CC2531?

Jsl-1, 2019-10-24:
Hi @Adminiuga , Phillips hue remote RWL021 use cluster 0xfc00 on endpoint 1 to send button information here is the detail of the cluster :

https://github.com/doudz/zigate/blob/e320afad5307273871821bdb325e9ba704f5a45d/zigate/clusters.py#L587-L599

Is this what you need here ?

Adorem, 2019-10-24:
@Jsl-1 Yeah thanks for these details!
Then @Adminiuga is this already managed by zha, zigpy and zigpy-zigate? If yes what shall I try then?

Also about this:
> I assume you don't have anything to run a zigbee sniffer to capture the traffic between zigate and hue? An ezsp based or a Conbee or CC2531?

Yes for the moment I have not but I would like to learn more about zigbee and I would like to buy a module for debug for a long time. Which one did you recommand me to buy?

Adminiuga, 2019-10-24:
Hrm, it's a loaded question :) If just for troubleshooting, Zigbee sniffing I probably would get a few CC2531 (you could flash them with router firmware too) and a programmer dongle for CC2531, unless you want to repurpose and Arduino for CC2531 programming.

I'm currently using CC2531 with ZBoss sniffer firmware.

Adorem, 2019-11-08:
I received yesterday my CC2531 stick and I will now try to capture the traffic.

But before that, according to the comment from @Jsl-1 can you please tell me @Adminiuga and @doudz if the fact that the Phillips hue remote RWL021 use cluster 0xfc00 is this already managed by zha, zigpy and zigpy-zigate?

Adminiuga, 2019-11-08:
Can't confirm nor deny. My dimmer doesn't appear to do so and I assume dmulcahey's doesn't do it as well, otherwise we would have a zha quirk for it. In a issue reported by someone else I do see traffic on 0xfc00 cluster. 
In your case, we still should see traffic come ng in and then zigpy complaining about an unknown cluster specific command, but we get nothing

Adminiuga, 2019-11-14:
any luck with the zigbee sniffer?

Adorem, 2019-11-15:
Thanks for asking. For the moment I do not success to sniff the traffic because of lack of time. I hope I will found some time this week-end.

Adorem, 2020-01-05:
@Adminiuga I tried to capture the traffic between zigate and hue without success for the moment.
I follow these instructions here https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html#with-cc2531
I can see some zigbee trafic but not the one for example when from the zigate I change the state of my hue light for exemple.
Did you have any idea of what I'm doing wrong?

Adminiuga, 2020-01-06:
1st double check you are sniffing on the correct channel. I don't know if there's a way to confirm on which channel Zigate formed the network. Try an each channel?

Next, make sure you have `ZigbeeAliance09` (confirm if `b` is upper or lower case) key in Wireshark preferences for Zigbee protocol, then permit joining for new devices and join a new device. In wireshark you should see a few beacon packets and a transport key request. Once you get the transport key, copy its value and put in Zigbee protocol dissector preferences so it could decode ZCL payload.

Adorem, 2020-01-07:
> 1st double check you are sniffing on the correct channel. I don't know if there's a way to confirm on which channel Zigate formed the network. Try an each channel?

Oh yes that's it! I have not enough time yesterday to get ell the traces but I will came back soon.

@doudz my zigate was configured on channel 15. Then, using ZWGUI, I change it to 11 but it seems to go back to channel 15. Is it something configurable?

doudz, 2020-01-07:
@Adorem yes and no...
Yes, currently zigpy set channel to 15 by default but any other channel could be used
No, because zha doesn't provide a way to configure it

zha call `await self.application_controller.startup(auto_form=True)`
https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/zha/core/gateway.py#L124

so https://github.com/doudz/zigpy-zigate/blob/master/zigpy_zigate/zigbee/application.py#L35-L36
```
if auto_form:
    await self.form_network()
```

and so https://github.com/doudz/zigpy-zigate/blob/master/zigpy_zigate/zigbee/application.py#L49
`async def form_network(self, channel=15, pan_id=None, extended_pan_id=None):
`

doudz, 2020-01-07:
@Adorem I suggest you to open an issue on Home Assistant to request the ability to change channel

Adminiuga, 2020-01-07:
Don't hangup on the channel. Is not the issue. When device joins the network it remembers the channel.

[8 more replies]

commit 81d44c4b (2019-10-08): Try to fix #4
closed 2020-10-19 by doudz