zigpy/zigpy-zigate #6: Usable or not?

nerdosity, 2019-09-21:
Hi,

Today since I had some problem with the zigate normal library (after the firmware upgrade every device were still there, but no device worked anymore) I tried the ZHA functions from Home assistant.

I want to clarify, I absolutely do not want to judge, just trying to understand where is the fault. Is your library already stable and usable?
Or maybe the ZHA portion is not still usable in the home assistant core?

I had a LOT of troubles, mainly these:

- pairing ("search for new devices" or the pairing service) worked 1 times on 2, sometimes it just continue to flash for longer time (5/10 minutes)
- two battery powered devices were seen as AC devices
- I had a lot of problems with pairing lamps and outlets, and they worked 2 times on 5
- after every reboot, EVERY device went missing. Everytime. That was the worst problem that let me gave this up.

Latest zigpy zigate lib, latest home assistant (.99)
Do you suggest something? I came back again to using your zigate module, just repairing every device and now it is working quite good apart of a few problems.

Rommie19, 2019-09-22:
i have the same problems, only the problem of "only paring once after a reboot" is solved since the last time
i am using hass.io

doudz, 2019-09-22:
Well, release 0.3.1 solve many bugs but HA 0.99 use release 0.2...

cgradl, 2019-09-23:
I can confirm above. tried it with ZHA.every HA reboot no devices working and device pairing very faulty. 
Initially i thought it was a network and connectivity problem but not even with devices next to the zigate it worked properly. I went back to your custom integration after family complained that they have to use lightswitches again which at least was good test to see the acceptance factor ;-)

doudz, 2019-09-23:
Should be fixed in HA 0.99.3

nerdosity, 2019-09-23:
Is there a quick way to try it without repairing everything? Like a device converter?
I would gladly test it but I have a LOT of devices, and my girlfriend would be angry to have half house not working for hours :D 

[image]

jgirardet, 2019-09-24:
I tested 0.99.3 and dev yesterday (i a virtualenv). My xiaomi aqara door sensor and heiman siren are lost after homeassistant restart

doudz, 2019-09-24:
You can't use dev branch, because it requires zigpy-zigate 0.4.0, which has not been released yet.

You should use 0.99.3 from rc branch (since 0.99.3 has not been released)

doudz, 2019-09-24:
@nerdosity wooo nice network !
I suppose I could make a converter

jgirardet, 2019-09-24:
indeed, it was 0.99.2 from pypi sorry for the mistake.
Is homeassistant dev testable over zigpy-zigate dev now ?

doudz, 2019-09-24:
no, you can use homeassistant dev
you can test using **rc** branch of homeassistant to have 0.99.3

doudz, 2019-09-26:
@nerdosity I made a converter, it's not garanted to work for all devices but it should avoid having to re-pair each device
https://github.com/doudz/zigate/blob/dev/zigate2zigpy.py
To use it you have to put the converter in the same folder as zigate.json and call
`python3 zigate2zigpy.py`
it will create/update the zigbee.db

nerdosity, 2019-09-26:
Perfect, this way I could help more easily on reports, repairing all devices everytime was a bit frustrating

jgirardet, 2019-09-27:
devices stay paired after reboot in RC :-)

doudz, 2019-09-28:
HA 0.99.3 is out

closed 2019-09-28 by doudz

nerdosity, 2019-10-14:
Converter worked (from shell) but devices were not created inside HA. Device list was empty. Any idea?

doudz, 2019-10-14:
> Converter worked (from shell) but devices were not created inside HA. Device list was empty. Any idea?

No time to really check it now, but did you try to restart HA twice ?

nerdosity, 2019-10-16:
Tried again.
After two reboots a lot of devices appeared, but every device was greyed out (not available)

[image]