zigpy/zigpy-zigate #7: Remove device with ZHA

guirem, 2019-10-15:
Hi,

I tried to remove a device via Configuration > ZHA > #pick a device# > Remove Device (or Configuration > ZHA > #pick a device# >  Reconfigure device)

Nothing seems to happen on zigate side. The device is temporary removed from the UI but will reappear if I refresh the page.
 In the logs I get only a notification from zha but nothing from zigate
```
2019-10-15 12:06:50 INFO (MainThread) [homeassistant.components.zha.api] Removing node 00:15:8d:00:01:b9:22:7e
2019-10-15 12:07:22 DEBUG (MainThread) [homeassistant.components.zha.api] Reconfiguring node with ieee_address: 00:0d:6f:ff:fe:a9:17:ce
```

Is it implemented ?

doudz, 2019-10-16:
not yet, this part is commented
https://github.com/doudz/zigpy-zigate/blob/master/zigpy_zigate/zigbee/application.py#L78-L80

But it should be ok in next release

doudz, 2019-10-20:
Fixed in v0.5.0

closed 2019-10-20 by doudz