zigpy/zigpy-zigate #9: Warn device

jgirardet, 2019-10-16:
Hi,
I have a heiman Warn device.
added via zha :

IEEE:    00:0d:6f:00:12:0a:bc:e6
Nwk:    0x55fd
LQI:    156
RSSI:    Unknown
Last Seen:    2019-10-16T20:54:46
Power Source:     Mains

Using the code example "full example data" : nothing happen.

Log :
```
2019-10-16 20:58:03 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.140049627285824] Error handling message: Unknown error
Traceback (most recent call last):
  File "/home/jimmy/rien/hass/.venv/lib/python3.6/site-packages/homeassistant/components/websocket_api/decorators.py", line 19, in _handle_async_response
    await func(hass, connection, msg)
  File "/home/jimmy/rien/hass/.venv/lib/python3.6/site-packages/homeassistant/components/zha/api.py", line 457, in websocket_get_bindable_devices
    for device in zha_gateway.devices.values()
  File "/home/jimmy/rien/hass/.venv/lib/python3.6/site-packages/homeassistant/components/zha/api.py", line 458, in <listcomp>
    if async_is_bindable_target(source_device, device)
  File "/home/jimmy/rien/hass/.venv/lib/python3.6/site-packages/homeassistant/components/zha/core/helpers.py", line 129, in async_is_bindable_target
    source_clusters = source_zha_device.async_get_std_clusters()
AttributeError: 'NoneType' object has no attribute 'async_get_std_clusters'

2019-10-16 20:59:54 ERROR (MainThread) [homeassistant.components.zha.api] Warning IASWD: ieee: [00:0d:6f:00:05:7d:2d:34] could not be found!
2019-10-16 20:59:54 DEBUG (MainThread) [homeassistant.components.zha.api] Warning IASWD: ieee: [00:0d:6f:00:05:7d:2d:34] mode: [1] strobe: [1] level: [2]
```

nerdosity, 2019-11-26:
I have a Heiman device too (SMOK_V16). It disappears after a while.
Why it looks like it's connected to "Mains"? It's a battery powered device (smoke sensor in my case).

Hedda, 2020-09-09:
@jgirardet Did you manage to fix this in latest HA and if so are you OK to close this yourself as resolved or will you keep it open?

jgirardet, 2020-09-09:
I can't test it right now, maybe at the end of the year. Close it if you want, I'll reopen if needed later
thanks

Hedda, 2020-09-30:
@jgirardet best is if you could close this yourself and then also reopen it yourself.

jgirardet, 2020-09-30:
ok I close it now and I come back if needed

closed 2020-09-30 by jgirardet