zigpy/zigpy-zigate #13: ZHA can't find my device (TRADFRI driver for wireless control (30 watt) (ICPSHC24-30EU-IL-1))

nerdosity, 2020-01-15:
Hello, I don't know if this issue (opened on HASS GitHub) is connected with your library or not.
Basically, my installation can't find it while using ZHA. With your custom module it previously worked fine.
See https://github.com/home-assistant/home-assistant/issues/30788

nerdosity, 2020-01-18:
They suggested me to mention it here in zigpy-zigate:
https://github.com/home-assistant/home-assistant/issues/30788#issuecomment-575818449

nerdosity, 2020-02-07:
@Adminiuga suggested me to escalate it here.
https://github.com/home-assistant/home-assistant/issues/30788#issuecomment-583480735

nerdosity, 2020-03-02:
Actually it solved today. Did they upgrade the zigate-zigpy library in HA?

Hedda, 2020-04-15:
No recent bumps of zigpy-zigate in Home Assistant core as far as I have seen, but plenty of other ZHA related changes that have been merged into Home Assistant core in the past months.

Usually, after code changes such as fixes is added to the libraries there is a release of that package and relatively soon after that new package is included as an update in Home Assistant. 

- https://github.com/zigpy/zigpy-zigate/releases

- https://pypi.org/project/zigpy-zigate/#history

You can track merged HA changes in closed pull requests:
https://github.com/home-assistant/core/pulls?q=is%3Aclosed

Tip is to search with keywords plus AND/OR like example "zigpy OR zha"
https://github.com/home-assistant/core/pulls?q=is%3Apr+is%3Aclosed+zigpy+or+zha

Know that zigpy-zigate is also dependent on zigpy and zha-device-handlers in Home Assistant

- https://github.com/zigpy/zigpy

- https://github.com/zigpy/zha-device-handlers

Improvements to zigpy and zha-device-handlers also means indirect improvements for zigate users.

Regardless, if this is solved by you then would simply close this now if I was you.

closed 2020-04-15 by nerdosity

Hedda, 2020-04-16:
@nerdosity FYI, submitted an update to readme in https://github.com/zigpy/zigpy-zigate/pull/29 on how-to test new releases.

## Testing new releases

Testing a new release of the zigpy-zigate library before it is released in Home Assistant.

If you are using Supervised Home Assistant (formerly known as the Hassio/Hass.io distro):
- Add https://github.com/home-assistant/hassio-addons-development as "add-on" repository
- Install "Custom deps deployment" addon
- Update config like: 
  ```
  pypi:
    - zigpy-zigate==0.5.1
  apk: []
  ```
  where 0.5.1 is the new version
- Start the addon

If you are instead using some custom python installation of Home Assistant then do this:
- Activate your python virtual env
- Update package with ``pip``
  ```
  pip install zigpy-zigate==0.5.1