zigpy/zigpy-zigate #17: [SUGGESTION] Migrate zigpy-zigate repository to upstream zigpy organization?

Hedda, 2020-02-16:
@doudz can I suggest that you please consider talking to @Adminiuga & @dmulcahey about the possibility of you joining the zigpy organization @ https://github.com/zigpy/ and maybe also having your zigpy-zigate repository here on GitHub moved to the zigpy organization @ https://github.com/zigpy/ so that the mainstream copy of this repo will stay there as its upstream instead and owned by the upstream zigpy organization? 

That way all developers that are members in the zigpy organization can have write access to one or more projects that it own for collaboration as a community? Hope that you could all come to an agreement that will make everyone happy with continuing with working on the zigpy-zigate repository together without getting offended by any ownership conflicts.

Hedda, 2020-03-20:
@doudz FYI, similar to zigpy-cc which @sanyatuning now moved upstream https://github.com/zigpy/zigpy-cc/issues/13

doudz, 2020-03-23:
Done !

closed 2020-03-23 by Adminiuga