zigpy/zigpy-zigate #18: Error setting up entry /dev/ttyUSB0 for zha

nerdosity, 2020-02-18:
My Zigate stop responding from some hours ago.
I got a zigbee2mqtt stick in the same USB (powered) hub, and it's still kicking.
This is the error trace on a new boot:

Error setting up entry /dev/ttyUSB0 for zha
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigpy_zigate/api.py", line 80, in command
    timeout=COMMAND_TIMEOUT
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 449, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 215, in async_setup
    hass, self
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 103, in async_setup_entry
    await zha_gateway.async_initialize()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 136, in async_initialize
    await self.application_controller.startup(auto_form=True)
  File "/usr/local/lib/python3.7/site-packages/zigpy_zigate/zigbee/application.py", line 29, in startup
    await self._api.set_raw_mode()
  File "/usr/local/lib/python3.7/site-packages/zigpy_zigate/api.py", line 105, in set_raw_mode
    await self.command(0x0002, data)
  File "/usr/local/lib/python3.7/site-packages/zigpy_zigate/api.py", line 84, in command
    raise NoResponseError
zigpy_zigate.api.NoResponseError

doudz, 2020-02-18:
Is it the prolific USB ttl adapter ?

nerdosity, 2020-02-18:
I think so, should be the original one shipped with my Zigate.

doudz, 2020-02-18:
The prolific model is known to have problem, and is now replaced by a "red" adapter, see Zigate.fr

nerdosity, 2020-02-18:
Ok, I will keep you posted. I wrote for a replacement, when it arrives I will update here.

nerdosity, 2020-05-07:
Hi, it arrived to me some days ago. Until today no errors, but it's a little early, sometimes it starts after some days of activity.

Hedda, 2020-09-09:
@nerdosity Did you manage to fix this in latest HA and if so are you OK to close this yourself as resolved or will you keep it open?

nerdosity, 2020-09-09:
> a

Hi, I removed the Zigate since it was not supported enough via zygpy. I am now using the BV 2010/10 with Silicon Labs chipset.

closed 2020-11-04 by doudz