zigpy/zigpy-zigate #19: Cannot create groups

nerdosity, 2020-03-02:
Hi, when I try to create a new group from HASS ZHA group editor I receive this:

```
Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 856, in _handle_cluster_request
    await self._handle_query_next_image(*args, tsn=tsn)
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 925, in _handle_query_next_image
    foundation.Status.NO_IMAGE_AVAILABLE, tsn=tsn
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 162, in request
    with self._pending.new(sequence) as req:
  File "/usr/local/lib/python3.7/site-packages/zigpy/util.py", line 236, in new
    return Request(self, sequence)
  File "/usr/local/lib/python3.7/site-packages/zigpy/util.py", line 205, in __init__
    assert sequence not in pending
AssertionError
```

The loader spinner just keep spinning.

nerdosity, 2020-03-02:
Actually the group was created, but no devices were added.
Then I selected two devices, and only one was added to the group.
Then I selecet every device, added everyone and it just added another device to the group.
Now I can't remove the group or remove device from the group.

nerdosity, 2020-03-02:
I have these errors too:

`Error handling message: Unknown error
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 20, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 347, in websocket_add_group_members
    await zha_group.async_add_members(members)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/group.py", line 58, in async_add_members
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 570, in async_add_to_group
    await self._zigpy_device.add_to_group(group_id)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 140, in add_to_group
    await ep.add_to_group(grp_id, name)
  File "/usr/local/lib/python3.7/site-packages/zigpy/endpoint.py", line 119, in add_to_group
    res = await self.groups.add(grp_id, name)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 187, in request
    self.nwk, dst_ep, cluster
zigpy.exceptions.DeliveryError: [0xcf92:1:0x0004]: Message send failure
Error handling message: Unknown error
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 20, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 347, in websocket_add_group_members
    await zha_group.async_add_members(members)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/group.py", line 58, in async_add_members
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 570, in async_add_to_group
    await self._zigpy_device.add_to_group(group_id)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 140, in add_to_group
    await ep.add_to_group(grp_id, name)
  File "/usr/local/lib/python3.7/site-packages/zigpy/endpoint.py", line 119, in add_to_group
    res = await self.groups.add(grp_id, name)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 187, in request
    self.nwk, dst_ep, cluster
zigpy.exceptions.DeliveryError: [0xcf92:1:0x0004]: Message send failure
Error handling message: Unknown error
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 20, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 324, in websocket_remove_groups
    await zha_gateway.async_remove_zigpy_group(group_ids[0])
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 539, in async_remove_zigpy_group
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 574, in async_remove_from_group
    await self._zigpy_device.remove_from_group(group_id)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 145, in remove_from_group
    await ep.remove_from_group(grp_id)
  File "/usr/local/lib/python3.7/site-packages/zigpy/endpoint.py", line 134, in remove_from_group
    res = await self.groups.remove(grp_id)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 187, in request
    self.nwk, dst_ep, cluster
zigpy.exceptions.DeliveryError: [0x5bb7:1:0x0004]: Message send failure
Error handling message: Unknown error
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 20, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 377, in websocket_remove_group_members
    await zha_group.async_remove_members(members)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/group.py", line 74, in async_remove_members
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 574, in async_remove_from_group
    await self._zigpy_device.remove_from_group(group_id)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 145, in remove_from_group
    await ep.remove_from_group(grp_id)
  File "/usr/local/lib/python3.7/site-packages/zigpy/endpoint.py", line 134, in remove_from_group
    res = await self.groups.remove(grp_id)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 187, in request
    self.nwk, dst_ep, cluster
zigpy.exceptions.DeliveryError: [0x5bb7:1:0x0004]: Message send failure
Error handling message: Unknown error
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 20, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 377, in websocket_remove_group_members
    await zha_group.async_remove_members(members)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/group.py", line 78, in async_remove_members
    ].async_remove_from_group(self.group_id)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 574, in async_remove_from_group
    await self._zigpy_device.remove_from_group(group_id)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 145, in remove_from_group
    await ep.remove_from_group(grp_id)
  File "/usr/local/lib/python3.7/site-packages/zigpy/endpoint.py", line 134, in remove_from_group
    res = await self.groups.remove(grp_id)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 187, in request
    self.nwk, dst_ep, cluster
zigpy.exceptions.DeliveryError: [0x2946:3:0x0004]: Message send failure`

Hedda, 2020-04-15:
Suggest that you try again with the latest Home Assistant (preferably the very latest beta) as support for groups was only recently added to Home Assistant ZHA, like really  just in the past 3-weeks or so, and many group specific issues as well as general ZHA/Zigbee/Zigpy related fixes have been added since then, see example:

- https://github.com/home-assistant/core/pull/33196
- https://github.com/home-assistant/core/pull/33207
- https://github.com/home-assistant/core/pull/33291
- https://github.com/home-assistant/core/pull/33378
- https://github.com/home-assistant/core/pull/31075
- https://github.com/home-assistant/core/pull/33977
- https://github.com/home-assistant/core/pull/33378
- https://github.com/home-assistant/core/pull/33308

nerdosity, 2020-04-18:
```
Error handling message: Unknown error
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 20, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 301, in websocket_add_group
    group = await zha_gateway.async_create_zigpy_group(group_name, members)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 595, in async_create_zigpy_group
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 550, in async_add_to_group
    await self._zigpy_device.add_to_group(group_id)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 140, in add_to_group
    await ep.add_to_group(grp_id, name)
  File "/usr/local/lib/python3.7/site-packages/zigpy/endpoint.py", line 118, in add_to_group
    res = await self.groups.add(grp_id, name)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 187, in request
    self.nwk, dst_ep, cluster
zigpy.exceptions.DeliveryError: [0x5bb7:1:0x0004]: Message send failure
```

From the latest stable. Should I try with the beta?

Hedda, 2020-04-20:
I do not think that there is a new beta yet after 0.108

https://rc.home-assistant.io/latest-release-notes/

There are however bug-releases where I think 0.108.6 is the currently latest

Suggest that you try 0.108.6 and then report the issue to Home Assistant Core and tagging "ZHA"

https://github.com/home-assistant/core/issues

Best would be if you also had other Zigbee adapter (like ex. CC2531) if can replicate same issue.

If you can replicate the same issue with two different adapters then unlikely issue is in zigpy-zigate

If you replicate the same issue on other adapters then can be ZHA in Home Assistant Core or zigpy

PS: I do not own a Zigate adapter myself.

Hedda, 2020-04-20:
@doudz Have you tested the new "groups" functions via Home Assistant's ZHA on a Zigate adapter?

- https://github.com/home-assistant/core/pull/33196
- https://github.com/home-assistant/core/pull/33207
- https://github.com/home-assistant/core/pull/33291
- https://github.com/home-assistant/core/pull/33378
- https://github.com/home-assistant/core/pull/31075
- https://github.com/home-assistant/core/pull/33977
- https://github.com/home-assistant/core/pull/33378
- https://github.com/home-assistant/core/pull/33308

doudz, 2020-04-20:
Sorry, not tested yet
Because of the covid-19 I have no time to work on zigate (family first)

nerdosity, 2020-05-07:
> Best would be if you also had other Zigbee adapter (like ex. CC2531) if can replicate same issue.
>
> If you can replicate the same issue with two different adapters then unlikely issue is in [...]

Hi, I have a CC2531 but I'm using it with zigbee2mqtt because home assistant doesn't manage ZHA Climate type and I used them perfectly with zigbee2mqtt, so it woul be a bit of trouble to use CC2531 with HASS now.

Hedda, 2020-05-07:
@nerdosity I don't think ZHA stores anything on the CC2531 so you could try temporarily disabling  Zigbee2mqtt and setup ZHA with your CC2531 for a quick test and then restore and re-enable Zigbee2mqtt

Hedda, 2020-05-07:
@nerdosity Have you by the way upgraded to Home Assistant 0.109 as it has many ZHA bug-fixes?

https://www.home-assistant.io/blog/2020/04/29/release-109/

Hedda, 2020-05-13:
Couple even newer pull requests changes how Home Assistant group ZHA devices in Zigbee groups:

- https://github.com/home-assistant/core/pull/34583
- https://github.com/home-assistant/frontend/pull/5602

Guess that those ZHA changes will make it into Home Assistant 0.110 RC (Release Candidate)?

nerdosity, 2020-06-23:
I tested groups with another zigbee stick (ezsp stick) and it works perfectly. Created a group, put a button with the light and it works immediately with the lamp just pairing it with home assistant.

Hedda, 2020-09-09:
@nerdosity Did you manage to fix this in latest HA and if so are you OK to close this yourself as resolved or will you keep it open?

doudz, 2020-10-19:
Just tried using the last HA and it works, so I close this issue

closed 2020-10-19 by doudz