zigpy/zigpy-zigate #37: Not able to include my valve

BornTKill, 2020-05-08:
Dear,

I was able to include my zigbee valve with this project https://github.com/doudz/homeassistant-zigate but not with ZHA Integration in HA.

Each time I try to add the device I receive ;
XX:XX:XX:XX:XX has left the Network.

Any ideas ?
Also can you tell me how to enable debug log ?

For you information my 5 Aqara sensors have been added and my Honeywell Gas Detector also.

Thank You.

Gamester17, 2020-09-03:
See troubleshooting section in https://www.home-assistant.io/integrations/zha/ for how to enable debug logs for ZHA in HA.

More details on debug logs and bug reporting is also available in https://github.com/zigpy/zigpy/blob/dev/CONTRIBUTING.md

Hedda, 2020-09-09:
@BornTKill Did you manage to fix this and if so are you OK to close this yourself as resolved or will you keep it open?

BornTKill, 2020-09-09:
@Hedda I didn’t test yet. I will reopen if need

closed 2020-09-09 by BornTKill

Hedda, 2020-09-10:
Suggest that you then Home Assistant 0.114 or 0.115 when it is released (beta now but the release is planned for 17 September)

https://rc.home-assistant.io/blog/2020/09/07/release-115/

If it still does not work then should follow troubleshooting steps in -> in https://www.home-assistant.io/integrations/zha/

BornTKill, 2020-09-10:
Ok !