zigpy/zigpy-zigate #38: Add command translation layer

pdecat, 2020-05-23:
Some Zigbee commands implemented by Zigate differ from Zigpy, e.g.:

- Management LQI request is 0x0031 in zigpy ([source](https://github.com/zigpy/zigpy/blob/4bd71dc1183456a31b861b580351e16322f4cfa1/zigpy/zdo/types.py#L362), [doc](https://books.google.fr/books?id=m5NYbUpqXY0C&pg=PA277&lpg=PA277&dq=zigbee+0x0031&source=bl&ots=9jO_ztd_5f&sig=ACfU3U0kaOXj61qPJyvgpU-EzkfPphkYZg&hl=fr&sa=X&ved=2ahUKEwi_jarz4snpAhUCxIUKHWiHC6IQ6AEwAXoECAoQAQ#v=snippet&q=0x0031&f=false)) and 0x004E in zigate ([source](https://github.com/fairecasoimeme/ZiGate/blob/v3.1c/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/SerialLink.h#L179), [doc](https://zigate.fr/wiki/commandes-zigate/)).
- Bind is 0x0021 in zigpy ([source](https://github.com/zigpy/zigpy/blob/4bd71dc1183456a31b861b580351e16322f4cfa1/zigpy/zdo/types.py#L358), [doc](https://books.google.fr/books?id=m5NYbUpqXY0C&pg=PA277&lpg=PA277&dq=zigbee+0x0021&source=bl&ots=9jO_ztd_5f&sig=ACfU3U0kaOXj61qPJyvgpU-EzkfPphkYZg&hl=fr&sa=X&ved=2ahUKEwi_jarz4snpAhUCxIUKHWiHC6IQ6AEwAXoECAoQAQ#v=onepage&q=0x0021&f=false)) and 0x0030 in zigate ([source](https://github.com/fairecasoimeme/ZiGate/blob/v3.1c/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/SerialLink.h#L137), [doc](https://zigate.fr/wiki/commandes-zigate/))
- Unbind is 0x0022 in zigpy ([source](https://github.com/zigpy/zigpy/blob/4bd71dc1183456a31b861b580351e16322f4cfa1/zigpy/zdo/types.py#L359), [doc](https://books.google.fr/books?id=m5NYbUpqXY0C&pg=PA277&lpg=PA277&dq=zigbee+0x0022&source=bl&ots=9jO_ztd_5f&sig=ACfU3U0kaOXj61qPJyvgpU-EzkfPphkYZg&hl=fr&sa=X&ved=2ahUKEwi_jarz4snpAhUCxIUKHWiHC6IQ6AEwAXoECAoQAQ#v=onepage&q=0x0022&f=false)) and 0x0031 in zigate ([source](https://github.com/fairecasoimeme/ZiGate/blob/v3.1c/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/SerialLink.h#L139), [doc](https://zigate.fr/wiki/commandes-zigate/))

Maybe a command translation layer could help here, WDYT?

Note: I'm came across this as it prevents zha_map from working (`ZDOCmd.Mgmt_Lqi_req: 49` is `0x0031` in hexadecimal):

```
2020-05-23 12:25:59 DEBUG (MainThread) [zigpy.util] Tries remaining: 1
2020-05-23 12:25:59 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x0000, 0, <ZDOCmd.Mgmt_Lqi_req: 49>, 0, 0, 113, b'q\x00', True, False)
2020-05-23 12:25:59 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'0200000000003100000000027100'
2020-05-23 12:25:59 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x0e{\x02\x00\x00\x00\x00\x001\x00\x00\x00\x00\x02q\x00'
2020-05-23 12:25:59 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x02\x1e{\x02\x12\x02\x10\x02\x10\x02\x10\x02\x10\x02\x101\x02\x10\x02\x10\x02\x10\x02\x10\x02\x12q\x02\x10'=
2020-05-23 12:25:59 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8000000516a600053000
2020-05-23 12:25:59 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'a6000530' LQI:0
2020-05-23 12:25:59 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [166, 0, 1328, b'']
2020-05-23 12:25:59 DEBUG (MainThread) [zigpy.device] [0x0000] Delivery error for seq # 0x71, on endpoint id 0 cluster 0x0031: Message send failure 166
2020-05-23 12:25:59 ERROR (MainThread) [custom_components.zha_map] failed to scan 00:15:8d:00:02:19:ff:ff device: [0x0000:0:0x0031]: Message send failure
```

Gamester17, 2020-09-03:
Probably need to tag @doudz to get his attention?

doudz, 2020-10-19:
There's a confusion between **zigate** command and **zigbee** command
zigpy use zigbee commands, not zigate commands

closed 2020-10-19 by doudz