zigpy/zigpy-zigate #40: Unable to use with dev branch

pipiche38, 2020-08-11:
I'm trying to use the zigpy-zigate lib and zigpy dev branch and I'm getting and error when starting.

```
python3 testzigpy.py
Traceback (most recent call last):
  File "testzigpy.py", line 5, in <module>
    from zigpy_zigate.zigbee.application import ControllerApplication
  File "/var/lib/domoticz/plugins/Domoticz-Zigpy/zigpy_zigate/zigbee/application.py", line 8, in <module>
    from zigpy_zigate import types as t
  File "/var/lib/domoticz/plugins/Domoticz-Zigpy/zigpy_zigate/types.py", line 132, in <module>
    class NWK(zigpy.types.HexRepr, uint16_t):
AttributeError: module 'zigpy.types' has no attribute 'HexRepr'
```

If adding the definition of HexRep (which seems to be removed from the latest zigpy library), I'm getting the following error

```
$ python3 testzigpy.py

DEBUG:asyncio:Using selector: EpollSelector
Traceback (most recent call last):
  File "testzigpy.py", line 71, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 587, in run_until_complete
    return future.result()
  File "testzigpy.py", line 48, in main
    "path": "/dev/ttyUSBRPI3",
TypeError: Can't instantiate abstract class ControllerApplication with abstract methods probe
```

Gamester17, 2020-09-03:
Probably need to tag @doudz  to get his attention?

pipiche38, 2020-09-11:
related to https://github.com/zigpy/zigpy/issues/453

Adminiuga, 2020-09-11:
Try latest release.

closed 2020-09-11 by Adminiuga

pipiche38, 2020-09-12:
> Try latest release.

thanks !