zigpy/zigpy-zigate #42: How to Erase PDM

pipiche38, 2020-09-02:
How can we erase the ZiGate PDM ?

doudz, 2020-09-02:
There's no way with the current implementation

pipiche38, 2020-09-02:
Shame, and I guess that is the same for Switching On/off the blue Led, switch from CE to FCC certification and reducing the Power ?

doudz, 2020-09-02:
Currently zigpy-zigate only support basic, minimal zigate functions, but we can do more.

Even if we add more api functions, I just don't know how we can call them from zigpy, but there's probably a way

doudz, 2020-09-02:
I haven't spent too much time on zigpy so I can't really help on that

pipiche38, 2020-09-03:
> I haven't spent too much time on zigpy so I can't really help on that

No problem. Thanks!

Gamester17, 2020-09-03:
FYI, I think that zigpy-cc (and maybe also zigpy-znp) might have LED control and TX power control or one of those. see:

https://github.com/zigpy/zigpy-cc/issues/42

https://github.com/zigpy/zigpy-cc/issues/33

https://github.com/zha-ng/zigpy-znp/issues/20

https://github.com/zha-ng/zigpy-znp/issues/11

Also see comments that functions like these can be added to zigpy if and when more than one radio library has support for it:

https://github.com/zigpy/zigpy/issues/372

https://github.com/zigpy/zigpy/issues/384

Gamester17, 2020-09-03:
Maybe change topic/subject to something like:

"[REQUEST] Possibility to reset the PDM and erase the all memory via the zigpy library?"

and add little more back-story/information why this is requested? Ex:

Requesting possibility to reset the PDM and erase the all memory on a ZiGate radio via the zigpy library? Perhaps extend zigpy API so can also be used by other radio libraries too?

As per https://github.com/zigpy/zigpy-zigate/issues/42

Posted this as a suggestion as also mentioned as a current limitation by @pipiche38 for his experimental project to make a Zigpy plugin for Domoticz with ZiGate support:

https://github.com/pipiche38/Domoticz-Zigpy

The idea is that his new project might at some point be able to replace his existing ZiGate plugin for Domoticz if and when functions and features get up to par.

https://github.com/pipiche38/Domoticz-Zigpy/blob/master/README.md

closed 2020-11-10 by doudz