zigpy/zigpy-zigate #50: Received unhandled response 0x8011

Cyr-ius, 2020-09-14:
the Homeassistant ZHA module correctly detected the Tuya On_Off switch but unable to control it
When I press the button here is the error
DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8011000704008d1801000600
DEBUG (MainThread) [zigpy_zigate.api] data received 0x8011 b'008d18010006' LQI:0
ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8011

is it possible to add the response 0x8011 which is missing from the zigpy_zigate file ?

Model : [SmartSwitch 1-Gang ](https://imall.com/product/Tuya-Zigbee-Smart-Switch-Without-Neutral-Line-EU-220V-1-2-3-Gang-Metal-Frame-Touch-Panel-Wireless-Remote-Control/%D0%91%D1%8B%D1%82%D0%BE%D0%B2%D0%B0%D1%8F-%D1%8D%D0%BB%D0%B5%D0%BA%D1%82%D1%80%D0%BE%D0%BD%D0%B8%D0%BA%D0%B0/aliexpress.com/4000465170042/144-38103968/en)

{
  "node_descriptor": "<NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4098 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3400_keyjqthh",
  "model": "TS0041",
  "class": "zigpy.device.Device"
}

doudz, 2020-09-14:
Message 0x8011 is just a data acknowledgement, and in that case status code is 0x00 meaning success.
see https://zigate.fr/documentation/commandes-zigate/
We should support that message but it should works without support

Cyr-ius, 2020-09-16:
> Message 0x8011 is just a data acknowledgement, and in that case status code is 0x00 meaning success.
> see https://zigate.fr/documentation/commandes-zigate/
> We should support that message but it should works without support

Is there a method to make these switches work successfully? 
What is the method to find a solution?

doudz, 2020-09-16:
set log level to debug and retry
then post here the log

Cyr-ius, 2020-09-16:
I have a lot of items on zigate but the one that has a problem is a model TS0041  => device 0xd9eb (ec: 1b: bd: ff: fe: 32: f9: 23)
Manfucturer : _TZ3400_keyjqthh.
I have 4, which all have the same concerns

For information, I have the same switch but powered without neutral, that does not pose a problem (_TZ3000_y9xprg33 , TS0011)

screenshot : 
[image]
[image]

zha join log:
[ZHA_JOIN.log](https://github.com/zigpy/zigpy-zigate/files/5232466/ZHA_JOIN.log)

extract homeassistant:
[HomeAssistant.log](https://github.com/zigpy/zigpy-zigate/files/5232460/HomeAssistant.log)

When i press single button on physical device , 
2020-09-16 15:31:52 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80110007a300d9eb01000600
2020-09-16 15:31:52 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8011 b'00d9eb010006' LQI:0
2020-09-16 15:31:52 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8011

When i press switch sensor on home assistant (switch rollback position)
2020-09-16 15:32:29 DEBUG (MainThread) [zigpy.device] [0xd9eb] Extending timeout for 0x27 request
2020-09-16 15:32:29 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0xD9EB, 260, 6, 1, 1, 39, b"\x01'\x01", True, False)
2020-09-16 15:32:29 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'02d9eb010100060104000003012701'
2020-09-16 15:32:29 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b"\x050\x00\x0f-\x02\xd9\xeb\x01\x01\x00\x06\x01\x04\x00\x00\x03\x01'\x01"
2020-09-16 15:32:29 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b"\x02\x150\x02\x10\x02\x1f-\x02\x12\xd9\xeb\x02\x11\x02\x11\x02\x10\x02\x16\x02\x11\x02\x14\x02\x10\x02\x10\x02\x13\x02\x11'\x02\x11"
2020-09-16 15:32:29 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800000051a00aa053000
2020-09-16 15:32:29 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00aa0530' LQI:0
2020-09-16 15:32:29 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 170, 1328, b'']
2020-09-16 15:32:37 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 87020008e7f0010102d9ebaa00
2020-09-16 15:32:37 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8702 b'f0010102d9ebaa' LQI:0
2020-09-16 15:32:37 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [240, 1, 1, <Address address_mode=ADDRESS_MODE.NWK address=0xd9eb>, 170]
2020-09-16 15:32:37 WARNING (MainThread) [zigpy_zigate.zigbee.application] Unexpected message send failure

Thank for help

doudz, 2020-10-17:
support of message 0x8011 will enable that :
https://github.com/zigpy/zigpy-zigate/blob/dev/zigpy_zigate/zigbee/application.py#L153-L159

closed 2020-11-10 by doudz