zigpy/zigpy-zigate #54: [REQUEST] Energy detection scan with zigpy-zigate for ZiGate based adapters

Hedda, 2020-09-21:
Requesting an "energy detection scan" feature in zigpy-zigate if possible matching what @puddly has recently added to zigpy-znp

Suggest implement similar channel energy scanning via some energy scanner tool but for ZiGate based firmware and hardware.

https://github.com/zha-ng/zigpy-znp

https://github.com/zha-ng/zigpy-znp/blob/dev/README.md

# Energy scan
Perform an energy scan to find a quiet Zigbee channel:

```console
$ python -m zigpy_znp.tools.energy_scan /dev/cu.usbmodem14101
Channel energy (mean of 1 / 5):
------------------------------------------------
 + Lower energy is better
 + Active Zigbee networks on a channel may still cause congestion
 + Using 26 in the USA may have lower TX power due to FCC regulations
 + Zigbee channels 15, 20, 25 fall between WiFi channels 1, 6, 11
 + Some Zigbee devices only join networks on channels 15, 20, and 25
------------------------------------------------
 - 11    61.57%  #############################################################
 - 12    60.78%  ############################################################
 - 13    12.16%  ############
 - 14    58.43%  ##########################################################
 - 15    57.65%  #########################################################
 - 16    29.80%  #############################
 - 17    38.82%  ######################################
 - 18    47.06%  ###############################################
 - 19    36.86%  ####################################
 - 20    10.98%  ##########
 - 21    16.47%  ################
 - 22    33.73%  #################################
 - 23    30.59%  ##############################
 - 24    20.39%  ####################
 - 25     5.88%  #####
 - 26*   20.39%  ####################
```

Hedda, 2020-09-21:
@doudz by the way, I do not think this is related to https://github.com/zigpy/zigpy-zigate/issues/44

doudz, 2020-10-19:
> @doudz by the way, I do not think this is related to #44

no it's a different think

it's not currently doable with zigate because the Msg 0x804a doesn't provide energy information

it could be done but requires a firmware change 
https://github.com/fairecasoimeme/ZiGate/blob/master/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/app_general_events_handler.c#L788

https://github.com/fairecasoimeme/ZiGate/blob/master/Module%20Radio/Firmware/src/sdk/JN-SW-4170/Components/ZPSAPL/Include/zps_apl_zdp.h#L1042