zigpy/zigpy-zigate #57: [REQUEST] NVRAM Backup and Restore + Export and Import NCP Trust Center parameters

Hedda, 2020-10-01:
Could it be possible to fully backup ZiGate NVRAM and restore to other ZiGate hardware via the zigpy-zigate radio library?

If possible also Export and Import NCP Trust Center parameters.

Please note that https://github.com/zha-ng/zigpy-znp also has a nice "NVRAM Backup and restore" function.

Also, bellows can Export (backup) and Import (restore) NCP Trust Center parameters, see ->  https://github.com/zigpy/bellows/pull/295

https://github.com/doudz/homeassistant-zigate and https://github.com/doudz/zigate do not feature backup and restore either?

pipiche38, 2020-10-01:
As far as I known, there is no simple possibility except by using Windows exec file provided by NXP

doudz, 2020-10-01:
Python ZiGate lib https://github.com/doudz/zigate provide a way to backup/restore the full memory (zigate firmware and PDM) 
Not sure we can only backup/restore PDM

doudz, 2020-10-01:
But to backup or restore firmware, zigate must be in flash mode.
It can be done automatically for pizigate and zigate din but requires manual activation for zigate usb and wifi (by pushing a button while plugging the zigate)

closed 2020-11-10 by doudz