zigpy/zigpy-zigate #59: [REQUEST] Support for Xiaomi Lumi Gateway MIxx01 Zigbee to WiFi bridge with ZiGate and OpenLumi firmware

Hedda, 2020-11-03:
@doudz Please consider adding for supporting Xiaomi Lumi Gateway with ZiGate and OpenLumi firmware in zigpy-zigate via WiFi

Please see -> https://github.com/Koenkk/zigbee-herdsman/pull/241 and https://github.com/Koenkk/zigbee-herdsman/issues/242

Summary: @G1K, @devbis, @kirovilya (from the @diyruz organization on GitHub) have hacked Xiaomi Lumi Gateway MIxx01 Zigbee to WiFi bridges (Xiaomi gateways DGNWG05LM and ZHWG11LM) with an OpenWRT based firmware on its IMX6 SoC and ZiGate firmware on its NXP JN516x chips (JN5168 and JN5169 chip), or you can make use your own DIY hardware.

They call their project "OpenLumi" and they are currently only testing this themselves with Zigbee2MQTT.

https://github.com/openlumi

As I understand this could maybe be used as ZiGate WiFi but in the practical format of a stand-alone bridge appliance format. 

I believe that this is a very similar concept to how you can use ITead's Sonoff ZBBridge with bellows if you hack its ESP8266 with Tasmota firmware and standard EZSP firmware so that you can use it as a remote coordinator over WiFi via its integrated TCP serial server. For that see -> https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

doudz, 2020-11-03:
good ! if it use the same protocol as ZiGate Wifi so it's already supported (I made recent change to fix ZiGate Wifi support)

Hedda, 2020-11-03:
According to https://github.com/Koenkk/zigbee-herdsman/pull/241 it requires unreleased ZiGate 3.1d firmware or ZiGate 3.1c with a back-ported bug-fix.

doudz, 2020-11-03:
ZiGate 3.1d firmware fix a memory leak and it is released https://github.com/fairecasoimeme/ZiGate/releases/tag/v3.1d

G1K, 2020-11-03:
you can flash any zigate firmware for 5169 and use ser2net to publish the port over tcp . You can also use domiticz with the zigate plugin directly on the gateway

Hedda, 2020-11-03:
@G1K That should mean that can also use your Open Lumi Gateway project with ZHA integration component in Home Assistant too:

https://www.home-assistant.io/integrations/zha/

> If you are use ZiGate or Sonoff ZBBridge you have to use some special usb_path configuration:
> 
> ZiGate USB TTL or DIN: /dev/ttyUSB0 or auto to auto discover the zigate
> PiZigate : pizigate:/dev/serial0
> **Wifi Zigate : socket://[IP]:[PORT] for example socket://192.168.1.10:9999**

ZHA integration component in Home Assistant uses on https://github.com/zigpy/zigpy-zigate and https://github.com/zigpy/zigpy

Hedda, 2020-11-03:
> you can flash any zigate firmware for 5169 and use ser2net to publish the port over tcp . You can also use domiticz with the zigate plugin directly on the gateway

@pipiche38 perhaps also of interest for your Domoticz-Zigpy project as well? https://github.com/pipiche38/Domoticz-Zigpy/

G1K, 2020-11-03:
> @G1K That should mean that can also use your Open Lumi Gateway project with ZHA integration component in Home Assistant too:
>
> https://www.home-assistant.io/integrations/zha/
>
>
> ZHA integration [...]

Sure. you can install ser2net 
`opkg install ser2net`
and write in the config /etc/ser2net. conf for example
```
9999:raw:0:/dev/ttymxc1:115200 8DATABITS NONE 1STOPBIT
```

doudz, 2020-11-03:
So you can close this issue, it's already supported

Hedda, 2020-11-03:
Yes closing this and opened PR for readme update in https://github.com/zigpy/zigpy-zigate/pull/60

closed 2020-11-03 by doudz