zigpy/zigpy-zigate #68: installation Pizigate ?

vini59fr, 2020-11-19:
REbonjour Doudz

Donc, venant de Zigate (qui est devenu obsolete), je cherche à installer mon Pizigate (3.1d) avec ZHA.
Mais probleme dès l'installation

[image]

en log, j'ai ca qui remonte : 
Logger: zigate
Source: /usr/local/lib/python3.8/site-packages/zigate/core.py:594
First occurred: 16:53:32 (6 occurrences)
Last logged: 16:57:20

Unknown response 0x0302

doudz, 2020-11-19:
Il est important de désactiver l'intégration Zigate afin de ne pas avoir de conflit

vini59fr, 2020-11-19:
Je viens de supprimer Zigate -> meme probleme

doudz, 2020-11-19:
C'est impossible car le message d'erreur mentionne la lib Zigate qui n'est pas utilisé par zha

vini59fr, 2020-11-19:
> C'est impossible car le message d'erreur mentionne la lib Zigate qui n'est pas utilisé par zha

meme probleme par rapport à la capture d'ecran je voulais dire :) 
Je n'ai plus rien dans la log ;)

vini59fr, 2020-11-19:
j'ai peut etre raté quelque chose, mais zigpy-zigate ca s'installe manuellement ou pas ?
Comme je ne trouve aucune doc pour l'installer, je suppose que c'est déjà integré à ZHA

vini59fr, 2020-11-19:
En mode automatique, l'assistant ZHA ne voit pas mon port ttyS0 (alors qu'il existe bien et qu'il fonctionne sans probleme avec Zigate) 
[image]

doudz, 2020-11-19:
Je me trompe peut être mais /dev/ttyAMA0 pourrait bien être la pizigate

doudz, 2020-11-19:
@vini59fr oui zigpy-zigate est intégré à zha

vini59fr, 2020-11-19:
> Je me trompe peut être mais /dev/ttyAMA0 pourrait bien être la pizigate

Ça ne fonctionne pas non plus avec ttyAMA0

vini59fr, 2020-11-19:
Pour l'intégration Zigate, j'avais bien suivi les modifications à faire dans config.txt et cmdline.txt, et tout fonctionnait bien sûr ttyS0

Ces 2 modifications sont toujours valable pour zha ? (Sachant que j'ai besoin du BT interne)

doudz, 2020-11-19:
il faudrait mettre en debug en suivant les instructions ici : https://www.home-assistant.io/integrations/zha/#debug-logging

doudz, 2020-11-19:
> Pour l'intégration Zigate, j'avais bien suivi les modifications à faire dans config.txt et cmdline.txt, et tout fonctionnait bien sûr ttyS0
> 
> Ces 2 modifications sont toujours valable pour zha ? (Sachant que j'ai besoin du BT interne)

oui les modifications sont toujours nécessaire et normalement le port qui fonctionnait avant doit toujours fonctionner avec zha

vini59fr, 2020-11-19:
je viens de trouver ca dans mes logs

2020-11-19 19:45:21 DEBUG (MainThread) [zigpy_zigate.uart] PiZiGate detected
2020-11-19 19:45:21 INFO (MainThread) [zigpy_zigate.common] Put PiZiGate in running mode
2020-11-19 19:45:23 DEBUG (MainThread) [zigpy_zigate.api] Unsuccessful radio probe of '/dev/serial0' port
Traceback (most recent call last):
File "/config/deps/lib/python3.8/site-packages/zigpy_zigate/api.py", line 257, in probe
await asyncio.wait_for(api._probe(), timeout=COMMAND_TIMEOUT)
File "/usr/local/lib/python3.8/asyncio/tasks.py", line 498, in wait_for
raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError

doudz, 2020-11-19:
ah effectivement ça doit venir de là, le timeout est trop court
étonnant je n'ai pas eu ce problème
il faudrait passer en manuel et sélectionner directement zigate comme périphérique radio

vini59fr, 2020-11-19:
Ok... Et on fait comment ? :D

closed 2020-11-19 by doudz

kirby0025, 2020-11-20:
Bonjour,

Est-ce que la modification du timeout a bien réglé le problème @vini59fr ?
Je rencontre un problème similaire et même après mise à jour il est toujours là. Ci-après les logs 
```
2020-11-21 00:04:18 DEBUG (MainThread) [zigpy_zigate.uart] PiZiGate detected
2020-11-21 00:04:18 INFO (MainThread) [zigpy_zigate.common] Put PiZiGate in running mode
2020-11-21 00:04:20 ERROR (MainThread) [zigpy_zigate.common] Unable to set PiZiGate GPIO, please check configuration
2020-11-21 00:04:20 ERROR (MainThread) [zigpy_zigate.common]
2020-11-21 00:04:20 DEBUG (MainThread) [zigpy_zigate.uart] Connection made
2020-11-21 00:04:20 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0002 b'01'
2020-11-21 00:04:20 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x00\x02\x00\x01\x02\x01'
2020-11-21 00:04:20 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x10\x02\x12\x02\x10\x02\x11\x02\x12\x02\x11'
2020-11-21 00:04:20 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800000078500000002000000
2020-11-21 00:04:20 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'000000020000' LQI:0
2020-11-21 00:04:20 DEBUG (MainThread) [zigpy_zigate.api] Unsuccessful radio probe of '/dev/ttyAMA0' port
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy_zigate/api.py", line 257, in probe
    try:
  File "/usr/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
```
Lorsque j'essaie en cli de me connecter à la pizigate cela fonctionne correctement, j'ai donc du mal à cerner d'où vient le problème.
```
python3 -m zigpy_zigate.tools.cli -p /dev/ttyAMA0 -d version
Port set to /dev/ttyAMA0
DEBUG:zigpy_zigate.uart:PiZiGate detected
INFO:zigpy_zigate.common:Put PiZiGate in running mode
/srv/homeassistant/lib/python3.7/site-packages/zigpy_zigate/common.py:61: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(17, GPIO.OUT)  # GPIO0
/srv/homeassistant/lib/python3.7/site-packages/zigpy_zigate/common.py:62: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(27, GPIO.OUT)  # GPIO2
DEBUG:zigpy_zigate.uart:Connection made
DEBUG:zigpy_zigate.uart:Send: 0x0010 b''
DEBUG:zigpy_zigate.uart:Frame to send: b'\x00\x10\x00\x00\x10'
DEBUG:zigpy_zigate.uart:Frame escaped: b'\x02\x10\x10\x02\x10\x02\x10\x10'
DEBUG:zigpy_zigate.uart:Frame received: 800000079700000010000000
DEBUG:zigpy_zigate.api:data received 0x8000 b'000000100000' LQI:0
DEBUG:zigpy_zigate.uart:Frame received: 80100005880003031d00
DEBUG:zigpy_zigate.api:data received 0x8010 b'0003031d' LQI:0
Firmware version 3.1d
```
Je pense (peut-être à tort) que les warnings concernant le channel déjà utilisé sont là parce que home-assistant est démarré.

J'utilise home-assistant en venv, j'ai upgrade zigpy-zigate manuellement vers la version 0.7.3 et HA en version 0.118.2.

L'intégration avec zigate fonctionnait à peu près bien, ce n'était pas le plus stable depuis quelques mise à jour mais cela fonctionnait, donc à priori la pizigate fonctionne bien.

kirby0025, 2020-11-20:
Bonjour,

Juste pour dire qu'après avoir mis à jour vers zigpy-zigate 0.7.3, le problème similaire que je rencontrais ne se manifeste plus.

Merci

vini59fr, 2020-11-21:
Oui, la 0.7.3 fonctionne nickel pour installer le pizigate avec zha.

PAR CONTRE, le Bluetooth interne semble ne plus fonctionner.

doudz, 2020-11-21:
la version 0.7.3 devrait être dans HA 0.118.3 je pense.
Concernant le bluetooth, ça doit être un autre problème car je n'y touche pas

commit 62d41441 (2020-11-19): increase probing timeout, fix #68