zigpy/zigpy-zigate #70: TODO : Auto reconnect

doudz, 2020-11-21:


ZeGuigui, 2020-12-15:
How complicated is this to add auto reconnect? Just switched to zha and I have a wifi zigate. I had not seen the issue before upgrading :(

doudz, 2020-12-15:
You shouldn't be hard since this feature already exists for other zigpy radio
Just need a bit of time to add this

ZeGuigui, 2020-12-15:
I started looking on my own. If what I understand is right (but my Python skills are not very developped) I might propose a pull request. I have to dig a little bit more and find a way to test my code on my HA instance (with some debug to check if reconnect do work)

doudz, 2020-12-15:
it should be ok in dev branch

ZeGuigui, 2020-12-15:
Thanks. I was miles away from something that complete. I will give it a try tonight.

ZeGuigui, 2020-12-16:
First tests with last commit not very conclusive.
Starting at 00:46:34 UTC I get some `ZiGate doesn't answer to command` error messages in my log files.
There is no trace of reconnect action (`grep -i reconnect` or `grep -i lost` return no line)

At 00:28:30 there is a log message `Closed serial connection`: 
```
# cat home-assistant.log | grep zigate | grep -i connect
2020-12-15 20:46:30 DEBUG (MainThread) [zigpy_zigate.uart] Connection made
2020-12-16 00:28:30 DEBUG (MainThread) [zigpy_zigate.uart] Closed serial connection
2020-12-16 05:12:07 DEBUG (MainThread) [zigpy_zigate.uart] Connection made
```

The last connection made is from a manual reload integration in HA. I attached my [zigate related logs](https://github.com/zigpy/zigpy-zigate/files/5700502/zigate.log.gz) if that can help.

doudz, 2020-12-16:
ok, thanks for feedback.
So the connection loss is not detected but we see "Closed serial connection" so there's something to check.
I have to test using a real ZiGate WiFi 

Maybe after many NoResponseError we could assume the connection is lost

ZeGuigui, 2020-12-16:
I traced back the error message to `def connection_lost(self, exc) -> None:` in `uart.py` so definitively the loss of the connection is registered. The real question is why is the reconnect not done? We might need to investigate zigpy code but as I mentionned above my python skills are rudimentary! I am more a Java or Perl afficionado.

baimard, 2021-01-29:
Dear team, any news for this issue ?

tygerlord, 2021-06-17:
> 
> 
> Dear team, any news for this issue ?

Seems not...