zigpy/zigpy-zigate #71: segmentation fault with new code for zigate din

totof129, 2020-11-29:
[log.txt](https://github.com/zigpy/zigpy-zigate/files/5612579/log.txt)
I'm using zigate usb (old prolific version) with home-assistant 0.118.3 in a docker container on a synology ds218+
Since the new code for zigate din has been integrated, when adding the zigate usb key in zha integration, a segmentation fault occurs, rebooting home-assistant and preventing to add the zigate key in ZHA. This seems linked to the load of libusb1 when zigpy-zigate code tests if the key is a zigate din
When checking the version of pyusb used, it is 1.0.0b1; this seg fault bug seems to be an already identified bug corrected in 1.0.2 and 1.1.0 : 
https://github.com/pyusb/pyusb/issues/232

doudz, 2020-11-29:
maybe we can fix it by adjusting requirements in setup.py
https://github.com/zigpy/zigpy-zigate/blob/dev/setup.py#L53
`'pyusb>=1.0.2',`

doudz, 2020-11-29:
I don't know if it's possible in a docker container but you could try to update pyusb with the command `python3 -m pip install pyusb>=1.0.2`

totof129, 2020-11-29:
Indeed it could be a good solution to force pyusb version in requirements (I would advice to directly put 1.1.0)

Regarding docker, it is indeed possible to open a bash terminal in the container and to update pyusb, but as home-assistant update is done by updating the docker image, the modification would be erased at each image update

doudz, 2020-11-29:
Yes , but if you can, try to update pyusb to check if it fix the problem, and I'll update setup.py

totof129, 2020-11-29:
I do confirm that it solves the pbl

[log.txt](https://github.com/zigpy/zigpy-zigate/files/5612876/log.txt)

commit 7748446a (2020-11-29): require pyusb>=1.1.0, fix #71
closed 2020-11-29 by doudz