zigpy/zigpy-zigate #72: Pizigate - ZHA installation error "No response to command 0x0002"

pedlou, 2020-11-30:
Hi,
I am using the setup with Pizigate, RPI3B+ and hassio, while trying the ZHA integration I am having some issues.
In Configuration+System+Hardware I have pizigate as:
/dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0
In Integration for ZHA I am using:
1. Serial device path: enter manually
2. Radio type: Zigate
3. Serial device path: /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0
result: Failed to connect.
In logs I can see the following:
2020-11-30 12:30:40 DEBUG (MainThread) [zigpy_zigate.uart] Connection made
2020-11-30 12:30:40 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0002 b'01'
2020-11-30 12:30:40 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x00\x02\x00\x01\x02\x01'
2020-11-30 12:30:40 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x10\x02\x12\x02\x10\x02\x11\x02\x12\x02\x11'
2020-11-30 12:30:41 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0002
2020-11-30 12:30:41 WARNING (MainThread) [zigpy_zigate.api] Retry command 0x0002
2020-11-30 12:30:41 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0002 b'01'
2020-11-30 12:30:41 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x00\x02\x00\x01\x02\x01'
2020-11-30 12:30:41 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x10\x02\x12\x02\x10\x02\x11\x02\x12\x02\x11'
2020-11-30 12:30:43 DEBUG (MainThread) [zigpy_zigate.api] Unsuccessful radio probe of '/dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0' port

Can you provide some assistance?
Thanks in advance
BR
Pedro

doudz, 2020-11-30:
I think there's a misunderstanding somewhere, you said you're using a pizigate 
https://zigate.fr/produit/pizigate-v1-0/
But you said the device path is /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 
CP2102 USB to UART is a USB TTL zigate not a pizigate
https://zigate.fr/produit/zigate-ttl/

doudz, 2020-11-30:
if you're really using a pizigate, the path is something like /dev/serial0 or /dev/ttyAMA0

pedlou, 2020-11-30:
Hi Doudz,
Thanks for the fast response.
It is a pizigate. let me copy/paste the Supervisor>System>Hardware:
serial:
  - /dev/ttyAMA0
  - /dev/ttyUSB0
  - >-
    /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0
If I try to use /dev/ttyAMA0 in ZHA integration I get:

2020-11-30 12:47:52 DEBUG (MainThread) [zigpy_znp.zigbee.application] Probing /dev/ttyAMA0
2020-11-30 12:47:52 DEBUG (MainThread) [zigpy_znp.uart] Connecting to /dev/ttyAMA0 at 115200 baud
2020-11-30 12:47:52 DEBUG (MainThread) [zigpy_znp.uart] Opened /dev/ttyAMA0 serial port
2020-11-30 12:47:52 DEBUG (MainThread) [zigpy_znp.uart] Toggling RTS/CTS to skip CC2652R bootloader
2020-11-30 12:47:52 DEBUG (MainThread) [zigpy_znp.uart] Connected to /dev/ttyAMA0 at 115200 baud
2020-11-30 12:47:52 DEBUG (MainThread) [zigpy_znp.api] Waiting 1s before sending anything
2020-11-30 12:47:53 DEBUG (MainThread) [zigpy_znp.api] Sending bootloader skip byte
2020-11-30 12:47:53 DEBUG (MainThread) [zigpy_znp.uart] Sending data: b'\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF\xEF'
2020-11-30 12:47:53 DEBUG (MainThread) [zigpy_znp.api] Waiting 1s or until a reset indication is received
2020-11-30 12:47:53 DEBUG (MainThread) [zigpy_znp.api] Creating one-shot listener OneShotResponseListener(matching_commands=(SYS.ResetInd.Callback(Reason=None, TransportRev=None, ProductId=None, MajorRel=None, MinorRel=None, MaintRel=None),), future=<Future pending cb=[ZNP.wait_for_responses.<locals>.<lambda>() at /usr/local/lib/python3.8/site-packages/zigpy_znp/api.py:449, <TaskWakeupMethWrapper object at 0x66c0e2f8>()]>)
2020-11-30 12:47:54 DEBUG (MainThread) [zigpy_znp.api] Removing listener OneShotResponseListener(matching_commands=(SYS.ResetInd.Callback(Reason=None, TransportRev=None, ProductId=None, MajorRel=None, MinorRel=None, MaintRel=None),), future=<Future cancelled>)
2020-11-30 12:47:54 DEBUG (MainThread) [zigpy_znp.api] Cleaning up empty listener list for header CommandHeader(id=0x80, subsystem=Subsystem.SYS, type=CommandType.AREQ)
2020-11-30 12:47:54 DEBUG (MainThread) [zigpy_znp.api] There are 0 callbacks and 0 one-shot listeners remaining
2020-11-30 12:47:54 DEBUG (MainThread) [zigpy_znp.api] Testing connection to /dev/ttyAMA0
2020-11-30 12:47:54 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Version.Req()
2020-11-30 12:47:54 DEBUG (MainThread) [zigpy_znp.api] Creating one-shot listener OneShotResponseListener(matching_commands=(SYS.Version.Rsp(TransportRev=None, ProductId=None, MajorRel=None, MinorRel=None, MaintRel=None, CodeRevision=None, BootloaderBuildType=None, BootloaderRevision=None), RPCError.CommandNotRecognized.Rsp(ErrorCode=None, RequestHeader=CommandHeader(id=0x02, subsystem=Subsystem.SYS, type=CommandType.SREQ))), future=<Future pending cb=[ZNP.wait_for_responses.<locals>.<lambda>() at /usr/local/lib/python3.8/site-packages/zigpy_znp/api.py:449, <TaskWakeupMethWrapper object at 0x66c0e400>()]>)
2020-11-30 12:47:54 DEBUG (MainThread) [zigpy_znp.uart] Sending data: b'\xFE\x00\x21\x02\x23'
2020-11-30 12:47:57 DEBUG (MainThread) [zigpy_znp.api] Removing listener OneShotResponseListener(matching_commands=(SYS.Version.Rsp(TransportRev=None, ProductId=None, MajorRel=None, MinorRel=None, MaintRel=None, CodeRevision=None, BootloaderBuildType=None, BootloaderRevision=None), RPCError.CommandNotRecognized.Rsp(ErrorCode=None, RequestHeader=CommandHeader(id=0x02, subsystem=Subsystem.SYS, type=CommandType.SREQ))), future=<Future cancelled>)
2020-11-30 12:47:57 DEBUG (MainThread) [zigpy_znp.api] Cleaning up empty listener list for header CommandHeader(id=0x00, subsystem=Subsystem.RPCError, type=CommandType.SRSP)
2020-11-30 12:47:57 DEBUG (MainThread) [zigpy_znp.api] Cleaning up empty listener list for header CommandHeader(id=0x02, subsystem=Subsystem.SYS, type=CommandType.SRSP)
2020-11-30 12:47:57 DEBUG (MainThread) [zigpy_znp.api] There are 0 callbacks and 0 one-shot listeners remaining
2020-11-30 12:47:57 WARNING (MainThread) [zigpy_znp.zigbee.application] Failed to probe ZNP radio with config {'path': '/dev/ttyAMA0', 'flow_control': None, 'baudrate': 115200}
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy_znp/zigbee/application.py", line 142, in probe
    await znp.connect()
  File "/usr/local/lib/python3.8/site-packages/zigpy_znp/api.py", line 236, in connect
    version = await self.request(c.SYS.Version.Req())
  File "/usr/local/lib/python3.8/site-packages/zigpy_znp/api.py", line 520, in request
[... cut ...]
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy_xbee/api.py", line 579, in probe
    await asyncio.wait_for(api._probe(), timeout=PROBE_TIMEOUT)
  File "/usr/local/lib/python3.8/asyncio/tasks.py", line 491, in wait_for
    return fut.result()
  File "/usr/local/lib/python3.8/site-packages/zigpy_xbee/api.py", line 600, in _probe
    raise APIException("Failed to configure XBee for API mode")
zigpy.exceptions.APIException: Failed to configure XBee for API mode

Thanks in advance
BR
Pedro

doudz, 2020-11-30:
What is the version of Home Assistant ?

pedlou, 2020-11-30:
Version | 0.118.4
-- | --
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
Virtual Environment | false
Python Version | 3.8.6
Operating System Family | Linux
Operating System Version | 4.19.127-v7
CPU Architecture | armv7l
Timezone | Europe/Lisbon

doudz, 2020-11-30:
Ok, did you adjust the boot config to disable internal bluetooth ? because it conflicts with pizigate on Rpi 3

pedlou, 2020-11-30:
Hi Doudz,

**/boot/config.txt inline:**
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# HassOS - don't change it!
disable_splash=1
kernel=u-boot.bin

# uncomment for aarch64 bit support
#arm_64bit=1

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[all]
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2
dtoverlay = pi3-disable-bt
enable_uart = 1

**/boot/cmdline.txt inline**

dwc_otg.lpm_enable=0 console=tty1

doudz, 2020-11-30:
I don't think it's important but did you tried to remove space around egal sign ?
Change
```
dtoverlay = pi3-disable-bt
enable_uart = 1

```
with
```
dtoverlay=pi3-disable-bt
enable_uart=1
```

pedlou, 2020-11-30:
will test it

pedlou, 2020-11-30:
It looks like it makes total difference as now integration is successful
[image]
I will now try to perform some Zigbee devices integration.
many thanks

closed 2020-12-01 by doudz