zigpy/zigpy-zigate #74: Issue with paring

pawelka, 2020-12-22:
Hi,

I migrated from zigate home assistant with firmware 3.1c to zha with firmware 3.1d and I have a lot of trouble with pairing aqara/xiaomi devices. I cannot pair few of them and I could pair rest after many tries.

Is it possible that some xiaomi/aqara "hack" for custom implementation of 3.0 spec are not migrated to zigpy-zigate?

For example I see weird data for reporting FF01 attribute and I found in zigate library that there is special implementation for this this attribute for xiaomi responses.

pawelka, 2020-12-22:
Generally wall switch is paired with zigate dongle, but it's not recognized by HA. Switch is show that is pair and I see commands in HA logs, but zha does know device

doudz, 2020-12-22:
Could you provide some debug log ?

pawelka, 2020-12-22:
Logs from pairing. Clean at the beginning, entered in zha pairing mode, till led on switch show paired and no activity. There can be also some other logs from other devices.

> 2020-12-22 18:24:59 WARNING (MainThread) [homeassistant.config_entries] Config entry for broadlink not ready yet. Retrying in 80 seconds
2020-12-22 18:25:01 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 801100083d00529a01000a6700
2020-12-22 18:25:01 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8011 b'00529a01000a67' LQI:0
2020-12-22 18:25:01 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0x529a, 1, 10, 103]
2020-12-22 18:25:01 DEBUG (MainThread) [zigpy_zigate.zigbee.application] ACK Data received 103 0
2020-12-22 18:25:02 DEBUG (MainThread) [zigpy_zigate.zigbee.application] Broadcast not implemented.
2020-12-22 18:25:02 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0049 b'fffc3c00'
2020-12-22 18:25:02 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x00I\x00\x04r\xff\xfc<\x00'
2020-12-22 18:25:02 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x10I\x02\x10\x02\x14r\xff\xfc<\x02\x10'
2020-12-22 18:25:02 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800000073b009f0049026800
2020-12-22 18:25:02 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'009f00490268' LQI:0
2020-12-22 18:25:02 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 159, 73, b'\x02h']
2020-12-22 18:25:02 INFO (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for 10.10.10.107: Error connecting to ('10.10.10.107', 6053): [Errno 113] Connect call failed ('10.10.10.107', 6053)
2020-12-22 18:25:02 INFO (MainThread) [homeassistant.components.esphome] Trying to reconnect to 10.10.10.107 in 60 seconds
2020-12-22 18:25:04 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001c9a00010400060201027014020000187c0a0000100100f02300147003ab
2020-12-22 18:25:04 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00010400060201027014020000187c0a0000100100f02300147003' LQI:171
2020-12-22 18:25:04 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 6, 2, 1, <Address address_mode=ADDRESS_MODE.NWK address=0x7014>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18|\n\x00\x00\x10\x01\x00\xf0#\x00\x14p\x03']
2020-12-22 18:25:04 DEBUG (MainThread) [zigpy_zigate.zigbee.application] No such device 0x7014
2020-12-22 18:25:04 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001c9a00010400060201027014020000187c0a0000100100f02300147003ab
2020-12-22 18:25:04 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00010400060201027014020000187c0a0000100100f02300147003' LQI:171
2020-12-22 18:25:04 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 6, 2, 1, <Address address_mode=ADDRESS_MODE.NWK address=0x7014>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18|\n\x00\x00\x10\x01\x00\xf0#\x00\x14p\x03']
2020-12-22 18:25:04 DEBUG (MainThread) [zigpy_zigate.zigbee.application] No such device 0x7014
2020-12-22 18:25:04 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001c9a00010400060201027014020000187c0a0000100100f02300147003ab
2020-12-22 18:25:04 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00010400060201027014020000187c0a0000100100f02300147003' LQI:171
2020-12-22 18:25:04 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 6, 2, 1, <Address address_mode=ADDRESS_MODE.NWK address=0x7014>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18|\n\x00\x00\x10\x01\x00\xf0#\x00\x14p\x03']
2020-12-22 18:25:04 DEBUG (MainThread) [zigpy_zigate.zigbee.application] No such device 0x7014
2020-12-22 18:25:05 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800200152200010400060401027014020000187d0a00001000a8
2020-12-22 18:25:05 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00010400060401027014020000187d0a00001000' LQI:168
2020-12-22 18:25:05 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 6, 4, 1, <Address address_mode=ADDRESS_MODE.NWK address=0x7014>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18}\n\x00\x00\x10\x00']
2020-12-22 18:25:05 DEBUG (MainThread) [zigpy_zigate.zigbee.application] No such device 0x7014
2020-12-22 18:25:05 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800200152200010400060401027014020000187d0a00001000a8
2020-12-22 18:25:05 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00010400060401027014020000187d0a00001000' LQI:168
2020-12-22 18:25:05 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 6, 4, 1, <Address address_mode=ADDRESS_MODE.NWK address=0x7014>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18}\n\x00\x00\x10\x00']
2020-12-22 18:25:05 DEBUG (MainThread) [zigpy_zigate.zigbee.application] No such device 0x7014
2020-12-22 18:25:05 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 801100085e00ad0801000a6900
2020-12-22 18:25:05 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8011 b'00ad0801000a69' LQI:0
2020-12-22 18:25:05 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0xad08, 1, 10, 105]
2020-12-22 18:25:05 DEBUG (MainThread) [zigpy_zigate.zigbee.application] ACK Data received 105 0
[... cut ...] -5': 39, '0xff01-154': 0, '0xff01-8': 4385, '0xff01-7': 0, '0xff01-9': 1283}]
2020-12-22 18:25:29 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 801100084d00eb5b01000a6f00
2020-12-22 18:25:29 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8011 b'00eb5b01000a6f' LQI:0
2020-12-22 18:25:29 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0xeb5b, 1, 10, 111]
2020-12-22 18:25:29 DEBUG (MainThread) [zigpy_zigate.zigbee.application] ACK Data received 111 0

doudz, 2020-12-22:
Since your device is already paired to zigate I suggest you to firstly remove the device.
To do this, I usually put the device in pair mode (for example on xiaomi by long press on pair button) without zigate in permit join mode, so the device will just leave zigate without re-join.
Then I activate the permit join mode and put device in pair mode again

pawelka, 2020-12-22:
Thank you for sugestion, I'll check this early next year ;-) and I'll comment this ticket.

pawelka, 2021-01-07:
You are right. It works correctly when first unpair/remove device and than pair again (no rejoin). 
It works. 

Thank you for support.

closed 2021-01-07 by pawelka