zigpy/zigpy-zigate #75: Can't turn on/off group entities

Jsl-1, 2020-12-23:
Hi,

We have an error when trying to turn on group entities in HA because mrequest is not implemented in zigpy_zigate/zigbee/application.py.

Is there any plan to implement this method ?

Thank you for your hard work !

2020-12-23 09:22:19 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140120683100576]
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 135, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1445, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1480, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await self.hass.helpers.service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 593, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 664, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 630, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 230, in async_handle_light_on_service
    await light.async_turn_on(**params)
  File "/usr/src/homeassistant/homeassistant/components/zha/light.py", line 529, in async_turn_on
    await super().async_turn_on(**kwargs)
  File "/usr/src/homeassistant/homeassistant/components/zha/light.py", line 225, in async_turn_on
    result = await self._on_off_channel.on()
  File "/usr/local/lib/python3.8/site-packages/zigpy/group.py", line 44, in request
    res = await self.application.mrequest(
  File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 255, in mrequest
    raise NotImplementedError
NotImplementedError

I made a working dirty fix to implement this : 

    async def mrequest(
        self,
        group_id,
        profile,
        cluster,
        src_ep,
        sequence,
        data,
        *,
        hops=0,
        non_member_radius=3,
    ):
        src_ep = 1
        dst_ep = src_ep
        LOGGER.debug('mrequest %s',
                     (group_id, profile, cluster, src_ep, sequence, data, hops, non_member_radius))
        try:
            v, lqi = await self._api.raw_aps_data_request(group_id, src_ep, dst_ep, profile, cluster, data, 1)
        except NoResponseError:
            return 1, "ZiGate doesn't answer to command"
        req_id = v[1]
        send_fut = asyncio.Future()
        self._pending[req_id] = send_fut

        if v[0] != 0:
            self._pending.pop(req_id)
            return v[0], "Message send failure {}".format(v[0])

        # disabled because of https://github.com/fairecasoimeme/ZiGate/issues/324
        # try:
        #     v = await asyncio.wait_for(send_fut, 120)
        # except asyncio.TimeoutError:
        #     return 1, "timeout waiting for message %s send ACK" % (sequence, )
        # finally:
        #     self._pending.pop(req_id)
        # return v, "Message sent"
        return 0, "Message sent"

doudz, 2020-12-23:
I didn't realized mrequest was missing, sorry.
I'll add it soon, thanks

Jsl-1, 2020-12-23:
Ok thank you :)
And take your time, it's not a big deal ! I don't see any important use case to this method...

commit 6b0592dc (2020-12-24): Fix #75
closed 2020-12-24 by doudz