zigpy/zigpy-zigate #76: Question about OTA update with zigpy-zigate

Jsl-1, 2020-12-23:
Hi,

All my ota update attempt have failed with zha/zigpy-zigate. Is there a known issue with this ? Does someone had to ota update  with zigpy-zigate ?

Here is a log with different attempt with tradfri provider and hue firmware in ota folder. All ota update stopped at file_offset = 40

https://termbin.com/n6be

Thanks !

Hedda, 2021-01-18:
Have you created otau_directory and then also manually downloaded + added the right OTA firmware update image files yourself?

https://www.home-assistant.io/integrations/zha/#ota-firmware-updates

zigpy only automatically download OTA firmware update image files for IKEA and Ledvance so far, so not for Philips Hue devices.

If want automatic OTA updates for Philips Hue then you need to code a provider ->  https://github.com/zigpy/zigpy/issues?q=ota

Hedda, 2021-01-18:
> If want automatic OTA updates for Hue then you need to code a provider -> https://github.com/zigpy/zigpy/issues?q=ota

If this is the case then suggest that you specifically check out https://github.com/zigpy/zigpy/issues/154

Jsl-1, 2021-01-18:
Hi @Hedda,

All OTA files are fine and detected correctly by zigpy.  OTA update transfert start but is stopped verry early in the process. I have tried Ikea provider and some hue update files  downloaded from https://github.com/Koenkk/zigbee-OTA/blob/master/index.json

I moved to zigpy-znp for now and ota update works perfectly without changing anything. So i'm pretty sure there's something broken in the process with zigate.

I can do more test next week with a clean HA instance and my unused zigate to help.

max5962, 2022-01-24:
I think i have the same issue when trying to update a zigbee device : https://github.com/fairecasoimeme/Zlinky_TIC/issues/44

Hedda, 2022-01-25:
@max5962 To rule out zigpy, any chance could also test ZHA with other compatible Zigbee Coordinator not using zigpy-zigate?

https://www.home-assistant.io/integrations/zha#compatible-hardware

@Jsl-1 for example wrote that tested a Texas Instruments based Zigbee Coordinator with "znp" which could help narrow it down:

> I moved to zigpy-znp for now and ota update works perfectly without changing anything. So i'm pretty sure there's something broken in the process with zigate.

Sadly there is not yet an easy way to migrate back and forth between ZiGate and others so advice testing on a separate instance.

pdecat, 2022-01-25:
A few months ago, I moved from using a Zigate v1 with zigpy-zigate to a zzh with zigpy-znp because the former had a maximum of 40 devices (I have 76 as of today), and there was no export/import migration process to Zigate v2.

I did the migration from Zigate to zzh progressively using Remote Home Assistant as one cannot have two ZHA integrations in a single HA instance.

Since then, all sort of issues I had with the Zigate v1 from time to time  are gone (devices not responding to commands, unreadable attributes, and painful OTA with some devices).

Updating the firmware of the ZLinky_TIC went well twice for me.

max5962, 2022-01-26:
@pdecat can you send a link of your coordinator ? ( i will updagrade too )

pdecat, 2022-01-26:
It's the CC2652R based zzh: https://electrolama.com/projects/zig-a-zig-ah/

max5962, 2022-01-26:
does 5btn ikea works with that coordinator ?

pdecat, 2022-01-26:
It does, I've got 4 of them and button presses events are properly reaching ZHA and I use some in my automations.  But one of the remotes is facing a [quick battery drain issue](https://www.google.com/search?q=ikea+5+button+battery+drain) since a recent firmware update I believe, despite all having the same firmware version. For now, I've replaced it by the new STYRBAR remote which uses rechargeable AAA batteries and had no apparent issue for over a month (got three of them, all fine).

PS: we should probably move this conversation elsewhere (maybe https://community.home-assistant.io/) to keep this issue on topic.