zigpy/zigpy-zigate #77: Listening to group messages

powerpaul17, 2021-01-03:
Hello and thanks already for this great software. I am currently running into problems receiving messages from an IKEA tradfri remote control in Home Assistant. As I found out the remote is sending its messages to a group. These messages are not showing up in the log thus I think they are not handled by zigpy-zigate. Also the coordinator (PiZiGate) is showing a device type of "unknown". This results in it not being found in the list of devices when I tried to create a ZigBee group for the remote and the coordinator.

I am able to code in python but I would need some assistance for pointing me into the right direction. Thank you in advance.

MattWestb, 2021-01-04:
If the devices is being flagged "unknown" is not OK paired to the coordinator or its one model that the system is not knowing (All normal IKEA controlling devices that is currently known is being supported by ZHA). Then one IKEA controlling device is successful pairing and its have one ZLL groupe ZHA is adding it to the system or if its one ZB3 its adding it to groupe 0x000 that is the coordinator so your system is getting the message its sending.
Make one debug log then pairing the device and posting it here so its being possible see what is happening.

powerpaul17, 2021-01-04:
Thanks for the response/explanation. The IKEA remote is recognized and paired correctly. The problem seems to be the coordinator. It is a PiZiGate and has the device type "unknown". As far as I saw in the source code it never requests the node descriptor. The DeConz device for example does that.

Maybe there was a `dev._initialize()` missing in `startup`

MattWestb, 2021-01-04:
If the coordinator is not correctly initiated / recodnoced / setted up its no way for ZHA to binding groups to it and you cant getting any messages from groups = light controlling devices that is sending commands to groups.
So its more than likely one PiZiGate undocumented future (=BUG).

LG (= not Luxurious Goldstar) MW

powerpaul17, 2021-01-05:
I managed to get the coordinator initialized by adding a `dev._initialize()` in the `startup()` routine. But it still doesn't listen to group messages. Also, in the `zha` integration, there is no option to issue a cluster command for the coordinator device.

I would really like to help getting this done, but can you point me to the code in e.g. the `zigpy-deconz` repo where this group b inding/listening happens?

MattWestb, 2021-01-05:
Its best @doudz is taking on look on it hi is the person that is knowing the code.

powerpaul17, 2021-01-06:
Ok, I got more information out of the log when pairing the remote control:

```
2021-01-05 21:48:13 WARNING (MainThread) [zhaquirks.ikea] unable to locate required group info - falling back to group 0x0000.
2021-01-05 21:48:13 DEBUG (MainThread) [zigpy.endpoint] [0x0000:1] Cannot add 0x0000 group, no groups cluster
2021-01-05 21:48:13 DEBUG (MainThread) [zigpy.endpoint] [0x0000:11] Cannot add 0x0000 group, no groups cluster
2021-01-05 21:48:13 DEBUG (MainThread) [zigpy.endpoint] [0x0000:10] Cannot add 0x0000 group, no groups cluster
2021-01-05 21:48:13 DEBUG (MainThread) [zigpy.endpoint] [0x0000:110] Cannot add 0x0000 group, no groups cluster
2021-01-05 21:48:13 DEBUG (MainThread) [zigpy.endpoint] [0x0000:21] Cannot add 0x0000 group, no groups cluster
2021-01-05 21:48:13 DEBUG (MainThread) [zigpy.endpoint] [0x0000:8] Cannot add 0x0000 group, no groups cluster
```

Apparently the coordinator has no groups cluster.

powerpaul17, 2021-02-04:
@doudz can you please have a look at this (when you have time) and point me into the right direction, I would really like to get this resolved. Thanks a lot for this project!

doudz, 2021-02-08:
Looks like there are many information missing on zigate device declaration, endpoint list and node descriptor are empty
I have to find why, maybe I have to provide a quirk
[image]

MattWestb, 2021-02-08:
I think bellows is very much different but can being good taking on look at.
https://github.com/zigpy/bellows/blob/dev/bellows/zigbee/application.py

doudz, 2021-02-08:
Yes, but if you compare belows, znp, deconz or xbee, none of them are doing the same, so I don't know what's the good practice.
Maybe @Adminiuga could give some information ?

Adminiuga, 2021-02-08:
so, this looks like two different problems:

1. What is the zigate API for the Zigate to  start receiving multicast (group) messages? e.g. for EZSP/XBee you have to "subscribe" to the group address, while ZNP and ConBee sticks just pass through any incoming multicast
2. the fake zigpy device for the coordinator, is just a helper in case the special API is required, as it implements "add_to_group" and "remove_from_group" methods, if there's a special API

as to "how" to add the zigpy device representing the coordinator, also depends on the coordinator and ATM I don't have a best practice. For devices which respond to "loopback" device interview/ZDO discovery, it could be as easy as `handle_device_join(self.ieee, self.nwk)` (deCONZ/ConBee implementation) in other cases may need to build the zigpy.Device instance manually and add to the devices, like XBee implementation.

powerpaul17, 2021-02-09:
> Yes, but if you compare belows, znp, deconz or xbee, none of them are doing the same, so I don't know what's the good practice.
> Maybe @Adminiuga could give some information ?

Yes, that's also what I found when I looked at the different implementations, it's different for every device it seems. I already tried initializing and got a node descriptor and some endpoints but no groups cluster. ConBee has a method for adding itself to the groups as far as I could see.

According to this: https://github.com/fairecasoimeme/ZiGate/issues/64 it should work with zigate, so it's a matter of implementation I guess.

max5962, 2021-04-16:
Hello,

Any update about this issue ? 

Thnaks every one !

doudz, 2021-05-11:
Groups are not yet supported

doudz, 2021-11-17:
no, sorry, I have no time to work on zigate 😞

powerpaul17, 2021-11-25:
@doudz Can you please point out where to start. I may have some time to look into this..

max5962, 2021-12-30:
@powerpaul17 have you done some progress about that issue ? thanks

powerpaul17, 2021-12-30:
> @powerpaul17 have you done some progress about that issue ? thanks

No, sorry, since nobody gave me any hints about what's exactly missing.. When I bought the zigate adapter I thought there would be some kind of abstraction layer so every adapter would behave the same but apparently it is not so.

I tried to understand the implementation but every one is different and I'm not familiar enough with the inner workings of a zigbee network to fix this correctly.

So, basically I gave up on this.. :disappointed:  maybe I'll buy a different zigbee adapter someday.. :slightly_frowning_face:

pipiche38, 2022-01-02:
Unfortunately the issue is on the firmware of the ZiGate, which is filtering those group message. In order to get those, you must have Zigate having the group membership.

majkrzak, 2022-01-02:
> the issue is on the firmware of the ZiGate
@pipiche38: is this reported somewhere? It should be the fixed there then.

pipiche38, 2022-01-02:
You can report here, and even do the changes
https://github.com/fairecasoimeme/ZiGatev2
https://github.com/fairecasoimeme/ZiGate

majkrzak, 2022-10-20:
@pipiche38 In see the request you have linked are closed now. I've recently upgraded to the latest firmware of the Pizigate, but I still can not setup the ikea buttons.

pipiche38, 2022-10-20:
The Zigate do not behave as ZNP or deConz or Bellows. At least the firmware developped for the Zigate.

So if you want to pair a ikea buttons with a zigate you have to :

1- pair the ikea button with the zigate to get the remote in the zigate network ( you press 4 times the pairing button)
2- pair an ikea bulb with the zigate
3- associate the ikea button with the ikea bulb by a long press ( around 10s )

then at that stage you should be able to command the bulb from the remote 

4- Now you need to do a group membership scan . I don't know how you do that with zigpy-zigate.
     you should receive the group Id which have created when the remote paired to the bulb

5- you have to make the Zigate a membership of this group

When that is done, the zigate will see all event from the remote, and you can even remove the bulb from the group.

PS/ Some other ikea remote are hard coded to belongs to group 0x0000. In that case, you just have to make the zigate part of group 0x0000 and that should work.

It is obvious that is a complex process, which is much simpler on the other coordinator.

majkrzak, 2022-10-20:
Ech.. I had hope it got finally fixed

majkrzak, 2022-11-09:
> 5- you have to make the Zigate a membership of this group

@pipiche38 
What will be the way of achieving this? Somehow through the cluster commands?

pipiche38, 2022-11-10:
you need to send a Add Group Membership command of the Zigate Nwkid 0x0000 and you need to provide the Group Id