zigpy/zigpy-zigate #81: Error doing job: Exception in callback SerialTransport._read_ready()

HRHL53, 2021-01-14:
Hello,

I updated all software packages on my RPi4 yesterday and ever since I get this error a lot and my zigbee devices stop working after a while. After a reboot they reconnect but only for a few minutes.

```
Error doing job: Exception in callback SerialTransport._read_ready()
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/asyncio/events.py", line 81, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.8/site-packages/serial_asyncio/__init__.py", line 119, in _read_ready
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy_zigate/uart.py", line 60, in data_received
    cmd, length, checksum, f_data, lqi = struct.unpack('!HHB%dsB' % (len(frame) - 6), frame)
struct.error: bad char in struct format
```

I tried the beta channel for HA supervised but without result.

## System Health

version | 2021.1.1
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.4.83-v7l+
arch | armv7l
timezone | Europe/Amsterdam

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Raspbian GNU/Linux 10 (buster)
-- | --
update_channel | beta
supervisor_version | 2021.01.1
docker_version | 20.10.2
disk_total | 14.3 GB
disk_used | 5.7 GB
healthy | true
supported | failed to load: Unsupported
supervisor_api | ok
version_api | ok
installed_addons | Check Home Assistant configuration (3.6.0), Node-RED (7.2.11), Home Assistant Google Drive Backup (0.103.0), Samba share (9.3.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
mode | auto-gen
resources | 0

</details>

doudz, 2021-01-14:
Could you provide log in debug mode ?
it looks like a unsupported zigate message.
Are you using firmware 3.1d ?

HRHL53, 2021-01-14:
This is a part of the log in debug mode: https://pastebin.com/yxvJ8d1u
I am using 3.1d.

doudz, 2021-01-14:
what kind of zigate ?

HRHL53, 2021-01-14:
The ZiGate USB-TTL

After these logs:
```
2021-01-14 20:24:52 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
2021-01-14 20:24:52 WARNING (MainThread) [zigpy_zigate.api] Retry command 0x0530
2021-01-14 20:24:52 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'02dcb8010100060104000005000f000000'
2021-01-14 20:24:52 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x11K\x02\xdc\xb8\x01\x01\x00\x06\x01\x04\x00\x00\x05\x00\x0f\x00\x00\x00'
2021-01-14 20:24:52 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x11K\x02\x12\xdc\xb8\x02\x11\x02\x11\x02\x10\x02\x16\x02\x11\x02\x14\x02\x10\x02\x10\x02\x15\x02\x10\x02\x1f\x02\x10\x02\x10\x02\x10'
```

The error in my starting post repeats thousands of times.

doudz, 2021-01-14:
are you sure there isn't another integration trying to use the same serial port ?
```
2021-01-14 20:24:50 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback SerialTransport._read_ready()
Traceback (most recent call last):
File "/usr/local/lib/python3.8/asyncio/events.py", line 81, in _run
self._context.run(self._callback, *self._args)
File "/usr/local/lib/python3.8/site-packages/serial_asyncio/__init__.py", line 119, in _read_ready
self._protocol.data_received(data)
File "/usr/local/lib/python3.8/site-packages/dsmr_parser/clients/protocol.py", line 91, in data_received
data = data.decode('ascii')
UnicodeDecodeError: 'ascii' codec can't decode byte 0x80 in position 0: ordinal not in range(128)
```

what is dsmr_parser ?

doudz, 2021-01-14:
it's "Library to parse Dutch Smart Meter Requirements (DSMR)"
nothing to do with zigpy or zigate

HRHL53, 2021-01-14:
Yes, I think you are right. I already disabled DSMR but after unplugging and deleting it, the errors are gone. Thank you for poiting me in the right direction!

closed 2021-01-16 by doudz