zigpy/zigpy-zigate #83: Ikea 5btn remote controller E1524 - problem

pawelka, 2021-01-18:
Hi,

When I'm paired E1524 device, I don't have zha_event when I'm click buttons. I tried to update firmware and it's fail on zero percent.

I check how it's works with cc2531 and zigbee2mqtt and I see mqtt messages for all buttons. I also update device to latest firmware.

When I paired on new firmware with zigate and zha, some switches start reacting on button (same group), so there was some communication in network, but no event.

I checked different projects and tried to find issue related with E1524 and zha, but I didn't find anything use full. 

I'm not sure if it's quirk problem or zigate problem. I have all logs enable and I saw only communication with dongle on main on-off button for other buttons no logs. 

I saw two quirks in handlers project, but I don't understand what condition device needs to meet to select zha version of quirk for this device.

Do you have any idea how to solve this problem?

powerpaul17, 2021-01-29:
Yes I have the same problem with E1810, I already opened an issue and I'm willing to debug but I got not answer yet. It seems like a configuration problem of zigate since it also isn't shown as coordinator but as "unknown" device.

max5962, 2021-02-13:
Same issue here ...
5 buttons or 2 buttons same issue : only battery reporting and no logs or events when button are pressed.

max5962, 2021-02-13:
> Hi,
>
> When I'm paired E1524 device, I don't have zha_event when I'm click buttons. I tried to update firmware and it's fail on zero percent.
>
> I check how it's works with cc2531 and zigbee2mqtt and I [...]

How to you update ikea firmware with ZHA and zigate ?

MattWestb, 2021-02-13:
All IKEA remotes is light controllers  and sending commands to groups of lights.
The problem is that Zigate is not listening on commands sent to gropes https://github.com/zigpy/zigpy-zigate/issues/77.

The devs is looking on the problem but dont have cooking any fix for it yet.

max5962, 2021-02-13:
> All IKEA remotes is light controllers and sending commands to groups of lights.
> The problem is that Zigate is not listening on commands sent to gropes #77.
> 
> The devs is looking on the problem but dont have cooking any fix for it yet.

Ok thanks. So, everything is working is just that zigate is ignoring all events message because they are broadcast to a group that the coordinator do not part of.

I bookmark that issue

MattWestb, 2021-02-13:
Yes or Zigate cant for the moment being member of all groups so it can catching the commands sent to the groups.

Hopefully they is getting it implanted so Zigates is working with all devices as the other radio modules is doing.

If you dont have so much devices you can trying adding all to group 0 if its possible doing with the Zigate (ii dont knowing is its possible but can being worth trying) then its perhaps is working.
One light can being member of more than one groups so group 0 for Zigeate an 2 for the kitchen light and so on.

max5962, 2021-02-13:
> Yes or Zigate cant for the moment being member of all groups so it can catching the commands sent to the groups.
>
> Hopefully they is getting it implanted so Zigates is working with all devices as the [...]

It only let me add outlet as part of group :

[image]

No ikea remote or zigate :-/

pipiche38, 2021-02-13:
ZiGate can be member of 5 groups at max.

MattWestb, 2021-02-13:
Controlling devices you is binding to one group (IKEA cant being bonded to devices only groups).
From the device card clicking "manage cluster" and then group binding.
Select the group you like to binding to the device and the On/Off cluster (its enough for the outlets but for lights also need level control, color and scenes can being needed). Click bind group and waking the remote so its not sleeping.

powerpaul17, 2021-02-14:
> It only let me add outlet as part of group :
>
> [image]
>
> No ikea remote or zigate :-/

Apparently from what I found out in the source, it only lists devices which have the "groups" cluster as input (or output, I always mix them up.. :D ), the remote has only the output cluster so you cannot add it to the group here.

> Controlling devices you is binding to one group (IKEA cant being bonded to devices only groups).
> From the device card clicking "manage cluster" and then group binding.
> Select the group you like to [...]

The problem here is that the zigate has a wrong configuration, it doesn't show any device signature or clusters. @doudz already discovered it (as I did myself) and will investigate. This is also found in #77.

powerpaul17, 2021-02-14:
Other zigbee dongles create some kind of "Default Lighting Group" and add themselves to it so they can receive the group messages, but zigate does not (yet) do it. I found it kind of very difficult to find all this in the sources, I supposed that it would be abstracted more so the devices would work similar despite their hardware differences..

mrIngrid, 2021-06-03:
Hi guys, i have the same problem with 3 ikea devices (E1524, E1743, E1745), on Zigate & Zigbee2mqtt.
The pairing is OK but don't receive anything in action when i press the button.

MattWestb, 2021-06-03:
@mrIngrid I think you need positing  _Zigate & Zigbee2mqtt_ problems in the [Z2M git](https://github.com/Koenkk/zigbee2mqtt).
As is being minted Zigate dont have the support for so many groups and the virtual coordinator is not implanted so its not possible binding the coordinator to one "default group" and getting the command being sent by controlling device like IKEA remotes.
I hope the Zigate devs can fixing sooner then later :-))

pipiche38, 2021-06-03:
As ZiGate is concerned, I don't see your issue. The Ikea Remote works perfectly with non-zigpy based plugin. Here is what you have to do (please translate for zipgy)

1/ pair your Led with the ZiGate
2/ pair your 5B remote with ZiGate 
3/ use the touch link to bind 5B with the LED (stay in the 10cm of the LED)
4/ from the controler request for group membership of the LEd

==> You'll get the group which has been created during the Touch link 

5/ add that groupId to the ZiGate

from that point all Actions on the 5B will be received by the ZiGate and pass through to the above layer. You can even remove the LEd from the group if you want.

As mentioned early, ZiGate can be members of 5 different groups. With ZiGate+ (the new generation) it is up to 15 groups

max5962, 2021-08-28:
> As ZiGate is concerned, I don't see your issue. The Ikea Remote works perfectly with non-zigpy based plugin. Here is what you have to do (please translate for zipgy)
>
> 1/ pair your Led with the [...]

It's very complicated. 
So in order to get it work, with have to add the remote to the same group ? (Remote/zigate/lamps) ?

So If  connecting ZigBee remote with yeelight (with home assistant routine) is not possible ?