zigpy/zigpy-zigate #86: Pizigate on odroid

CinPoU, 2021-04-30:
Could you please add the odroid as a compatible hardware for the pizigate?

Hedda, 2021-06-10:
FYI, ZiGate compatible with different computers is not decided at the zigpy-zigate level at all. ZiGate models compatible with different computers depends on ZiGate hardware and not the zigpy-zigate project. To make this fact a little more clear suggest change readme as per https://github.com/zigpy/zigpy-zigate/issues/86

- [PiZiGate + / PiZiGate (ZiGate HAT/Shield module for Raspberry Pi compatible GPIO header)](https://zigate.fr/produit/pizigatev2/)
   - Tip! PiZiGate are not limited to Raspberry Pi series as works with all computers with a Raspberry Pi compatible GPIO header.

MattWestb, 2021-06-10:
It can being problem then the Pi is using 5 V and normal TTL is using 3.3 V but very likely the HAT cant handle both OK.

Hedda, 2021-06-10:
Obviously the comment "Raspberry Pi compatible GPIO header" also means that it must also offer the same volatage as Raspberry Pi

CinPoU, 2021-06-10:
Hi!

The pizigate is compatible with odroid, i used it with raspbian and jeedom.

But the zigpy-zigate's code check the model of the motherboard, and if it's not a raspberry, there's an error.
So, could you add the odroid motherboard as a compatible device please?

Best regards

CinPoU, 2021-06-10:
in setup.py

`try:
                with open('/proc/cpuinfo', 'r') as cpuinfo:
                    found = False
                    for line in cpuinfo:
                        if line.startswith('Hardware'):
                            found = True
                            label, value = line.strip().split(':', 1)
                            value = value.strip()
                            if value not in (
                                'BCM2708',
                                'BCM2709',
                                'BCM2835',
                                'BCM2836'
                            ):
                                if raise_on_errors:
                                    raise ValueError(
                                        'This system does not appear to be a Raspberry Pi.'
                                    )
                                else:  
                                    return False
`

Hedda, 2021-06-10:
@doudz did you intend that zigpy-zigate should only work with Raspberry Pi? 

If so then it should not either work with OpenLumi ("Open Lumi Gateway" DIY ZiGate WiFi bridge hacked from an Xiaomi Lumi Gateway with modded OpenWRT firmware for the OS and ZiGate firmware on its NXP module for Zigbee), or?

https://openlumi.github.io

https://github.com/openlumi

https://github.com/zigpy/zigpy-zigate/issues/59

Hedda, 2021-06-11:
> in setup.py
>
> `try: with open('/proc/cpuinfo', 'r') as cpuinfo: found = False for line in cpuinfo: if line.startswith('Hardware'): found = True label, value = line.strip().split(':', 1) value = [...]

@doudz  This should maybe not be closed as CinPoU points out that zigpy-zigate setup is currently requiring Raspberry Pi to run?

doudz, 2021-06-11:
The Rpi detection is used to add rpi.gpio requirement dynamically during setup.
Pizigate need it , not other zigate boards.

doudz, 2021-06-11:
Is there a way to check if we are on odroid board ?

doudz, 2021-06-11:
Maybe on odroid we need to install Odroid.GPIO instead of Rpi.GPIO ?