zigpy/zigpy-zigate #89: Zigpy Zigate Wifi - Close connection to ZHA

baimard, 2021-05-16:
Dear,

I use Zigate wifi with ZHA in Home Assistant, after random time (30 min to 4hours), Home Assistant show that : 

```
Logger: zigpy_zigate.api
Source: /usr/local/lib/python3.8/site-packages/zigpy_zigate/api.py:138
First occurred: 19:46:24 (20 occurrences)
Last logged: 19:47:26

No response to command 0x0530
```

After this message, all my sensor light etc. not work with the device.

Home assistant don't send any request to my zigate wifi.

Tcpdump :

```
19:22:09.071920 ARP, Request who-has 192.168.4.227 tell 192.168.4.252, length 46
19:22:09.073167 ARP, Reply 192.168.4.227 is-at 3c:61:05:d1:1f:d6, length 46
19:25:59.740061 IP 192.168.4.227.9999 > 192.168.4.252.54258: Flags [P.], seq 2250:2330, ack 860, win 1824, length 80
19:25:59.740679 IP 192.168.4.252.54258 > 192.168.4.227.9999: Flags [.], ack 2330, win 64068, length 0
19:25:59.741956 IP 192.168.4.227.9999 > 192.168.4.252.54258: Flags [P.], seq 2330:2340, ack 860, win 1824, length 10
19:25:59.743470 IP 192.168.4.252.54258 > 192.168.4.227.9999: Flags [.], ack 2340, win 64068, length 0
19:26:04.848226 ARP, Request who-has 192.168.4.227 tell 192.168.4.252, length 46
19:26:04.850873 ARP, Reply 192.168.4.227 is-at 3c:61:05:d1:1f:d6, length 46
19:30:59.877528 IP 192.168.4.227.9999 > 192.168.4.252.54258: Flags [F.], seq 2340, ack 860, win 1824, length 0
19:30:59.879422 IP 192.168.4.252.54258 > 192.168.4.227.9999: Flags [F.], seq 860, ack 2341, win 64068, length 0
19:30:59.880991 IP 192.168.4.227.9999 > 192.168.4.252.54258: Flags [.], ack 861, win 1823, length 0
19:31:05.139535 ARP, Request who-has 192.168.4.227 tell 192.168.4.252, length 46
19:31:05.141093 ARP, Reply 192.168.4.227 is-at 3c:61:05:d1:1f:d6, length 46
19:37:21.514434 ARP, Request who-has 192.168.4.227 tell 192.168.4.1, length 46
```

So two questions : 

- Why zigate Wifi close the connection ? 
- Is it possible to make ZHA restart the connection ?

Thank you

totof129, 2021-05-17:
Hi,

your bug seems to be similar to issue #55 
Did you upgrade Zigate's firmware ? 
A new release (3.1e) has just been delivered

baimard, 2021-05-17:
Do you know how I could watch my firmware version of zigate ?

totof129, 2021-05-17:
you should be able to see it in the visualization tab in ZHA configuation

baimard, 2021-05-17:
3.1d for me, new question ... where is the last firmware ...

totof129, 2021-05-17:
https://github.com/fairecasoimeme/ZiGate/releases

baimard, 2021-05-17:
Merci beaucoup !

baimard, 2021-05-18:
another bug : 

```
home_assistant.1.mag30mglug26@x-kaki    | 2021-05-18 22:55:43 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
home_assistant.1.mag30mglug26@x-kaki    | 2021-05-18 22:55:43 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
home_assistant.1.mag30mglug26@x-kaki    | 2021-05-18 22:55:44 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
home_assistant.1.mag30mglug26@x-kaki    | 2021-05-18 22:55:44 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
home_assistant.1.mag30mglug26@x-kaki    | 2021-05-18 22:55:44 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
home_assistant.1.mag30mglug26@x-kaki    | 2021-05-18 22:55:44 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
```
Always same deconnection ...

baimard, 2021-05-18:
```
home_assistant.1.k28tajqju8bv@x-kaki    | 2021-05-18 23:25:57 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=light, service=turn_off, service_data=entity_id=light.beta_level_on_off>
home_assistant.1.k28tajqju8bv@x-kaki    | 2021-05-18 23:25:57 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x5231, 260, 6, 1, 1, 18, b'\x01\x12\x00', True, False)
home_assistant.1.k28tajqju8bv@x-kaki    | 2021-05-18 23:25:57 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'025231010100060104000003011200'
home_assistant.1.k28tajqju8bv@x-kaki    | 2021-05-18 23:25:57 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x0fH\x02R1\x01\x01\x00\x06\x01\x04\x00\x00\x03\x01\x12\x00'
home_assistant.1.k28tajqju8bv@x-kaki    | 2021-05-18 23:25:57 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x02\x1fH\x02\x12R1\x02\x11\x02\x11\x02\x10\x02\x16\x02\x11\x02\x14\x02\x10\x02\x10\x02\x13\x02\x11\x12\x02\x10'
home_assistant.1.k28tajqju8bv@x-kaki    | 2021-05-18 23:25:58 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
home_assistant.1.k28tajqju8bv@x-kaki    | 2021-05-18 23:25:58 WARNING (MainThread) [zigpy_zigate.api] Retry command 0x0530
home_assistant.1.k28tajqju8bv@x-kaki    | 2021-05-18 23:25:58 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'025231010100060104000003011200'
home_assistant.1.k28tajqju8bv@x-kaki    | 2021-05-18 23:25:58 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x0fH\x02R1\x01\x01\x00\x06\x01\x04\x00\x00\x03\x01\x12\x00'
home_assistant.1.k28tajqju8bv@x-kaki    | 2021-05-18 23:25:58 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x02\x1fH\x02\x12R1\x02\x11\x02\x11\x02\x10\x02\x16\x02\x11\x02\x14\x02\x10\x02\x10\x02\x13\x02\x11\x12\x02\x10'
home_assistant.1.k28tajqju8bv@x-kaki    | 2021-05-18 23:26:00 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
home_assistant.1.k28tajqju8bv@x-kaki    | 2021-05-18 23:26:00 DEBUG (MainThread) [zigpy.device] [0x5231] Delivery error for seq # 0x12, on endpoint id 1 cluster 0x0006: ZiGate doesn't answer to command
home_assistant.1.k28tajqju8bv@x-kaki    | 2021-05-18 23:26:00 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5231:1:0x0006]: command failed: 'off' args: '()' kwargs '{}' exception: '[0x5231:1:0x0006]: Message send failure'
home_assistant.1.k28tajqju8bv@x-kaki    | 2021-05-18 23:26:00 DEBUG (MainThread) [homeassistant.components.zha.entity] light.beta_level_on_off: turned off: [0x5231:1:0x0006]: Message send failure

```

letroll, 2021-06-14:
Same here, I can also help with debug if you need information

tygerlord, 2021-06-14:
I use firmware 3.1e
Yesterday all is working. This morning: devices Offlines

Alindil, 2021-06-14:
Hi, 
I have the same problem.
There is a query that is disconnected and no longer makes the plugin work. A little try catch would at least prevent the plugin from crashing

tygerlord, 2021-06-14:
```
2021-06-13 21:04:43 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530                                                               ┤
│2021-06-13 21:04:43 WARNING (MainThread) [zigpy_zigate.api] Retry command 0x0530                                                                        ┤
│2021-06-13 21:04:43 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'02a6e10101000001040000050057000400'                                           ┤
│2021-06-13 21:04:43 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x112\x02\xa6\xe1\x01\x01\x00\x00\x01\x04\x00\x00\x05\x00W\x00\x04\┤
│2021-06-13 21:04:43 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x112\x02\x12\xa6\xe1\x02\x11\x02\x11\x02\x10\x02\x10\x02\x┤
│2021-06-13 21:04:44 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530                                                               ┤
│2021-06-13 21:04:44 DEBUG (MainThread) [zigpy.device] [0xa6e1] Delivery error for seq # 0x57, on endpoint id 1 cluster 0x0000: ZiGate doesn't answer to command│
```

ZeGuigui, 2021-06-14:
I just performed a test against the code in the dev branch which is supposed to implement the reconnect feature.
```
2021-06-14 14:51:37 DEBUG (MainThread) [zigpy_zigate.uart] Closed serial connection
```
but no reconnection attempt was made by the lib. I had to force integration reload to have the connection back:
```
2021-06-14 14:55:06 DEBUG (MainThread) [zigpy_zigate.uart] Connection made
```

ZeGuigui, 2021-06-14:
I added some debug code into the lib, just to be sure.
```
    def connection_lost(self, exc) -> None:
        """Port was closed expecteddly or unexpectedly."""
        if self._connected_future and not self._connected_future.done():
            if exc is None:
                LOGGER.info("Connection lost - no exception");
                self._connected_future.set_result(True)
            else:
                LOGGER.info("Connection lost with exception %s", exc);
                self._connected_future.set_exception(exc)
        if exc is None:
            LOGGER.debug("Closed serial connection")
            return

        LOGGER.error("Lost serial connection: %s", exc)
        self._api.connection_lost(exc)
```

The "connection lost" messages I added are never called when we hit a disconnect. We might need to examine zigpy implementation in details to understand if this code (which is common to xbee, combee...) do work with other devices or if there is just a duplicated bug. Connection lost with USB devices is probably a rare issue

tygerlord, 2021-06-16:
Like it seems to come  after 'No response to command 0x0530', that is to say, a 'hardware' crash, this is probably common to xbee, combee and other but they don't crash so we don't see the same trouble. 

I think we are dealing with 2 bugs. To check.

baimard, 2021-06-16:
> Like it seems to come after 'No response to command 0x0530', that is to say, a 'hardware' crash, this is probably common to xbee, combee and other but they don't crash so we don't see the same trouble.
> 
> I think we are dealing with 2 bugs. To check.

For me it's a global connection problem. After a delay, connection crash and we have lot of errors.

Depending on the command that we send we have a different return. But the basic problem is the connection

pipiche38, 2021-06-16:
My 2 cents, don't know if this is related, but in Wifi, the plugin needs to maintain the IP connection, otherwise after a Timeout ( 4 or 5 minutes) the line is droped

baimard, 2021-06-16:
is it possible to add simply a "ping" or other every 30s to maintain the connection ?

baimard, 2021-06-16:
> > > Like it seems to come after 'No response to command 0x0530', that is to say, a 'hardware' crash, this is probably common to xbee, combee and other but they don't crash so we don't see the same trouble.
> > > I think we are dealing with 2 bugs. To check.
> > 
> > 
> > For me it's a global connection problem. After a delay, connection crash and we have lot of errors.
> > Depending on the command that we send we have a different return. But the basic problem is the connection
> 
> Do other Zigbee radio modules have the same troubles ?

that a very good question, with the title of this issue I don't know if other product owner read this thread

ZeGuigui, 2021-06-16:
Most other zigbee radio modules are USB modules

letroll, 2021-06-16:
I can confirm I'm affected with zigate on usb. For now I have rollback to HASS version 2021.3.0 (docker install)

MattWestb, 2021-06-16:
deCONZ radios have much problems with USB disconnections and comports is not mapped OK in docker (in native deCONZ and ZHA).

Tasmota, and ESPHome that using ESP chips looks working OK with EFR32 and TI chips also tuya ZBGW is working OK with EFR32 radios over Ethernet.

EFR32 / bellows is going in fault state if the counter is going of sync then loosing packages in the WiFi / Eth network but is nicely logged in the error log.

tygerlord, 2021-06-16:
> 
> 
> I can confirm I'm affected with zigate on usb. For now I have rollback to HASS version 2021.3.0 (docker install)

Interesting, can you confirm that you don't use any wifi between zigate/usb and your home assistant ?

letroll, 2021-06-16:
I confirm. I'm using Hass with docker on a NUC running archlinux. I don't know if it can influence but my USB Zigate is on a USB hub. I can try without tonight if you want?

tygerlord, 2021-06-16:
> 
> 
> I confirm. I'm using Hass with docker on a NUC running archlinux. I don't know if it can influence but my USB Zigate is on a USB hub. I can try without tonight if you want?

Thank you, I don't think that it's necessary. The main point is that wifi seem not be the cause. As far as I'm remember when I put my PiZigate directly on my raspberry pi3, I already have 'Devices Offline' disconnects on my zigbee2mqtt, so this seems to exonerate wifi

tygerlord, 2021-06-16:
> 
> 
> is it possible to add simply a "ping" or other every 30s to maintain the connection ?

I need to check logs, but I think that plugin already ask periodically for state of sensors...

tygerlord, 2021-06-16:
> 
> 
> ```
> 2021-06-13 21:04:43 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530                                                               ┤
> │2021-06-13 21:04:43 WARNING (MainThread) [zigpy_zigate.api] Retry command 0x0530                                                                        ┤
> │2021-06-13 21:04:43 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'02a6e10101000001040000050057000400'                                           ┤
> │2021-06-13 21:04:43 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x112\x02\xa6\xe1\x01\x01\x00\x00\x01\x04\x00\x00\x05\x00W\x00\x04\┤
> │2021-06-13 21:04:43 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x112\x02\x12\xa6\xe1\x02\x11\x02\x11\x02\x10\x02\x10\x02\x┤
> │2021-06-13 21:04:44 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530                                                               ┤
> │2021-06-13 21:04:44 DEBUG (MainThread) [zigpy.device] [0xa6e1] Delivery error for seq # 0x57, on endpoint id 1 cluster 0x0000: ZiGate doesn't answer to command│
> ```
> 
> And after nothing

To my mind, after Retry command, if module don't answer again then we should initialize connection like we do on start. What do you think about?

tygerlord, 2021-06-16:
```
Citation de pipiche le 16 juin 2021, 15 h 27 min
Domoticz ping régulièrement la ZiGate (si il n'y a pas eu de communication In et Out depuis un laps de temps)
Si on ne reçoit pas de réponse au ping, on va insister
Si malgré les retry pas de réponse, alors le plugin est relancé avec un restart de la ZiGate
```

This seems a good process.

pipiche38, 2021-06-16:
> ```
> Citation de pipiche le 16 juin 2021, 15 h 27 min
> Domoticz ping régulièrement la ZiGate (si il n'y a pas eu de communication In et Out depuis un laps de temps)
> Si on ne reçoit pas de réponse au [...]
To be complet, Wifi module (as per their documentation and specifications) has a Timeout and this is up to the client to keep the line live!

baimard, 2021-06-16:
So the problem is in HASS core ?

tygerlord, 2021-06-16:
>

Yes, but as I can see in logs, we have a 0x0530 exchange every minute. (more over this seem to happen with usb module too that are not concern about wifi timeout)

tygerlord, 2021-06-16:
> 
> 
> So the problem is in HASS core ?

At least the core can be improved by adding same process, retry and restart if failure stay.

doudz, 2021-07-19:
not yet, sorry for that

letroll, 2021-07-27:
By the past, I wasn't experiment these disconnection. Anyone know how I can rollback HASS before the changes in causes?

baimard, 2021-11-07:
I have solved the problem temporary by changing this function : 

```
 async def command(self, cmd, data=b'', wait_response=None, wait_status=True):
        tries = 2
        while tries > 0:
            tries -= 1
            await self._lock.acquire()
            try:
                return await asyncio.wait_for(
                    self._command(cmd, data, wait_response, wait_status),
                    timeout=COMMAND_TIMEOUT
                )
            except asyncio.TimeoutError:
                LOGGER.warning("No response to command 0x%04x", cmd)
                if tries > 0:
                    LOGGER.warning("Retry command 0x%04x", cmd)
                else:
                    raise NoResponseError
            finally:
                self._lock.release()
```

by :

```
 async def command(self, cmd, data=b'', wait_response=None, wait_status=True):
        tries = 2
        while tries > 0:
            tries -= 1
            await self._lock.acquire()
            try:
                return await asyncio.wait_for(
                    self._command(cmd, data, wait_response, wait_status),
                    timeout=COMMAND_TIMEOUT
                )
            except asyncio.TimeoutError:
                LOGGER.warning("No response to command 0x%04x", cmd)
                if tries > 0:
                    LOGGER.warning("Retry command 0x%04x", cmd)
                    LOGGER.warning("Close uart")
                    self.close()
                    LOGGER.warning("connect uart")
                    await self.connect()
                else:
                    raise NoResponseError
            finally:
                self._lock.release()
```

Finally, the update of the zigpy_zigae does not have update in HASS.

jquievreux, 2021-11-09:
@baimard thanks for sharing, 
on battery powered sensor too? 
Which firmware is your zigate on?

baimard, 2021-11-09:
ZiGate-WiFi v2.00 https://github.com/fairecasoimeme/ZiGate-WiFi/releases/tag/v2.00

Zigate 3.20 https://github.com/fairecasoimeme/ZiGate/releases

I try to check the connection every minute but I don't know how to deal with the asyncio librairie.

In fact, I don't which command I can send to the ZigateWifi to check connectivity. :(

Because actually my modification work for every hardware interrogated by HASS, but if it's the hardware who need to send data without request (opening door sensor), there is a possibility that the connection is down and that I don't have the information.

baimard, 2021-11-10:
> To keep the line active you can use some zigate command like 0x0014 get status 0x0010 get zigate version Il faut une tache que s’assure qu’il y ait eu au moins un échange dans les 4’ sinon faire une commande pour garder la liaison TCP/IP’s-wifi Envoyé de mon iPhone
> […](#)

Merci beaucoup, je vais lancer un thread qui va tourner en permanence pour vérifier l'état de la connexion merci !

pipiche38, 2021-11-10:
Absolument, l'idéal, c'est de regarder si il y a du trafic dans la fenetre des 4 minutes et dans ce cas ne rien faire, sinon envoyer une commande.

baimard, 2021-11-10:
Pour faire simple un simple get_status toutes les 4 minutes permettra de 
détecter la connexion, sinon je vais devoir implémenter des traceurs 
d'action partout ... ou un loggeur

Cordialement,
Benjamin AIMARD

baimard, 2021-11-16:
so I do that simply ...

```
    async def connect(self):
        assert self._uart is None
        self._uart = await zigpy_zigate.uart.connect(self._config, self)
        self._thread = threading.Thread(target=self.loop_check_connection) #89
        self._thread.start() #89
```

```
    def loop_check_connection(self):
        while True:
            time.sleep(300)
            LOGGER.warning("Check ---")
            try:
                asyncio.run(self.get_network_state())
            except:
                LOGGER.warning("thread error") #Because of concurrent access
```

I have no more error ... : 

logs
```
home_assistant.1.01xbc0atho3i@ocre    | 2021-11-16 10:20:18 WARNING (Thread-4) [zigpy_zigate.api] Check ---
home_assistant.1.01xbc0atho3i@ocre    | 2021-11-16 10:20:18 WARNING (MainThread) [zigpy_zigate.api] data_received
home_assistant.1.01xbc0atho3i@ocre    | 2021-11-16 10:20:18 WARNING (MainThread) [zigpy_zigate.api] data_received
home_assistant.1.01xbc0atho3i@ocre    | 2021-11-16 10:21:50 WARNING (MainThread) [zigpy_zigate.api] data_received
home_assistant.1.01xbc0atho3i@ocre    | 2021-11-16 10:23:02 WARNING (MainThread) [zigpy_zigate.api] data_received
home_assistant.1.01xbc0atho3i@ocre    | 2021-11-16 10:23:02 WARNING (MainThread) [zigpy_zigate.api] data_received
home_assistant.1.01xbc0atho3i@ocre    | 2021-11-16 10:23:02 WARNING (MainThread) [zigpy_zigate.api] data_received
home_assistant.1.01xbc0atho3i@ocre    | 2021-11-16 10:23:02 WARNING (MainThread) [zigpy_zigate.api] data_received
home_assistant.1.01xbc0atho3i@ocre    | 2021-11-16 10:23:02 WARNING (MainThread) [zigpy_zigate.api] data_received
home_assistant.1.01xbc0atho3i@ocre    | 2021-11-16 10:23:02 WARNING (MainThread) [zigpy_zigate.api] data_received
home_assistant.1.01xbc0atho3i@ocre    | 2021-11-16 10:23:02 WARNING (MainThread) [zigpy_zigate.api] data_received
home_assistant.1.01xbc0atho3i@ocre    | 2021-11-16 10:23:02 WARNING (MainThread) [zigpy_zigate.api] data_received
```

Entyre file : 

```
import asyncio
import binascii
import functools
import logging
import enum
import datetime
from typing import Any, Dict

import serial
import zigpy.exceptions

import zigpy_zigate.config
import zigpy_zigate.uart

from . import types as t

import time
import threading

LOGGER = logging.getLogger(__name__)

COMMAND_TIMEOUT = 1.5
PROBE_TIMEOUT = 3.0

RESPONSES = {
    0x004D: (t.NWK, t.EUI64, t.uint8_t, t.uint8_t),
    0x8000: (t.uint8_t, t.uint8_t, t.uint16_t, t.Bytes),
    0x8002: (t.uint8_t, t.uint16_t, t.uint16_t, t.uint8_t, t.uint8_t,
[... 250 lines cut ...]
            asyncio.TimeoutError,
            serial.SerialException,
            zigpy.exceptions.ZigbeeException,
        ) as exc:
            LOGGER.debug(
                "Unsuccessful radio probe of '%s' port",
                device_config[zigpy_zigate.config.CONF_DEVICE_PATH],
                exc_info=exc,
            )
        finally:
            api.close()

        return False

    async def _probe(self) -> None:
        """Open port and try sending a command"""
        try:
            device = next(serial.tools.list_ports.grep(self._config[zigpy_zigate.config.CONF_DEVICE_PATH]))
            if device.description == 'ZiGate':
                return
        except StopIteration:
            pass
        await self.connect()
        await self.set_raw_mode()
```

pipiche38, 2021-12-12:
Why are you not submitting a PR ? so your patch would be merge into the code

baimard, 2021-12-12:
Because I've tested, and I have an other error I need to solve that before the PR.

After 24 h of usage I have this : the thread goes into an error, and I don't know why :(

```
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:07:15 ERROR (Thread-4) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:12:11 ERROR (Thread-566) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:12:11 ERROR (Thread-559) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:12:13 WARNING (Thread-554) [zigpy_zigate.api] No response to command 0x0009
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:12:13 WARNING (Thread-554) [zigpy_zigate.api] Retry command 0x0009
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:12:13 WARNING (Thread-554) [zigpy_zigate.api] Close uart
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:12:13 ERROR (Thread-554) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:12:15 WARNING (Thread-571) [zigpy_zigate.api] No response to command 0x0009
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:12:15 WARNING (Thread-571) [zigpy_zigate.api] Retry command 0x0009
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:12:15 WARNING (Thread-571) [zigpy_zigate.api] Close uart
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:12:15 WARNING (Thread-571) [zigpy_zigate.api] connect uart
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:12:15 ERROR (Thread-4) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:12:16 WARNING (Thread-571) [zigpy_zigate.api] No response to command 0x0009
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:12:16 ERROR (Thread-571) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:13:13 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:13:13 WARNING (MainThread) [zigpy_zigate.api] Retry command 0x0530
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:13:13 WARNING (MainThread) [zigpy_zigate.api] Close uart
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:13:13 WARNING (MainThread) [zigpy_zigate.api] connect uart
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:13:15 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:13:15 WARNING (MainThread) [zigpy_zigate.api] Retry command 0x0530
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:13:15 WARNING (MainThread) [zigpy_zigate.api] Close uart
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:13:15 WARNING (MainThread) [zigpy_zigate.api] connect uart
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:13:17 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:13:18 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:16:25 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:16:25 WARNING (MainThread) [zigpy_zigate.api] Retry command 0x0530
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:16:25 WARNING (MainThread) [zigpy_zigate.api] Close uart
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:16:25 WARNING (MainThread) [zigpy_zigate.api] connect uart
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:17:12 ERROR (Thread-559) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:17:13 ERROR (Thread-554) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:17:15 ERROR (Thread-577) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:18:15 ERROR (Thread-580) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:22:13 ERROR (Thread-554) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:22:13 ERROR (Thread-566) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:22:16 ERROR (Thread-4) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:23:15 ERROR (Thread-579) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:27:13 ERROR (Thread-554) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:27:13 ERROR (Thread-559) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:27:16 ERROR (Thread-577) [zigpy_zigate.api] thread error
home_assistant.1.5ktr66b42u1p@ocre    | 2021-12-11 20:28:17 ERROR (Thread-580) [zigpy_zigate.api] thread error

puddly, 2021-12-13:
You don't need to use threads when working with asyncio, you can use tasks:

```Python
async def connect(self):
    ...
    self._watchdog_task = asyncio.create_task(self.watchdog_loop())

def close(self):
    ...
    self._watchdog_task.cancel()

async def watchdog_loop(self):
    while True:
        await asyncio.sleep(30)
        LOGGER.warning("Check ---")

        try:
            await self.get_network_state()
        except Exception as exc:
            LOGGER.warning("Failed to get network state, treating as disconnect", exc_info=exc)

            self.connection_lost(exc)  # Treat this is a disconnect
            break
```

I don't have a ZiGate to test but this is roughly what other radio libraries do.

baimard, 2021-12-13:
I don't know why I can't deal with asyncio, but I'm gone try again with your sample code. Thank you.

baimard, 2022-03-16:
I tried updating to 0.8.0, it just doesn't work, my solution still works with the 0.7.3 package. I downgrade everything to HASS...

Very sad situation!

puddly, 2022-03-16:
@baimard What does the debug log show with 0.8.0? https://www.home-assistant.io/integrations/zha/#debug-logging

baimard, 2022-03-16:
```
home_assistant.1.iayelvp8clwa@home3    | 2022-03-16 20:52:17 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code 133
home_assistant.1.iayelvp8clwa@home3    | 2022-03-16 20:52:17 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code 133
home_assistant.1.iayelvp8clwa@home3    | 2022-03-16 20:52:17 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code 133
home_assistant.1.iayelvp8clwa@home3    | 2022-03-16 20:52:17 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code 131
home_assistant.1.iayelvp8clwa@home3    | 2022-03-16 20:52:17 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code 131
home_assistant.1.iayelvp8clwa@home3    | 2022-03-16 20:52:17 WARNING (MainThread) [homeassistant.components.zha.core.channels.base] [0x1500:1:0x0001]: async_initialize: all attempts have failed: [DeliveryError('[0x1500:1:0x0001]: Message send failure'), DeliveryError('[0x1500:1:0x0001]: Message send failure'), DeliveryError('[0x1500:1:0x0001]: Message send failure'), DeliveryError('[0x1500:1:0x0001]: Message send failure')]
home_assistant.1.iayelvp8clwa@home3    | 2022-03-16 20:52:18 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code 133
home_assistant.1.iayelvp8clwa@home3    | 2022-03-16 20:52:18 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code 131
home_assistant.1.iayelvp8clwa@home3    | 2022-03-16 20:52:18 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code 131
home_assistant.1.iayelvp8clwa@home3    | 2022-03-16 20:52:18 WARNING (MainThread) [homeassistant.components.zha.core.channels.base] [0x1500:1:0x0405]: async_initialize: all attempts have failed: [DeliveryError('[0x1500:1:0x0405]: Message send failure'), DeliveryError('[0x1500:1:0x0405]: Message send failure'), DeliveryError('[0x1500:1:0x0405]: Message send failure'), DeliveryError('[0x1500:1:0x0405]: Message send failure')]

```

you need more ?

puddly, 2022-03-16:
The whole debug log, not just the warnings.

baimard, 2022-03-16:
ok I will show you tomorow.

baimard, 2022-03-28:
Sorry for delay, I have new born at home :)

```
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:00:32 INFO (MainThread) [homeassistant.components.camera] Setting up camera.netatmo
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:00:32 INFO (MainThread) [homeassistant.components.climate] Setting up climate.netatmo
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:00:32 INFO (MainThread) [homeassistant.components.light] Setting up light.netatmo
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:00:32 INFO (MainThread) [homeassistant.components.select] Setting up select.netatmo
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:14 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.netatmo_chambre_co2, old_state=<state sensor.netatmo_chambre_co2=1299; state_class=measurement, attribution=Data provided by Netatmo, unit_of_measurement=ppm, device_class=carbon_dioxide, friendly_name=Netatmo Chambre CO2 @ 2022-03-28T10:51:14.106610+02:00>, new_state=<state sensor.netatmo_chambre_co2=1265; state_class=measurement, attribution=Data provided by Netatmo, unit_of_measurement=ppm, device_class=carbon_dioxide, friendly_name=Netatmo Chambre CO2 @ 2022-03-28T11:01:14.181534+02:00>>
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:14 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.netatmo_chambre_pressure, old_state=<state sensor.netatmo_chambre_pressure=1018.6; state_class=measurement, attribution=Data provided by Netatmo, unit_of_measurement=mbar, device_class=pressure, friendly_name=Netatmo Chambre Pressure @ 2022-03-28T10:45:14.113598+02:00>, new_state=<state sensor.netatmo_chambre_pressure=1018.5; state_class=measurement, attribution=Data provided by Netatmo, unit_of_measurement=mbar, device_class=pressure, friendly_name=Netatmo Chambre Pressure @ 2022-03-28T11:01:14.182445+02:00>>
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:14 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.netatmo_chambre_noise, old_state=<state sensor.netatmo_chambre_noise=40; state_class=measurement, attribution=Data provided by Netatmo, unit_of_measurement=dB, icon=mdi:volume-high, friendly_name=Netatmo Chambre Noise @ 2022-03-28T10:51:14.107684+02:00>, new_state=<state sensor.netatmo_chambre_noise=38; state_class=measurement, attribution=Data provided by Netatmo, unit_of_measurement=dB, icon=mdi:volume-high, friendly_name=Netatmo Chambre Noise @ 2022-03-28T11:01:14.183131+02:00>>
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:28 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xFB80](TRADFRIbulbE14WWclear250lm): Attempting to checkin with device - missed checkins: 1
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:28 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0xFB80, 260, 0, 1, 1, 33, b'\x00!\x00\x04\x00', True, False)
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:28 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'02fb800101000001040000050021000400'
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:28 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x11x\x02\xfb\x80\x01\x01\x00\x00\x01\x04\x00\x00\x05\x00!\x00\x04\x00'
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:28 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x11x\x02\x12\xfb\x80\x02\x11\x02\x11\x02\x10\x02\x10\x02\x11\x02\x14\x02\x10\x02\x10\x02\x15\x02\x10!\x02\x10\x02\x14\x02\x10'
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:28 WARNING (MainThread) [asyncio] socket.send() raised exception.
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:28 DEBUG (MainThread) [zigpy_zigate.api] Wait for status to command 0x0530
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:30 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:30 DEBUG (MainThread) [zigpy_zigate.api] Tries count 2
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:30 WARNING (MainThread) [zigpy_zigate.api] Retry command 0x0530
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:30 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'02fb800101000001040000050021000400'
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x11x\x02\xfb\x80\x01\x01\x00\x00\x01\x04\x00\x00\x05\x00!\x00\x04\x00'
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x11x\x02\x12\xfb\x80\x02\x11\x02\x11\x02\x10\x02\x10\x02\x11\x02\x14\x02\x10\x02\x10\x02\x15\x02\x10!\x02\x10\x02\x14\x02\x10'
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:30 WARNING (MainThread) [asyncio] socket.send() raised exception.
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:30 DEBUG (MainThread) [zigpy_zigate.api] Wait for status to command 0x0530
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:31 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:31 DEBUG (MainThread) [zigpy_zigate.api] Tries count 1
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:01:31 WARNING (MainThread) [zigpy_zigate.api] Retry command 0x0530
[... cut ...]
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 11:03:08 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.fp3_battery_temperature, old_state=<state sensor.fp3_battery_temperature=30.2; state_class=measurement, unit_of_measurement=°C, device_class=temperature, icon=mdi:battery, friendly_name=FP3 Battery Temperature @ 2022-03-28T10:59:34.856810+02:00>, new_state=<state sensor.fp3_battery_temperature=29.7; state_class=measurement, unit_of_measurement=°C, device_class=temperature, icon=mdi:battery, friendly_name=FP3 Battery Temperature @ 2022-03-28T11:03:08.254379+02:00>>

baimard, 2022-03-28:
As you can see, after changing my ZHA librairy I can't retrieve temperature information.

[image]

baimard, 2022-03-28:
After a while

```
t_home; source_type=gps, latitude=44.8671356, longitude=-0.7290534, gps_accuracy=87, altitude=82, course=0, speed=0, vertical_accuracy=14, friendly_name=FP3 @ 2022-03-28T16:27:29.701762+02:00>, new_state=<state device_tracker.fp3_2=not_home; source_type=gps, latitude=44.8871812, longitude=-0.7411826, gps_accuracy=19, altitude=82, course=0, speed=0, vertical_accuracy=3, friendly_name=FP3 @ 2022-03-28T16:30:32.719032+02:00>>
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:30:32 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=person.benjamin, old_state=<state person.benjamin=not_home; editable=False, id=benjamin, latitude=44.8671356, longitude=-0.7290534, gps_accuracy=87, source=device_tracker.fp3_2, user_id=8274e2fea1344dc795b63f9d4b49db97, entity_picture=/api/image/serve/47ffc39db49252b0e9cf22e30e8baddc/512x512, friendly_name=Benjamin @ 2022-03-28T13:42:57.492469+02:00>, new_state=<state person.benjamin=not_home; editable=False, id=benjamin, latitude=44.8871812, longitude=-0.7411826, gps_accuracy=19, source=device_tracker.fp3_2, user_id=8274e2fea1344dc795b63f9d4b49db97, entity_picture=/api/image/serve/47ffc39db49252b0e9cf22e30e8baddc/512x512, friendly_name=Benjamin @ 2022-03-28T13:42:57.492469+02:00>>
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:29 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x1500](TH01): Attempting to checkin with device - missed checkins: 1
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:29 DEBUG (MainThread) [zigpy.device] [0x1500] Extending timeout for 0x32 request
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:29 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x1500, 260, 0, 1, 1, 50, b'\x002\x00\x04\x00', True, False)
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:29 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'0215000101000001040000050032000400'
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:29 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x11\x05\x02\x15\x00\x01\x01\x00\x00\x01\x04\x00\x00\x05\x002\x00\x04\x00'
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:29 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x11\x02\x15\x02\x12\x15\x02\x10\x02\x11\x02\x11\x02\x10\x02\x10\x02\x11\x02\x14\x02\x10\x02\x10\x02\x15\x02\x102\x02\x10\x02\x14\x02\x10'
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:29 WARNING (MainThread) [asyncio] socket.send() raised exception.
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:29 DEBUG (MainThread) [zigpy_zigate.api] Wait for status to command 0x0530
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:30 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:30 DEBUG (MainThread) [zigpy_zigate.api] Tries count 2
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:30 WARNING (MainThread) [zigpy_zigate.api] Retry command 0x0530
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:30 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'0215000101000001040000050032000400'
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x11\x05\x02\x15\x00\x01\x01\x00\x00\x01\x04\x00\x00\x05\x002\x00\x04\x00'
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:30 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x11\x02\x15\x02\x12\x15\x02\x10\x02\x11\x02\x11\x02\x10\x02\x10\x02\x11\x02\x14\x02\x10\x02\x10\x02\x15\x02\x102\x02\x10\x02\x14\x02\x10'
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:30 WARNING (MainThread) [asyncio] socket.send() raised exception.
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:30 DEBUG (MainThread) [zigpy_zigate.api] Wait for status to command 0x0530
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:32 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:32 DEBUG (MainThread) [zigpy_zigate.api] Tries count 1
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:32 WARNING (MainThread) [zigpy_zigate.api] Retry command 0x0530
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:32 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'0215000101000001040000050032000400'
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:32 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x11\x05\x02\x15\x00\x01\x01\x00\x00\x01\x04\x00\x00\x05\x002\x00\x04\x00'
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:32 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x11\x02\x15\x02\x12\x15\x02\x10\x02\x11\x02\x11\x02\x10\x02\x10\x02\x11\x02\x14\x02\x10\x02\x10\x02\x15\x02\x102\x02\x10\x02\x14\x02\x10'
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:32 WARNING (MainThread) [asyncio] socket.send() raised exception.
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:32 DEBUG (MainThread) [zigpy_zigate.api] Wait for status to command 0x0530
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:33 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:31:33 DEBUG (MainThread) [zigpy_zigate.api] Tries count 0
[... cut ...]
home_assistant.1.tuok51suk2y3@home3    | 2022-03-28 16:36:30 DEBUG (MainThread) [homeassistant.core] Bus:Handling **<Event state_changed[L]: entity_id=button.ewelink_th01_e1496922_identify, old_state=<state button.ewelink_th01_e1496922_identify=unknown; device_class=update, friendly_name=eWeLink TH01 e1496922 identify @ 2022-03-28T09:35:29.423165+02:00>, new_state=<state button.ewelink_th01_e1496922_identify=unavailable; device_class=update, friendly_name=eWeLink TH01 e1496922 identify @ 2022-03-28T16:36:30.100990+02:00>>**

max5962, 2022-04-13:
Same issue.
I have to unplug/plug and reboot my VM twice a day :-(

puddly, 2022-07-12:
What version of Home Assistant are you running? Can post post debug logs?

Rudd-O, 2022-07-12:
Absolutely, what logs do you want?

I'm on the latest version of HA on top of Fedora 36 ARM64.  I'm using the serial UART ZiGate, not the Wi-Fi one.

Rudd-O, 2022-07-12:
Some warning logs:

```
Jul 12 17:43:14 verena.dragonfear systemd[1]: systemd-unit-state-collector.service: Consumed 1.481s CPU time.
Jul 12 17:43:18 verena.dragonfear hass[873]: 2022-07-12 17:43:18 INFO (SyncWorker_5) [homeassistant.components.ubus.device_tracker] Checking hostapd
Jul 12 17:43:28 verena.dragonfear hass[873]: 2022-07-12 17:43:28 INFO (SyncWorker_7) [homeassistant.components.ubus.device_tracker] Checking hostapd
Jul 12 17:43:38 verena.dragonfear hass[873]: 2022-07-12 17:43:38 INFO (SyncWorker_8) [homeassistant.components.ubus.device_tracker] Checking hostapd
Jul 12 17:43:43 verena.dragonfear hass[873]: 2022-07-12 17:43:43 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:43 verena.dragonfear hass[873]: 2022-07-12 17:43:43 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:43 verena.dragonfear hass[873]: 2022-07-12 17:43:43 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:43 verena.dragonfear hass[873]: 2022-07-12 17:43:43 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:43 verena.dragonfear hass[873]: 2022-07-12 17:43:43 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:43 verena.dragonfear hass[873]: 2022-07-12 17:43:43 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:43 verena.dragonfear hass[873]: 2022-07-12 17:43:43 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:44 verena.dragonfear hass[873]: 2022-07-12 17:43:44 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:44 verena.dragonfear hass[873]: 2022-07-12 17:43:44 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:44 verena.dragonfear hass[873]: 2022-07-12 17:43:44 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:46 verena.dragonfear kernel: hrtimer: interrupt took 52969 ns
Jul 12 17:43:48 verena.dragonfear hass[873]: 2022-07-12 17:43:48 INFO (SyncWorker_9) [homeassistant.components.ubus.device_tracker] Checking hostapd
Jul 12 17:43:57 verena.dragonfear hass[873]: 2022-07-12 17:43:57 INFO (MainThread) [homeassistant.components.automation.nook_shortcut_button_toggles_lamp] Nook: button toggles lamp: Running automation actions
Jul 12 17:43:57 verena.dragonfear hass[873]: 2022-07-12 17:43:57 INFO (MainThread) [homeassistant.components.automation.nook_shortcut_button_toggles_lamp] Nook: button toggles lamp: Executing step call service
Jul 12 17:43:57 verena.dragonfear hass[873]: 2022-07-12 17:43:57 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:57 verena.dragonfear hass[873]: 2022-07-12 17:43:57 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:58 verena.dragonfear hass[873]: 2022-07-12 17:43:58 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:58 verena.dragonfear hass[873]: 2022-07-12 17:43:58 INFO (MainThread) [homeassistant.components.automation.nook_shortcut_button_toggles_lamp] Nook: button toggles lamp: Running automation actions
Jul 12 17:43:58 verena.dragonfear hass[873]: 2022-07-12 17:43:58 INFO (MainThread) [homeassistant.components.automation.nook_shortcut_button_toggles_lamp] Nook: button toggles lamp: Executing step call service
Jul 12 17:43:58 verena.dragonfear hass[873]: 2022-07-12 17:43:58 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:58 verena.dragonfear hass[873]: 2022-07-12 17:43:58 INFO (MainThread) [homeassistant.components.automation.nook_shortcut_button_toggles_lamp] Nook: button toggles lamp: Running automation actions
Jul 12 17:43:58 verena.dragonfear hass[873]: 2022-07-12 17:43:58 INFO (MainThread) [homeassistant.components.automation.nook_shortcut_button_toggles_lamp] Nook: button toggles lamp: Executing step call service
Jul 12 17:43:58 verena.dragonfear hass[873]: 2022-07-12 17:43:58 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:58 verena.dragonfear hass[873]: 2022-07-12 17:43:58 INFO (MainThread) [homeassistant.components.automation.nook_shortcut_button_toggles_lamp] Nook: button toggles lamp: Running automation actions
Jul 12 17:43:58 verena.dragonfear hass[873]: 2022-07-12 17:43:58 INFO (MainThread) [homeassistant.components.automation.nook_shortcut_button_toggles_lamp] Nook: button toggles lamp: Executing step call service
Jul 12 17:43:58 verena.dragonfear hass[873]: 2022-07-12 17:43:58 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:43:58 verena.dragonfear hass[873]: 2022-07-12 17:43:58 INFO (SyncWorker_7) [homeassistant.components.ubus.device_tracker] Checking hostapd
Jul 12 17:43:58 verena.dragonfear hass[873]: 2022-07-12 17:43:58 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
[... cut ...]
Jul 12 17:43:58 verena.dragonfear hass[873]: 2022-07-12 17:43:58 INFO (MainThread) [homeassistant.components.automation.nook_shortcut_button_toggles_lamp] Nook: button toggles lamp: Executing step call service
Jul 12 17:43:58 verena.dragonfear hass[873]: 2022-07-12 17:43:58 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
Jul 12 17:44:08 verena.dragonfear hass[873]: 2022-07-12 17:44:08 INFO (SyncWorker_5) [homeassistant.components.ubus.device_tracker] Checking hostapd
```

Rudd-O, 2022-07-12:
To me it looks like there is a resource leak somewhere, and the HA system runs out of "APSAckHandles" (whatever those are) shortly after startup.  I'm going to try and reload the integration and see if that makes a difference.

Rudd-O, 2022-07-12:
Hot damn, restarted Home Assistant and nothing is wrong now.  WTF.

Hang on, I may have spoken too soon tho.

Rudd-O, 2022-07-12:
Yeah spoke too soon.  I'm not seeing any errors in the logs right now, but my sensors definitely aren't updating.

[3 more replies]