zigpy/zigpy-zigate #93: zigpy-zigate 0.8.0 or 0.8.x release tag missing?

Hedda, 2021-06-08:
@doudz Looks like you bumped zigpy-zigate version to 0.8.0 but missed tagging it for release so not available via PyPI/pip?

ZeGuigui wrote this in https://github.com/zigpy/zigpy-zigate/commit/69b02f1c58f5642b032a7fccb90504bfd463cd0d

> Do you plan to add a tag so this release will be pushed to pypi.org or should I hard install it to test the reconnect feature?

https://github.com/zigpy/zigpy-zigate/releases

https://github.com/zigpy/zigpy-zigate/tags

https://pypi.org/project/zigpy-zigate/#history

https://pypi.org/project/zigpy-zigate/#files

Hedda, 2021-06-08:
Maybe also need to add/update workflows for release management?

Comparing

https://github.com/zigpy/zigpy/tree/dev/.github/workflows or https://github.com/zigpy/bellows/tree/dev/.github/workflows

verses

https://github.com/zigpy/zigpy-zigate/tree/dev/.github/workflows

doudz, 2021-06-09:
the release is not ready that's why I didn't tag it

Hedda, 2021-06-10:
> the release is not ready that's why I didn't tag it

OK. Just assumed based on https://github.com/zigpy/zigpy-zigate/commit/69b02f1c58f5642b032a7fccb90504bfd463cd0d bump as usually major and minor version bumps are done last just before release

Hedda, 2021-11-16:
@doudz Any chance you could test and bump the release version 0.8.1 

Home Assistant core looks to still be using zigpy-zigate 0.7.3 release:

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/manifest.json

https://github.com/home-assistant/core/pulls?q=zigpy-zigate

That does not contain the later fixes that you added after that.

See comment by @Xavier-M in https://github.com/zigpy/zigpy-zigate/issues/101 so maybe he is willing to test it.

doudz, 2021-11-17:
Sorry, no time to work on zigate 😞

MichaelBitard, 2022-01-22:
Hi @doudz, do you think you could release the 0.8.0? We could take care of the testing in home assistant, and do the PR to integrate it in ZHA if needed.

Hedda, 2022-01-25:
@MichaelBitard Check out the somewhat related discussion in https://github.com/zigpy/zigpy-zigate/issues/104 about help with test mentioned fork with Home Assistant.

Help if needed regression testing mentioned zigpy-zigate fork as if it does not break anything PRs can be submitted upstream.

Gemies, 2022-02-04:
That would be very much appreciated when 0.8.0 is released so at least.whe can test it. I am not relying on the Zigate wifi. So testing would be nice.

Hedda, 2022-02-06:
FYI, see the very related discussions in https://github.com/zigpy/zigpy-zigate/issues/116 and https://github.com/zigpy/zigpy/discussions/880 about zigpy-zigate needing new maintainers to volunteer first.