zigpy/zigpy-zigate #104: Home assistant error Received unhandled response 0x8012 tradfri IKEA

baimard, 2021-11-07:
``https://github.com/zigpy/zigpy-zigate/blob/818c3a1f76d183c962afb838afa5d083e8c04c31/zigpy_zigate/api.py#L22

My IKEA light send me a 0x8012 answer.

Is it possible to had this response ? 
```
home_assistant.1.mn6u4y319qpd@ocre    | 2021-11-07 14:18:38 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000009be00b7053001b7000300
home_assistant.1.mn6u4y319qpd@ocre    | 2021-11-07 14:18:38 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00b7053001b70003' LQI:0
home_assistant.1.mn6u4y319qpd@ocre    | 2021-11-07 14:18:38 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 183, 1328, b'\x01\xb7\x00\x03']
home_assistant.1.mn6u4y319qpd@ocre    | 2021-11-07 14:18:38 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8012000a47000101025231b700030a
home_assistant.1.mn6u4y319qpd@ocre    | 2021-11-07 14:18:38 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8012 b'000101025231b70003' LQI:10
home_assistant.1.mn6u4y319qpd@ocre    | 2021-11-07 14:18:38 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012

```

baimard, 2021-11-16:
```
RESPONSES = {
    0x004D: (t.NWK, t.EUI64, t.uint8_t, t.uint8_t),
    0x8000: (t.uint8_t, t.uint8_t, t.uint16_t, t.Bytes),
    0x8002: (t.uint8_t, t.uint16_t, t.uint16_t, t.uint8_t, t.uint8_t,
             t.Address, t.Address, t.Bytes),
    0x8009: (t.NWK, t.EUI64, t.uint16_t, t.uint64_t, t.uint8_t),
    0x8010: (t.uint16_t, t.uint16_t),
    0x8011: (t.uint8_t, t.NWK, t.uint8_t, t.uint16_t, t.uint8_t),
    0x8012: (t.uint32_t,),
    0x8017: (t.uint32_t,),
    0x8024: (t.uint8_t, t.NWK, t.EUI64, t.uint8_t),
    0x8035: (t.uint8_t, t.uint32_t),
    0x8048: (t.EUI64, t.uint8_t),
    0x8701: (t.uint8_t, t.uint8_t),
    0x8702: (t.uint8_t, t.uint8_t, t.uint8_t, t.Address, t.uint8_t),
    0x8806: (t.uint8_t,),
}
```

I just add this line : `    0x8012: (t.uint32_t,),`

nlhomme, 2021-11-28:
I'm having the same issue on Home Assistant with the ZHA integration:
```
Logger: zigpy_zigate.api
Source: /usr/local/lib/python3.9/site-packages/zigpy_zigate/api.py:115
First occurred: 12:03:31 (2 occurrences)
Last logged: 12:03:33
Received unhandled response 0x8012
```
My Zigbee controller is a Zigate v1 with firmware 3.1e

nlhomme, 2021-12-16:
It seems that this issue has been treated in #107

Hedda, 2022-01-03:
@baimard You might want to consider reopening this and @nlhomme might want to create a new pull request based on https://github.com/zigpy/zigpy-zigate/pull/107

See the last comment by pipiche38 in https://github.com/zigpy/zigpy-zigate/pull/107

> **_we are currently working on zigpy-zigate library under a fork dedicated to our project, we don't have the possibility to test it under the ZHA environment but if you are ready to do, you can eventually use it https://github.com/zigbeefordomoticz/zigpy-zigate_**
> 
> _**When ready we will be providing PRs against the main stream**_

Forking a project/library means that the library has been copied and some or any code patches (like new fixes or features) to forks might or might not be actively contributed back to the original upstream project by the developers that are working on that forked version.

In this case, it means that there is at least one patch in the fork https://github.com/zigbeefordomoticz/zigpy-zigate that is not available in upstream https://github.com/zigpy/zigpy-zigate but those patches in that fork are not currently being actively tested to check for any breakage in the ZHA integration component in Home Assistant by the developers working on that fork of zigpy-zigate as they are testing the library with another software application and do not have time or interest in regression testing with other applications using the same library, like ZHA.

As in, if you want those patches to be in upstream now rather hoping that they might be added later then regression testing of that fork (or just specific patches) is needed by other volunteers and then once confirmed not broken anything, someone, either the tester or some other volunteer will need to backport that and any other acceptable fixes from that fork back here to the upstream zigpy-zigate.

Terminology for reference:
* https://en.wikipedia.org/wiki/Fork_(software_development)
* https://en.wikipedia.org/wiki/Regression_testing
* https://en.wikipedia.org/wiki/Backporting

If you have the ability to test that fork with your setup then please do so and report your result here and also to https://github.com/zigpy/zigpy-zigate/pull/107 and https://github.com/zigpy/zigpy-zigate/pull/106

There is some more information about testing here that would need to be adapted when working with a forked library version: 

* https://github.com/zigpy/zigpy-zigate#testing-new-releases
* https://github.com/zigpy/zigpy/blob/dev/CONTRIBUTING.md
* https://github.com/zigpy/zha-device-handlers/#testing-quirks-in-development-in-docker-based-install

baimard, 2022-01-03:
done

pipiche38, 2022-01-03:
> @baimard You might want to consider reopening issue and @nlhomme might want to create a new pull request based on #107
> 
> See the last comment by pipiche38 in #107
> 
> > **_we are currently working on zigpy-zigate library under a fork dedicated to our project, we don't have the possibility to test it under the ZHA environment but if you are ready to do, you can eventually use it https://github.com/zigbeefordomoticz/zigpy-zigate_**
> > _**When ready we will be providing PRs against the main stream**_
> 
> Forking a project/library means that the library has been copied and some or any code patches (like new fixes or features) to forks might or might not be actively contributed back to the original upstream project by the developers that are working on that forked version.
> 
> In this case, it means that there is at least one patch in the fork https://github.com/zigbeefordomoticz/zigpy-zigate that is not available in upstream https://github.com/zigpy/zigpy-zigate but those patches in that fork are not currently being actively tested to check for any breakage in the ZHA integration component in Home Assistant by the developers working on that fork of zigpy-zigate as they are testing the library with another software application and do not have time or interest in regression testing with other applications using the same library, like ZHA.
> 
> As in, if you want those patches to be in upstream now rather than hoping that they might be added later then regression testing of that fork (or just specific patches) is needed by other volunteers and then once confirmed not broken anything, someone, either the tester or some other volunteer will need to backport that and any other acceptable fixes from that fork back here to the upstream zigpy-zigate.
> 
> Terminology for reference:
> 
>     * https://en.wikipedia.org/wiki/Fork_(software_development)
> 
>     * https://en.wikipedia.org/wiki/Regression_testing
> 
>     * https://en.wikipedia.org/wiki/Backporting
> 
> 
> If you have the ability to test that fork with your setup then please do so and report your result here and also to #107 and #106
> 
> There is some more information about testing here that would need to be adapted when working with a forked library version:
> 
>     * https://github.com/zigpy/zigpy-zigate#testing-new-releases
> 
>     * https://github.com/zigpy/zigpy/blob/dev/CONTRIBUTING.md
> 
>     * https://github.com/zigpy/zha-device-handlers/#testing-quirks-in-development-in-docker-based-install
> 
> 
> PS: As I understand it, the main reason why they choose to fork to https://github.com/zigbeefordomoticz/zigpy-zigate rather than just keep submitting pull requests to upstream https://github.com/zigpy/zigpy-zigate is that no zigpy developers who own any ZiGate hardware and have a personal interest in maintaining support for are currently actively working on maintaining upstream https://github.com/zigpy/zigpy-zigate thus new volunteering developers with a personal interest in maintaining support is needed.

I don't understand all of this communication from @hedda, if you are against what we are currently doing, tell us frankly , but so far the zigpy-zigate library is stuck and not maintained. So it is unbelievable to read such messages !

MattWestb, 2022-01-03:
I agree with @pipiche38 but if hi and some more like forking some part (or all) of the Zigpy libs and patching it like more devs have doing in other system its more than OK.
The best is they can making the patches upstream for easter maintaining there system but its not primary one Zigpy problem but its best for all if its possible doing it.

In the end if more support for different hardware /system then better for all parts and our community.

Also having more devs working with different parts is also one win win situation then getting more devices supported in the upstream (pipiche38 have helping getting more device added and some other system have doing the same).

I hope all is working well and getting all part in place and the system working with Zigpy libs as underlying radio lib.

Witch one great 2022 to all that is helping all working well in our system and many hours or nice "dream coeding" for getting more function in place !!

pipiche38, 2022-01-03:
@MattWestb this is exactly my point, we are currently trying to enhance the zigpy-zigate layer, because we benefits from all good work done on zigpy and especially zigpy-znp.
At the end we are not interested to use zigpy-zigate as we have our own native way to communicate with zigate and for now there is no plan. Our approach was to get zigpy-zigate to a point where it is usable , now if the zigpy community @Hedda  feels this is not interested for them, fine.

Hedda, 2022-01-07:
> I don't understand all of this communication from @Hedda, if you are against what we are currently doing, tell us frankly , but so far the zigpy-zigate library is stuck and not maintained. So it is unbelievable to read such messages !

@Hedda Sorry for the full on misunderstanding. I am not against what you are doing at all, not in any way. I am very much a fan of what you are trying to achieve and think what you are doing is a very good thing. I think that it is great that more projects are trying to use zigpy. 

I was just trying to inform others what it means by a fork as do not think that was made clear when the pull requests were closed.

Personally, I understand why you forked and understand that is will hopefully not be your long term goal but something that has to be done in order to speed up the development process. 

I do hope that you at some point get to having a stable version of zigpy-zigate, no matter if it is a fork or not.

baimard, 2022-01-10:
> @Hedda <https://github.com/Hedda> I have been contacted by several HA
> users willing to test the current fork that we have. Is there any one
> from HA whom can give a short procedure on how to shift from the
> current zigpy-zigate package to the one available here
> https://github.com/zigbeefordomoticz/zigpy-zigate
>
> Currently we have made changes to:
>
>   * Support Zigbee group and broadcast
>   * Support Zigate firmware 321 (which is the minimum)
>   * Support and handle the message flow ( 8011/8012/8702 )

Maybe a quick procedure can be to replace the package by this one in :

/usr/local/lib/python3.9/site-packages/zigpy_zigate

Cordialement,
Benjamin AIMARD

pipiche38, 2022-01-10:
Don't know if that will be enought to overwrite the standard package

pip3 install git+https://github.com/zigbeefordomoticz/zigpy-zigate.git

baimard, 2022-01-10:
It maybe better to directly replace the package, because of docker user.

With this command I don't know if you replace existing package ...

Cordialement,
Benjamin AIMARD

pipiche38, 2022-01-10:
So this is out of my knowledge

baimard, 2022-01-10:
git clone https://github.com/zigbeefordomoticz/zigpy-zigate.git

mv /usr/local/lib/python3.9/site-packages/zigpy_zigate 
/usr/local/lib/python3.9/site-packages/zigpy_zigate.back

cp -r ./zigpy-zigate /usr/local/lib/python3.9/site-packages

C'est tout :)

Cordialement,
Benjamin AIMARD

Gemies, 2022-01-10:
Benjamin,

Thanks a lot for you're small tutorial on how to update the zigpy-zigate library. 
Now I have to find a way to acces usr/local/lib/python3.9...........  file. From the command line I don't get acces to the file location. 
Any ideas?

Thanks alot?
Gr, Gemies

Hedda, 2022-01-11:
> @Hedda I have been contacted by several HA users willing to test the current fork that we have. Is there any one from HA whom can give a short procedure on how to shift from the current zigpy-zigate package to the one available here https://github.com/zigbeefordomoticz/zigpy-zigate

Sorry I'm not good with either Linux or Python packages myself, and especially not when the use of Docker containers is involved.

Home Assistant have this official documentation on adding new dependencies to a test environment:

https://developers.home-assistant.io/docs/development_testing/#adding-new-dependencies-to-test-environment

I'm not sure how it works if want to test alternative Python packages from alternative sources when a users is using "Home Assistant Container" "Home Assistant Operating System" (a.k.a. "HAOS" Home Assistant OS embedded Linux distribution) installations which both uses/depends on Docker containers? 

Probably a lot easier for testers using Home Assistant without Docker, as is when have only installed "Home Assistant Core" directly under any other Linux distribution. The differences between the different installation methods of Home Assistant are described here:

https://www.home-assistant.io/installation/

> Don't know if that will be enought to overwrite the standard package
> 
> pip3 install git+https://github.com/zigbeefordomoticz/zigpy-zigate.git

Probably first want to start by uninstalling whichever existing zigpy-zigate package which is already installed:

```console
python -m pip uninstall zigpy-zigate
```

Then do a pip install of "branch" (where replace "branch.zip" with branch) that want ZHA users with ZiGate to test, with either:

pip install https://github.com/zigbeefordomoticz/zigpy-zigate/archive/branch.zip

Example:
```console
pip install https://github.com/zigbeefordomoticz/zigpy-zigate/archive/dev.zip
```

or to download each file individually via git as you wrote which takes longer but might be simpler (again replace "branch"):

pip install git+https://github.com/zigbeefordomoticz/zigpy-zigate.git@branch

Example:
```console
pip install git+https://github.com/zigbeefordomoticz/zigpy-zigate.git@dev
```

PS: Many used to use a this "zha-custom-radios" custom component to test zigpy-znp and other custom radio modules  in Home Assistant via HACS (Home Assistant Community Store) before zigpy-znp was merged into the mainline which made testing simple: 

https://github.com/zha-ng/zha-custom-radios

Today that is not maintained and I guess would have to be updated and maybe even rebased on this to make updating tests easier:

https://github.com/MatthewFlamm/pytest-homeassistant-custom-component

PPS: Maybe start new issue with the subject topic "Testing fork of zigpy-zigate"? -> https://github.com/zigpy/zigpy-zigate/issues/

Hedda, 2022-01-11:
Once got a working guide for an easy way to test specific branches from other forks it might be good idea to add that to README?

https://github.com/zigpy/zigpy-zigate/blob/dev/README.md#testing-new-releases

pipiche38, 2022-01-11:
@Gemies I think the best is for you to use this Issue to report any problem/issue. I'm sure you'll find some help

Hedda, 2022-01-11:
FYI, also now posted a general plea for help with testing in Home Assistant community forum here:

https://community.home-assistant.io/t/zigate-users-help-wanted-to-test-zha-with-fork-of-zigpy-zigate-with-new-patches/378362

baimard, 2022-01-11:
@Gemies are you on Linux Windows Docker ?

Hedda, 2022-01-11:
> are you on Linux Windows Docker ?

It should maybe also be that Home Assistant is officially not supported on Microsoft Windows OS or Mac OS, only Linux.

Microsoft Windows OS or Mac OS is only options if install Home Assistant in Linux running inside a virtual machine (VM).

https://www.home-assistant.io/installation/

baimard, 2022-01-11:
Even with WSL on windows with a docker container ? ...

Gemies, 2022-01-11:
@baimard I run HA on Synology Nas with a VM.

baimard, 2022-01-11:
> @baimard I run HA on Synology Nas with a VM.

I'm sorry I don't know where you can find the package ... try with `find` command

Hedda, 2022-01-11:
> I run HA on Synology Nas with a VM.

Which exact Linux distribution are you running inside that VM? Are you running "Home Assistant Operating System" or ... ?

https://www.home-assistant.io/installation/

> Even with WSL on windows with a docker container ? ...

Yes I believe so, at least as I understand it. That might work differently now however with the later versions of Windows Subsystem for Linux in Microsoft Windows operating systems are getting more and more compatible with real Linux with each newer version.

You need to understand and remember that WSL (Windows Subsystem for Linux) is not a full Linux distribution. WSL only a compatibility layer for running Linux binary executables (in ELF format) natively on Microsoft Windows 10 and Windows 11.

So I think it is very a bad idea to use WSL on Windows as a testing environment platform when it is not officially supported.

Anyway, I know that if you would [submit a ZHA bug-report to Home Assistant core](https://github.com/home-assistant/core/issues) then they will close it if you are not running it under either Home Assistant OS or any other proper Linux distribution. Thus it is really a moot point to spend time testing in WSL.

Gemies, 2022-01-11:
I use the vm manager from Synology. Installed the official HA OS image which provided by home assistant on the website.

https://www.home-assistant.io/installation/alternative/#install-home-assistant-operating-system

Hedda, 2022-01-18:
> The purpose of zigpy is to be cross plateform ? (domoticz/ha/jeedom/ect ?)
> So if it work on domoticz that should work on home assistant. isn't it ?

@max5962 What are you asking? FYI, see back-story on Zigbee for Domoticz in zigpy general discussion here -> https://github.com/zigpy/zigpy/discussions/865

Xavier-M, 2022-01-18:
Thank you all for your work on this.

I think I was able to install this into hassio. I have the plugin "SSH & Web Terminal", I then connect with SSH and enter the following lines (it's an image, i cannot copy paste using this...):
[image]

After that I restarted home assistant. But then I don't really know how I can know if it is taken into account, or if I have to do something else for it to work ;-)
In the log I have things like that:
```
2022-01-18 16:28:02 ERROR (MainThread) [zigpy_zigate.api] data_received : error details received 0x9999 error:0x83 
2022-01-18 16:28:02 ERROR (MainThread) [zigpy_zigate.api] command : error status cmd:0x530 error:163
2022-01-18 16:28:03 ERROR (MainThread) [zigpy_zigate.api] data_received : error details received 0x9999 error:0x83 
2022-01-18 16:28:03 ERROR (MainThread) [zigpy_zigate.api] command : error status cmd:0x530 error:163
2022-01-18 16:28:04 ERROR (MainThread) [zigpy_zigate.api] data_received : error details received 0x9999 error:0x83 
2022-01-18 16:29:53 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback SerialTransport._read_ready()
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.9/site-packages/serial_asyncio/__init__.py", line 119, in _read_ready
    self._protocol.data_received(data)
  File "/config/deps/lib/python3.9/site-packages/zigpy_zigate/uart.py", line 90, in data_received
    self._api.data_received(cmd, f_data, lqi)
  File "/config/deps/lib/python3.9/site-packages/zigpy_zigate/api.py", line 213, in data_received
    fut.set_result((data, lqi))
asyncio.exceptions.InvalidStateError: invalid state
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.9/site-packages/serial_asyncio/__init__.py", line 119, in _read_ready
    self._protocol.data_received(data)
  File "/config/deps/lib/python3.9/site-packages/zigpy_zigate/uart.py", line 90, in data_received
    self._api.data_received(cmd, f_data, lqi)
  File "/config/deps/lib/python3.9/site-packages/zigpy_zigate/api.py", line 213, in data_received
    fut.set_result((data, lqi))
```

pipiche38, 2022-01-18:
> 2022-01-18 16:28:02 ERROR (MainThread) [zigpy_zigate.api] data_received : error details received 0x9999 error:0x83
> 2022-01-18 16:28:02 ERROR (MainThread) [zigpy_zigate.api] command : error status [...]

These are ZiGate errors reported by the firmware via 0x9999 .
"No free APS acknowledgement handles (resource error) - the number of handles is set in the Maximum Number of Simultaneous Data Requests with Acks field of the APS layer configuration section of the ZPS Configuration Editor"

For the stack trace, it would be important to have more debug to understand what was the context. So better to enable debug for zigpy-zigate

baimard, 2022-07-19:
everythings work well now !

closed 2022-07-19 by baimard

baimard, 2022-07-19:
[image]

baimard, 2022-07-20:
When I change this value on this new parameter, everything seems to be ok.

[image]