zigpy/zigpy-zigate #111: HA [zigpy_zigate.api] Received unhandled response 0x8008

romsworld, 2021-12-19:
Hi,

I have a Zigate+ on RPI4 and I have this error every minute on HA log. Do you know this problem,

Thanks

2021-12-19 20:55:35 WARNING (MainThread) [zigpy.application] Device is partially initialized: <ZiGateDevice model='ZiGate USB-TTL 3.21' manuf='ZiGate' nwk=0x0000 ieee=00:15:xx:xx:xx:xx:xx:xx is_initialized=False>
2021-12-19 20:55:50 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8008
2021-12-19 20:56:49 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8008
2021-12-19 20:57:49 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8008
2021-12-19 20:58:49 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8008
2021-12-19 20:59:48 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8008
2021-12-19 21:00:48 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8008
2021-12-19 21:01:48 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8008
2021-12-19 21:02:48 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8008

pipiche38, 2021-12-30:
This is surprising, I was under the impression that this feature was not realeased !
This is an heartbeat that the Zigate is sending !

@fairecasoimeme

fairecasoimeme, 2022-01-03:
Hi,
This feature (heartbeat) is usefull for ZHA user (or zigpy-zigate) with ZiGate-WiFi when there is few zigbee devices.
If an update of the library occur which solve the TCP connection issue https://github.com/zigpy/zigpy-zigate/issues/89, I'll rollback this feature.

PS: this feature is only used with zigate RAW mode

pipiche38, 2022-01-03:
This feature is counter productive as this is the purpose of the zigpy layer to maintain the connection.