zigpy/zigpy-zigate #116: Campaign to find additional maintainers of this upstream zigpy-zigate repository?

Hedda, 2022-01-25:
@doudz expressed that he, unfortunately, do not have enough spare time to spare on actively developing and maintaining zigpy-zigate.

Wondering if it would be a good idea to try to find additional maintainers that can help test, verify, commit pull requests, and do releases?

I'm not suggesting to push out @doudz and not volunteering myself, but don't think any will step up until make a public request for help?

Believe getting more testers onboard might help a little but think ultimately need developers with interest who also use ZiGate hardware. 

Guess I'm suggesting asking previous contributors as well as posting some kind of plea for help in both ZiGate and ZHA communities?

As I understand, [zigpy "organization" on GitHub]( https://github.com/zigpy/) was set up for loose collboration and multiple developers already who have access to repositories within the zigpy organization (and [matching PyPI published packages](https://pypi.org/search/?q=zigpy)) where others are acting as the primary developer, which is a great concept to hopefully avoid having to fork if a developer no longer has interest or time to spend on one of the projects that is part of the zigpy organization. Note that "organization" here just referred to GitHub term for a grouping of Git repositories for collaboration purposes.

_Originally posted by @doudz in https://github.com/zigpy/zigpy-zigate/issues/93#issuecomment-971954177_
> Sorry, no time to work on zigate 😞

_Originally posted by @doudz in https://github.com/zigpy/zigpy-zigate/pull/113#issuecomment-1016853361_
> I just have no time to work on this repo but PR are welcome :)

Perhaps @fairecasoimeme @MichaelBitard @nlhomme @Jsl-1 @onkelbeh @devbis @badzz @pipiche38 know of potential candidates?

nlhomme, 2022-01-26:
I'm sorry, I don't have the skills to maintain the code or make reviews :(

Plus I don't have the time to test, and I don't really wish to as I am trying to keep my Home Assistant instance the most stable as possible.

The best I can do is to open issues (with logs extracts, screenshots etc...) and test the proposed fixes in the context of thoose issues

Hedda, 2022-02-04:
Please also check out the very much related discussion here -> https://github.com/zigpy/zigpy/discussions/880

ZiGate users help wanted to test ZHA with fork of zigpy-zigate with new patches.