zigpy/zigpy-zigate #143: How is the data of zdo and zcl sent and received?

liangyingy, 2022-11-01:
We received this [request](https://github.com/Xinyuan-LilyGO/T-ZigBee/issues/3).

I'm learning how zigpy works?

How is the data of zdo and zcl sent and received?

puddly, 2022-11-01:
Zigpy expects low-level access to the stack. Radio libraries are expected to implement only a few methods:

 - `send_packet`/`packet_received`, which sends/receives [`ZigbeePacket` objects](https://github.com/zigpy/zigpy/blob/f6d7e4b510492765c86b10568ac5f01709decf94/zigpy/types/named.py#L560-L597)
 - `load_network_info`/`write_network_info`, which read/write `NetworkInfo` and `NodeInfo` to/from the NCP: https://github.com/zigpy/zigpy/blob/8a2c7548065367d7e94ddf5857652ca1cd087e70/zigpy/state.py#L81
 - `connect`/`disconnect`

That being said, it seems that your radio is based on the TLSR8258 chip. Is there any widely-available USB stick using this chip? If not, there is not going to be much of a benefit to writing a new radio library when we already have good support for CC2652, EFR32, and Conbee chips.

---

@Hedda I appreciate your enthusiasm but please do not recruit developers to write new radio libraries without first consulting with the zigpy team.

The maintenance burden of the existing legacy zigpy radio libraries is significant. Adding yet another radio type will not help with Zigbee adoption when we already have five to choose from, it will just increase the amount of time it takes to make changes to zigpy.

AskAlice, 2022-12-07:
I just wrote a huge paragraph but I think Doing a little digging I see zigpy is mostly used for coordinators, but could it be used in a device context? Communicating with the esp32 over serial, then relaying that over to another zigpy-controlled device (z2m or zha) ?

puddly, 2022-12-07:
> Doing a little digging I see zigpy is mostly used for coordinators, but could it be used in a device context?

Totally, but only if the firmware can be configured to join a network. There's no standard zigpy API for this since we have never needed to do it. I believe the EmberZNet and ZNP coordinators can be configured to join a network as a router but I have not tried it in a very long time.

> which leaves me fairly in the dark about how to implement custom messaging that doesn't conform to ZCL or ZDO.

Custom messaging can be implemented with ZCL if you define a manufacturer-specific cluster with manufacturer-specific commands. You'd just have the overhead of the ZCL header, which is just a couple of bytes. No reason not to use ZCL.

Multi-device can also be done with multi-PAN commands, but I have only tested that with a Z-Stack radio and it's not really all that useful, since I don't believe the packets are relayed, nor are they encrypted.

MattWestb, 2022-12-07:
@AskAlice You can (normally) only having one coordinator in one Zigbee network (or more true one trust center pair) that is controlling all devices.
But if you can compiling own firmware you can doing one `Zigbee Control Bridge` that you having one serial port and CLI and can sending and receiving commands over the comport and controlling all devices in the network. Its the same way IKEA Tradfri is working and you can communicating with all devices in the network but not doing the trust center things.
Search for it and you is finding one easy way doing it.

PS: I have cooking firmware for the old IKEA module with it and its working but i have not testing connecting it ESP32 but shall working OK.